schreibfaul1/ESP32-audioI2S #920: intergrate you library to the Arduino library manager

RASPIAUDIO, 2024-12-02:
Hello 
It could be great if your great library could be available directly in the arduino IDE, it will ease the use of you library for many users. 
But maybe is there a reason why you have not made it already?

All you need to do is to add a library.properties here an example : 
https://github.com/RASPIAUDIO/Muse_library/blob/main/library.properties

and do a PR to https://github.com/arduino/library-registry

KMTsvetanov, 2024-12-03:
I tried to add this library to Arduino IDE and I got: 
Multiple libraries were found for "SD.h"

did you run into this?

KMTsvetanov, 2024-12-03:
> Hello,
> 
> It would be great if your library were available directly in the Arduino IDE. This would make it much easier for many users to utilize your library.
> 
> Is there a particular reason why it hasn't been added already?
> 
> All you need to do is add a library.properties file. Here's an example for reference: [Example library.properties](https://github.com/RASPIAUDIO/Muse_library/blob/main/library.properties)
> 
> Then, you can submit a pull request to the [Arduino Library Registry](https://github.com/arduino/library-registry).

I tried to add this library to Arduino IDE and I got:
Multiple libraries were found for "SD.h"

did you run into this?

RASPIAUDIO, 2024-12-03:
> > Hello,
> > It would be great if your library were available directly in the Arduino IDE. This would make it much easier for many users to utilize your library.
> > Is there a particular reason why it hasn't been added already?
> > All you need to do is add a library.properties file. Here's an example for reference: [Example library.properties](https://github.com/RASPIAUDIO/Muse_library/blob/main/library.properties)
> > Then, you can submit a pull request to the [Arduino Library Registry](https://github.com/arduino/library-registry).
> 
> I tried to add this library to Arduino IDE (sketch -> include library from zip) and I got: Multiple libraries were found for "SD.h"
> 
> did you run into that?

It's not the subject of this topic.

LucasSaintrain, 2024-12-31:
Hello, I also wanted to have this library at the Arduino Library Registry.
It's stragne that this wasn't done already considering that this project already has a library.properties file.

I've done myself a pull request to the registry, adding the github url to the list:
https://github.com/arduino/library-registry/pull/5705

The pull request was immediatly accepted and the bot said:
"The library(s) will be available for installation via Library Manager within a day's time."

By the way, I'm not a contributor to this library, I'm just interested in using it.

RASPIAUDIO, 2024-12-31:
I did not want to submit it myself as I didn't want to take any credit for this, but I'm glad you did ,😅

closed 2026-03-23 by github-actions[bot]