schreibfaul1/ESP32-audioI2S #927: Is there any function that can get the total duration of a song?

Explorerlowi, 2024-12-15:
I tried audio.getAudioFileDuration() but it's not correct.

Explorerlowi, 2024-12-15:
Sorry, this is the function, but it has some errors. It is very large in the early stage of the song, about tens of seconds, and gradually decreases as the music plays. There is still an error of about 5 seconds at the end.

Pako2, 2024-12-16:
This is because you are using a variable bit rate. There is no way to calculate it exactly here. See the [wiki](https://github.com/schreibfaul1/ESP32-audioI2S/wiki):
`getAudioFileDuration() Indicates the expected length of an audio file in seconds. With a constant bit rate, CBR, the value is exact, with a variable bit rate, VBR, the duration is estimated based on the first 100 mp3 frames and can therefore deviate slightly from the actual playback time`

Explorerlowi, 2024-12-18:
Thanks for your answer. I have a new problem now. I often encounter the following problem when playing music from the SD card:
`audio_info: MP3 decode error -6: INVALID_FRAMEHEADER`
This will cause the music progress to jump and the `void audio_eof_mp3(const char *info)` cannot be triggered. 

Sometimes it will also have more serious problems as follows, which will cause esp32 to restart:
`[Audio.cpp:125] bytesWritten(): m_writePtr 1032508517, m_endPtr 1032508506`
`audio_info: MP3 decode error -1 : INDATA_UNDERFLOW`
`audio_info: MP3 decode error -6: INVALID_FRAMEHEADER`
After restarting, it will prompt that the SD card initialization failed:
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x403833a0  PS      : 0x00060133  A0      : 0x803839b8  A1      : 0x3fcb6ae0  
A2      : 0x3d800014  A3      : 0x3d8b6ba0  A4      : 0xdfb225ce  A5      : 0x05034bc4  
A6      : 0x00060f20  A7      : 0x00000001  A8      : 0x0000001a  A9      : 0x00000010  
A10     : 0xc05817f1  A11     : 0x00000914  A12     : 0x3d800038  A13     : 0x3d80007c  
A14     : 0x00060123  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xdfb225da  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x4038339d:0x3fcb6ae0 0x403839b5:0x3fcb6b00 0x40383ae4:0x3fcb6b20 0x403783a3:0x3fcb6b40 0x4037846a:0x3fcb6b70 0x403780e5:0x3fcb6bc0 0x4003ecad:0x3fcb6be0 |<-CORRUPTED

ELF file SHA256: cd27126af446285b

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377934
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   390][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   436][W][sd_diskio.cpp:186] sdCommand(): token error [0] 0x6
[   436][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[   437][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
SD Card initialization failed!
```

Is this a hardware problem or something else? This is my wiring diagram:
[image]

Pako2, 2024-12-18:
I don't have any specific explanation for the problem. However, I can say that I have similar problems, but with .flac and especially .aac formats. I also sometimes get some error during decoding and it has the same result - "audio_eof_mp3" is not triggered.
Things got a little better when I used a faster SD card.
Note: I am using the SD_MMC library (and 1-bit mode).

schreibfaul1, 2024-12-18:
I don't think the SD card is the problem. If you use SD_MMC, the SD card can be connected directly to the GPIOs (Vcc, GND, D0, CMD, CLK), which is better than your card adapter, especially since a level shifter is not required.

Pako2, 2024-12-18:
I have an SD card connected directly to the GPIOs (of course I have pull-up resistors there). But that is not the case of the questioner, as can be seen in the picture he attached. Fortunately in my case it is rare that the error occurs. If I notice any pattern, I will open a new issue.

schreibfaul1, 2024-12-18:
Pull-ups are not necessary with SD_MMC, but they do not interfere either. A good connection is much more important; small contact problems are often enough to trigger a malfunction. This is a possible cause with a breadboard.

Crazymanmusings, 2025-01-15:
> Thanks for your answer. I have a new problem now. I often encounter the following problem when playing music from the SD card: `audio_info: MP3 decode error -6: INVALID_FRAMEHEADER` This will cause the music progress to jump and the `void audio_eof_mp3(const char *info)` cannot be triggered.
> 
Not sure if you solved your problem but you may want to check the sd-card module as it may be a 5 volt module and may work intermittently on 3.3 v.