schreibfaul1/ESP32-audioI2S #935: Some m3u8s content are not parsed correctly

mifarelight, 2024-12-28:
A series of station (e.g.: https://hls.nntv.cn/nnradio/895_24.m3u8) can be played normally with VLC, but ESP32-audioI2S cannot play it normally. Its characteristics:
The content returned when accessing the m3u8 link for the first time is a link to another url (needs to be redirected, like https://numbers.livehwc3.cn/hls.nntv.cn/nnradio/895_24.m3u8?sub_m3u8=true&edge_slice=true&user_session_id=numbers), and ESP32-audioI2S does not redirect to that new url.

Is it caused by the station m3u8 content not meeting the standard?

mifarelight, 2024-12-28:
When analyzing the m3u8 content, because no line containing CODECS=mp4a, the second line "#EXT-X-VERSION:3" is taken as part of the link, resulting in the error "HTTP/1.1 404 Not Found".

schreibfaul1, 2024-12-29:
I can't get any further with the stream https://hls.nntv.cn/nnradio/895_24.m3u8.
A code string "mp4a...." is expected in the m3u8 playlist, which is missing here.
If I assume that it is aac, the second ts packet is invalid. TS packets usually start with 0x47 and are 188 bytes long. Something is different, the stream is added to the list of unsolved problems.

mifarelight, 2024-12-29:
Thanks for your attention to this issue.

I also noticed that the m3u8 content of this radio station did not conform to the conventional standard format requirements.

The reason why the VLC player application (usually running on a CPU with more abundant resources) can successfully parse and play it may be that it uses a relatively fuzzy parsing and matching method.

mifarelight, 2024-12-29:
I tried to access the links provided by m3u8 step by step with the FireFox browser, and sometimes a 404 error was reported, but from the successfully downloaded .ts file (attached below), it did start with 0x47.
[895_24_208.zip](https://github.com/user-attachments/files/18269608/895_24_208.zip)

schreibfaul1, 2024-12-29:
Your ts file looks good, every 188 bytes comes 0x47 for the start of the packet. 
[image]
The question is why it doesn't work.

schreibfaul1, 2024-12-30:
It's not perfect, but I now understand how this stream works

closed 2024-12-31 by mifarelight