schreibfaul1/ESP32-audioI2S #937: Problem playing mp3 files

lormaccone, 2024-12-30:
I started having problems playing mp3 files that played fine with an earlier version of this same library. Some mp3s (e.g.  eagle_landed.mp3) work fine, others play at increased speed (e.g. A17A1203615.mp3). 

One of the mp3s (launch.mp3) causes a bootloop error:
E (16500) I2S: dma error, interrupt status: 0x00b83fc0

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace: 0x4008385d:0x3ffc040c |<-CORRUPTED

All of them used to work with an earlier version of the library (which compiles fine but no playback (only silence) with my upgraded arduino 1.8.19).
[z.zip](https://github.com/user-attachments/files/18273032/z.zip)

p.s. Great library, by the way! Thanks

schreibfaul1, 2024-12-30:
Hi [lormaccone](https://github.com/lormaccone),
All three files work as expected for me. The A17A1203615.mp3 has a runtime of 18s, just like in the VLC player.
In contrast to earlier versions of this library, audio decoding and output runs via a separate task. As a result, audi.loop() is no longer time-critical and can be called much more slowly. Try inserting a delay for load balancing (1ms...10ms are good values):
[image]
The new internal task requires 3300 DWORDs. If the memory is running low, the Arduino task can be reduced by this amount.

lormaccone, 2024-12-30:
Thanks for your immediate reply!!! I really appreciate your help. 

I pruned down my sketch to the bare minimum, and the playback is still too fast.  I'm using an ESP32 WROOM 32 using the arduino IDE. There must be something weird with my setup. Thanks.

Here is my sketch:

#include "FS.h"
#include "SPIFFS.h"
#include "Audio.h"
#define FORMAT_SPIFFS_IF_FAILED false
Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);

void setup(void) {
  if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)){
      Serial.println("SPIFFS Mount Failed");
      return;
   }
  //   listDir(SPIFFS, "/", 0);
   audio.setVolume(21); // 0...21 

   
 audio.connecttoFS(SPIFFS,"A17A1203615.mp3");
  do{audio.loop();vTaskDelay(1);}while(audio.isRunning());
}

void loop(){delay(1);}

schreibfaul1, 2024-12-30:
Use `vTaskDelay(1)` instead of `delay(1)`
This allows you to balance the load between the tasks.

lormaccone, 2024-12-31:
Actually, delay(1) gets called AFTER all the mp3s have stopped playing, so it doesn't matter. I used vTasksDelay(1) during playback as per your suggestion. 

As a further example, I've changed the sketch using the same structure as the TalkingClock.ino example. I can confirm what I said: A17A1203615.mp3 plays back at about 4x speed, one-small-step-for-man.mp3 plays fine and launch.mp3 causes a bootloop with error (different from the one above given by the full sketch):
Guru Meditation Error: 
Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
[...] Backtrace: 0x4008ba21:0x3ffbeecc |<-CORRUPTED
[...] Rebooting...

I include my stripped down sketch. Thanks a lot!!!!

#include "FS.h"
#include "SPIFFS.h"
#include "Audio.h"

#define FORMAT_SPIFFS_IF_FAILED false
Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);

int8_t timefile = 3;

void setup(void) {
  Serial.begin(115200);

  if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)){
      Serial.println("SPIFFS Mount Failed");
      return;
   }
  //   listDir(SPIFFS, "/", 0);
   audio.setVolume(21); // 0...21 
   
 audio.connecttoFS(SPIFFS,"A17A1203615.mp3");
}

void loop(){audio.loop();vTaskDelay(1);}

void audio_eof_mp3(const char *info){  //end of file
    //Serial.printf("file :%s\n", info);
    if(timefile>0){
      Serial.println(timefile);
        if(timefile==1){ audio.connecttoFS(SPIFFS,"launch.mp3"); timefile--;}  // stroke
        if(timefile==2){ audio.connecttoFS(SPIFFS,"A17A1203615.mp3");     timefile--;}  // precisely
        if(timefile==3){ audio.connecttoFS(SPIFFS,"one-small-step-for-man.mp3"); timefile--;}
    }
}

lormaccone, 2025-01-02:
Well that's not very useful, since the problem clearly lies with the part you changed: I'm not using a I2S amplifier, but I'm using the internal DAC of the ESP32, note the "Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);" declaration. (compare this with the ESP32-audioI2S/examples/Internal DAC/internalDAC.ino example).

Now I have updated the esp32 board using the arduino ide and the sketch doesn't even compile anymore: it complains about

'I2S_DAC_CHANNEL_BOTH_EN' was not declared in this scope

It did compile before the upgrade. And it did work fine with an older version of this library. So at some point there was a regression... By the way, I tried compiling the above example (Internal DAC/internalDAC.ino), and it fails to compile with the same error.

lormaccone, 2025-01-02:
Ok, I managed to fix it by reverting the library to an older version (that I had kept on my computer) and by switching to the ESP32 version 1.0.5. Now everything works. 
Clearly, the latest version of ESP32 (3.x) doesn't support I2S_DAC_CHANNEL_BOTH_EN anymore.
Thanks!

schreibfaul1, 2025-01-02:
I had not tested it with the internal DAC. As of Arduino V2 there are problems with low bit rates, https://github.com/espressif/esp-idf/pull/8327 at the bottom. If you increase the bit rate of the audio files, you can use this library with Arduino version 2..0.15. From Arduino V3 it only works with an external DAC.

lormaccone, 2025-01-03:
No, apparently it does not even compile with the latest ESP32 update (see previous error) and that is true also for the stock example ESP32-audioI2S/examples/Internal DAC/internalDAC.ino

In any case, for me the issue is resolved, as I have managed to compile and everything works by reverting to ESP32 version 1.0.5. Thanks for your time

closed 2025-01-03 by lormaccone