schreibfaul1/ESP32-audioI2S #940: All Https type stations result in "Request (URL) failed!" when

renatoianhez, 2025-01-01:
I finally updated the library and the ESP32-Arduino. The ESP32-Arduino version I was using was 2.0.11. I did this after a fresh installation of Arduino IDE 2.3.4. It compiled normally and worked as expected.
However, all the stations whose address was "Https" resulted in "Request (URL) failed!".
The code is:
```
#include <ESP_I2S.h>
#include <wav_header.h>
#include <Arduino.h>
#include <TFT_eSPI.h> // Biblioteca da tela
#include <SPI.h> // SPI
#include <WiFiMulti.h> // Pra mais de um local de WiFi
#include "Audio.h" // Biblioteca do audio
#include "examples/320 x 240/Free_Font_Demo/Free_Fonts.h" // Fontes
#include "BluetoothA2DPSink.h" // Para o Bluetooth
#include <HTTPClient.h>

#define canalPraCima  0   // 0 nos antigos - 15 nos Wrover - Botoes para mudar a estacao
#define canalPraBaixo 4
#define I2S_DOUT      25  // Pinos do audio I2S
#define I2S_BCLK      27
#define I2S_LRCK      26
#define modoAparelho  34  // Chave seletora radio-bluetooth

bool mudouEstacao;
int radioTocando;
char* tituloBT;
char* artistaBT;
char* albumBT;

TFT_eSPI tela = TFT_eSPI(); // ATENCAO os pinos da tela sao definidos na biblioteca
Audio audio;
I2SClass i2s;
BluetoothA2DPSink a2dp_sink(i2s);
WiFiMulti outrosWiFi;
HTTPClient http;
bool connected = true;

//Radios e enderecos
String conteudoBaixado;
String tudo[120];
String nomeRadio[30];
String tipoRadio[30];
String cidadeRadio[30];
String urlRadio[30];
String endTudo = "https://gist.githubusercontent.com/renatoianhez/4b5726d5ef32ad0a24fa3db6521989d6/raw";
int indice = 0;

void baixaDados(const String& urlDado, int maxTokens, String* tokens) {
  http.begin(urlDado);
  int httpCode = http.GET();
  if (httpCode == HTTP_CODE_OK) {
    conteudoBaixado = http.getString();
    Serial.println("Arquivo de dados das radios baixado com sucesso");
  }
  else {
    Serial.printf("Falha ao baixar o arquivo. Código de erro HTTP: %d\n", httpCode);
  }
  http.end();
  int startIndex = 0;
  int endIndex = 0;
  int tokenIndex = 0;
  while (endIndex >= 0 && tokenIndex < maxTokens) {
    endIndex = conteudoBaixado.indexOf('\n', startIndex);
    if (endIndex >= 0) {
      tokens[tokenIndex] = conteudoBaixado.substring(startIndex, endIndex);
      startIndex = endIndex + 1;
    } else {
      tokens[tokenIndex] = conteudoBaixado.substring(startIndex);
    }
    tokenIndex++;
  }
[... 310 lines cut ...]
    tipoRadio[indice] = tudo[(j + 1)];
    cidadeRadio[indice] = tudo[(j + 2)];
    urlRadio[indice] = tudo[(j + 3)];
    indice++;
  }
}

void loop() {
  audio.loop();
  vTaskDelay(20);
  if (digitalRead(canalPraCima) == LOW) {  //Se apertar o botao pra subir a estacao
    sobeCanal();
    mudouEstacao = true;  // Avisa la que mudou de estacao!
  }
  if (digitalRead(canalPraBaixo) == LOW) {  //Se apertar o botao para descer a estacao:
    desceCanal();
    mudouEstacao = true;  //Avisa la que mudou de estacao tambem!
  }
  if (mudouEstacao) {  //Se a estacao mudou, atualiza o LCD e o terminal
    mudouEstacaoMesmo();
    mudouEstacao = false;  // Mantem a estacao que esta tocando
  }
}
```
I imagine that the "Bluetooth" portion of the code does not interfere with the internet connection via WiFi, since the "Http" addresses (without the "s") work. What could be interfering with Https? Or is it the size of the code that leaves little memory for these addresses?

renatoianhez, 2025-01-01:
I removed all the "s" from the addresses. Some stations don't support this, like [Tonic Ska Radio](https://eu10.fastcast4u.com/tonicska). Others redirect to the Https: address.
```
info        connect to: "streaming.educadorafm.ba.gov.br" on port 80 path "/stream-01"
info        Connection has been established in 61 ms, free Heap: 35680 bytes
info        redirect to new host "https://streaming.educadorafm.ba.gov.br/stream-01"
info        buffers freed, free Heap: 35884 bytes
info        connect to: "streaming.educadorafm.ba.gov.br" on port 443 path "/stream-01"
info        Request https://streaming.educadorafm.ba.gov.br/stream-01 failed!
```

schreibfaul1, 2025-01-04:
I think it will be a memory problem, encrypted connections need up to ~50KBytes additional RAM. If you are using Arduino as a component you can tell "mbedtls" in menuconfig to use the PSRAM.

renatoianhez, 2025-01-05:
Oh, thanks! I've never been able to use Platformio, something in my attempts to install it and in my head doesn't work properly with it. It's like they say here in Brazil: 'our saints didn't cross'. I'll try one more time!

schreibfaul1, 2025-01-05:
I have updated my template. https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF
This should not be a problem,
Install Git and VSCode, then load the PlatformIO extension and then insert the link in "Clone repository".

renatoianhez, 2025-01-05:
I believe I solved it without needing Platformio (I ran into library issues again): there seems to be a dispute for memory between WiFi and Bluetooth. Well, since my device has a 'radio mode' and a 'bluetooth mode', I freed the memory supposedly allocated for Bluetooth when in radio (WiFi) mode, and vice versa:

```
void setup() {
  pinMode(canalPraCima, INPUT_PULLUP);  // Buttons mode
  pinMode(canalPraBaixo, INPUT_PULLUP);
  pinMode(modoAparelho, INPUT);
  Serial.begin(115200);  
  tela.init();            // display
  tela.setRotation(1);   
  tela.invertDisplay(0); 

  if (digitalRead(modoAparelho)) { // If in radio mode
    esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);  // memory allocated for Bluetooth
    radioTocando = random(0, 30);  
    mudouEstacao = true;
    WiFi.disconnect();  // 
    WiFi.mode(WIFI_STA);
    outrosWiFi.addAP("SSID", "passw"); 
    //outrosWiFi.addAP("xxx", "xxx");
    //outrosWiFi.addAP("xxx","xxx");
    while (outrosWiFi.run() != WL_CONNECTED) delay(1500);  
    audio.setPinout(I2S_BCLK, I2S_LRCK, I2S_DOUT);     
    audio.setVolume(16);                                  
    audio.setConnectionTimeout(3000, 7200);          
    audio.setTone(0, 6, 4);                                
    // audio.setBufsize(0, 4000000);
    // audio.forceMono(true); // Força o mono
    // audio.setBalance(0); // Balanco estereo direito - esquerdo (-16 a 16)
    baixaDados(endTudo, 120, tudo);        
    for (int j = 0; j < 120; j = j + 4) { 
      nomeRadio[indice] = tudo[j];
      tipoRadio[indice] = tudo[(j + 1)];
      cidadeRadio[indice] = tudo[(j + 2)];
      urlRadio[indice] = tudo[(j + 3)];
      indice++;
    }
  } else {  //If in Bluetooth mode
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    i2s.setPins(I2S_BCLK, I2S_LRCK, I2S_DOUT);
    if (!i2s.begin(I2S_MODE_STD, 44100, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO, I2S_STD_SLOT_BOTH)) {
      Serial.println("Failed to initialize I2S!");
      while (1)
        ;  // do nothing
    }
    a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
    a2dp_sink.set_auto_reconnect(true);
    a2dp_sink.start("BTRadio");
  }
}
```

renatoianhez, 2025-01-05:
I ran some tests using a function from the ESP32 API to measure the amount of SRAM released when doing esp_bt_controller_mem_release(ESP_BT_MODE_BTDM), and, on average, it released 36KB. It seems to be enough for HTTPS streams to work properly.

schreibfaul1, 2025-01-05:
Sounds good, good luck with your project!