schreibfaul1/ESP32-audioI2S #941: BBC radio stream play sound of low quality

mifarelight, 2025-01-02:
I recently tried to play some BBC link (provided by you in #645), for example: [BBC Radio 1](http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=96000.norewind.m3u8).

The sound quality is very good for about 1 second at the beginning of successful connection, but then it gets worse (there is always sound distortion). I checked the log and there is not any abnormal prompt.

I think the network problem (tried wired and wireless network connection) and the audio source problem (VLC/MPC plays well on PC) can be ruled out.

Is it the problem of the decoder? Or the hardware decoding capability is insufficient (I am using ESP32-S3)? Or is it just me who has this problem?

mifarelight, 2025-01-02:
I manually downloaded the linked m3u8 content.
```
#EXTM3U
#EXT-X-VERSION:3
## Created with Unified Streaming Platform(version=1.8.4)
#EXT-X-MEDIA-SEQUENCE:271221806
#EXT-X-INDEPENDENT-SEGMENTS
#EXT-X-TARGETDURATION:6
#USP-X-TIMESTAMP-MAP:MPEGTS=7210089888,LOCAL=2025-01-02T12:05:52Z
#EXT-X-PROGRAM-DATE-TIME:2025-01-02T12:05:52Z
#EXTINF:6.4, no desc
bbc_radio_one-audio=96000-271221806.ts
#EXTINF:6.4, no desc
bbc_radio_one-audio=96000-271221807.ts
#EXTINF:6.4, no desc
bbc_radio_one-audio=96000-271221808.ts
#EXTINF:6.4, no desc
bbc_radio_one-audio=96000-271221809.ts
#EXTINF:6.4, no desc
bbc_radio_one-audio=96000-271221810.ts
```
Each .ts file is 6.5 seconds play length. In my experience, ESP32-audioI2S plays audio streams with segments of more than 5 seconds better than those with segments of about 1 second. So far BBC these links are the exception.

mifarelight, 2025-01-03:
You are right, my ping time is close to 200mS, which seems to be unsolvable for the network I am using.

mifarelight, 2025-01-04:
Strangely, the [VoA online radio station](https://voa-ingest.akamaized.net/hls/live/2035200/161_352R/playlist.m3u8), which also has a ping time of 200mS, can be played smoothly.

It seems that there are other problems with BBC link playback.

monupurohit, 2025-01-12:
i can play stream smoothly with this code
sometime hits issue need to handle buffering

https://github.com/schreibfaul1/ESP32-audioI2S/issues/810