schreibfaul1/ESP32-audioI2S #945: AAC decoder issue (since 3.0.12)

Pako2, 2025-01-12:
My favorite stations work with aac stream. Very often, when I switch to such a station, it doesn't connect to the stream properly and I can't hear the sound. Then I have to repeat the station selection (sometimes several times) until it's OK.
This is what the log looks like when the problem occurs:
```[2025-01-12 11:29:57.274] I2S AUDIO_INFO        PSRAM found, inputBufferSize: 638965 bytes
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        buffers freed, free Heap: 176280 bytes
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        connect to: "rozhlas.stream" on port 80 path "/dvojka_high.aac"
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        Connection has been established in 56 ms, free Heap: 175268 bytes
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        AACDecoder has been initialized, free Heap: 174028 bytes , free stack 6204 DWORDs
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        stream ready
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        syncword found at pos 433
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        AAC decode error -32 : Bitstream value not allowed by specification
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        syncword found at pos 25
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.274] I2S AUDIO_INFO        AAC decode error -21 : Unexpected channel configuration change
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 608
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        AAC decode error -32 : Bitstream value not allowed by specification
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 9
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        AAC decode error -21 : Unexpected channel configuration change
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 9
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        AAC decode error -32 : Bitstream value not allowed by specification
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        AAC decode error -32 : Bitstream value not allowed by specification
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 480
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        AAC decode error -21 : Unexpected channel configuration change
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 9
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.291] I2S AUDIO_INFO        AAC decode error -21 : Unexpected channel configuration change
[2025-01-12 11:29:57.303] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.303] I2S AUDIO_INFO        AAC decode error -6 : Channel coupling not yet implemented
[2025-01-12 11:29:57.303] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.303] I2S AUDIO_INFO        AAC decode error -32 : Bitstream value not allowed by specification
[2025-01-12 11:29:57.303] I2S AUDIO_INFO        syncword found at pos 441
[2025-01-12 11:29:57.303] I2S AUDIO_INFO        syncword found at pos 0
[2025-01-12 11:29:57.303] I2S AUDIO_INFO        AAC decode error -21 : Unexpected channel configuration change
```
Sometimes (and this is not a rare case) an exception even occurs and ESP crashes:
```11:37:31.835 > I2S AUDIO_INFO        AAC decode error -21 : Unexpected channel configuration change
11:37:31.843 > I2S AUDIO_INFO        syncword found at pos 9
11:37:31.846 > I2S AUDIO_INFO        syncword found at pos 0
11:37:31.851 > I2S AUDIO_INFO        AAC decode error -12 : Invalid number of channels
11:37:31.858 > I2S AUDIO_INFO        syncword found at pos 0
11:37:31.862 > I2S AUDIO_INFO        AAC decode error -32 : Bitstream value not allowed by specification
11:37:31.869 > I2S AUDIO_INFO        syncword found at pos 469
11:37:31.874 > I2S AUDIO_INFO        syncword found at pos 0
11:37:31.878 > I2S AUDIO_INFO        AAC decode error -21 : Unexpected channel configuration change
11:37:31.885 > I2S AUDIO_INFO        syncword found at pos 9
11:37:31.889 > I2S AUDIO_INFO        syncword found at pos 0
11:37:31.893 > Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
11:37:31.900 >
11:37:31.900 > Core  0 register dump:
11:37:31.902 > PC      : 0x400e7b91  PS      : 0x00060530  A0      : 0x800e8b59  A1      : 0x3ffc4f60
11:37:31.910 > A2      : 0x3f8a6ad8  A3      : 0x3f8ad676  A4      : 0x00000800  A5      : 0x00060023
11:37:31.918 > A6      : 0x3ffc8514  A7      : 0x3ffc509c  A8      : 0x00000028  A9      : 0x3f8ad686  
11:37:31.926 > A10     : 0x3f4223ac  A11     : 0x00000000  A12     : 0x00040000  A13     : 0x00000780
11:37:31.933 > A14     : 0x3f8ade06  A15     : 0x00000080  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
11:37:31.942 > EXCVADDR: 0x00040000  LBEG    : 0x4008b30c  LEND    : 0x4008b317  LCOUNT  : 0x00000000
11:37:31.950 > 
11:37:31.950 >
11:37:31.950 > Backtrace: 0x400e7b8e:0x3ffc4f60 0x400e8b56:0x3ffc4f80 0x400e96b2:0x3ffc4fa0 0x400e973a:0x3ffc4fc0 0x400eb076:0x3ffc4fe0 0x400eb106:0x3ffc5020 0x400eb215:0x3ffc5050 0x400eb462:0x3ffc5070 0x400eb60d:0x3ffc50f0 0x400e3e27:0x3ffc5110 0x400dd7ed:0x3ffc5140 0x400de023:0x3ffc51a0 0x400de1d5:0x3ffc51c0 0x4008ea52:0x3ffc51e0
11:37:40.523 >   #0  0x400e7b8e in window_grouping_info(NeAACDecStruct*, ic_stream*) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.cpp:6812
11:37:40.523 >   #1  0x400e8b56 in ics_info(NeAACDecStruct*, ic_stream*, _bitfile*, unsigned char) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.cpp:8194
11:37:40.523 >   #2  0x400e96b2 in side_info(NeAACDecStruct*, element*, _bitfile*, ic_stream*, unsigned char) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.cpp:8636
11:37:40.523 >   #3  0x400e973a in individual_channel_stream(NeAACDecStruct*, element*, _bitfile*, ic_stream*, unsigned char, short*) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.cpp:8691
11:37:40.523 >   #4  0x400eb076 in single_lfe_channel_element(NeAACDecStruct*, _bitfile*, unsigned char, unsigned char*) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.cpp:8060
11:37:40.523 >   #5  0x400eb106 in decode_sce_lfe(NeAACDecStruct*, NeAACDecFrameInfo*, _bitfile*, unsigned char) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.cpp:7829
11:37:40.523 >   #6  0x400eb215 in raw_data_block(NeAACDecStruct*, NeAACDecFrameInfo*, _bitfile*, program_config*, drc_info*) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/aac_decoder/libfaad/neaacdec.cpp:7922
[... 3 lines cut ...]
11:37:40.523 >   #10 0x400dd7ed in Audio::sendBytes(unsigned char*, unsigned int) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/Audio.cpp:4371
11:37:40.523 >   #11 0x400de023 in Audio::playAudioData() at .pio/libdeps/noota4mb/ESP32-audioI2S/src/Audio.cpp:3585 (discriminator 2)
11:37:40.523 >       (inlined by) Audio::playAudioData() at .pio/libdeps/noota4mb/ESP32-audioI2S/src/Audio.cpp:3555 (discriminator 2)
11:37:40.523 >   #12 0x400de1d5 in Audio::performAudioTask() at .pio/libdeps/noota4mb/ESP32-audioI2S/src/Audio.cpp:6217 
11:37:40.523 >       (inlined by) Audio::performAudioTask() at .pio/libdeps/noota4mb/ESP32-audioI2S/src/Audio.cpp:6210  
11:37:40.523 >       (inlined by) Audio::audioTask() at .pio/libdeps/noota4mb/ESP32-audioI2S/src/Audio.cpp:6205
11:37:40.523 >       (inlined by) Audio::taskWrapper(void*) at .pio/libdeps/noota4mb/ESP32-audioI2S/src/Audio.cpp:6199  
11:37:40.523 >   #13 0x4008ea52 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
11:37:40.523 >
11:37:40.524 > 
11:37:40.524 >
11:37:40.524 >
```
Interestingly, the problem only occurs since version 3.0.12. If I change the ESP32-AudioI2S library version to 3.0.8, the problem does not occur and the connection to the stream works completely reliably.
Radio station URLs:
http://rozhlas.stream/dvojka_high.aac
http://rozhlas.stream/radiozurnal_high.aac
http://rozhlas.stream/radiozurnal_sport_high.aac
http://rozhlas.stream/pohoda_high.aac
http://rozhlas.stream/radio_junior_high.aac
http://rozhlas.stream/radio_junior_pisnicky_high.aac
http://rozhlas.stream/jazz_high.aac
http://rozhlas.stream/vltava_high.aac
http://rozhlas.stream/liberec_high.aac
http://rozhlas.stream/radio_dab_praha_high.aac

schreibfaul1, 2025-01-14:
Hello @Pako2,
when the station http://rozhlas.stream/dvojka_high.aac is running, the error sometimes occurs. That is rare, maybe something is being sent outside the aac stream. I will save the raw data at the same time to be able to analyse the cause. The old aac decoder was probably more fault-tolerant.

schreibfaul1, 2025-01-14:
With many stations, the aac stream always starts with a syncword followed by an ADTS header. This is not the case here. The next syncword 0xFFFx is then searched for. As 0xFFF also occurs in the audio data, it does not have to be followed by an ADTS header. Then the decoder receives completely wrong information. I now determine the length of the aac frame and check whether a further syncword follows. I think this is more reliable.

Pako2, 2025-01-15:
Hello @schreibfaul1 !
That looks promising. I was sure you would know how to do it. I'll try it when I get home and let you know.

Pako2, 2025-01-15:
Hello @schreibfaul1  !
It works like a charm!
I've made 128 switchings from one stream to another and not a single hesitation! That wouldn't have been possible before.
Thanks for the quick fix.

closed 2025-01-15 by Pako2