schreibfaul1/ESP32-audioI2S #951: Starting to play MP3 caused a crash , please help me

DifficultyToCreatName, 2025-01-15:
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define SD_CS 5
#define SPI_MOSI 23
#define SPI_MISO 19
#define SPI_SCK 18
#define I2S_DOUT 25
#define I2S_BCLK 27
#define I2S_LRC 26

Audio audio;

void setup() {
  pinMode(SD_CS, OUTPUT);
  digitalWrite(SD_CS, HIGH);
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SPI.setFrequency(1000000);

  Serial.begin(115200);
  SD.begin(SD_CS);
  // WiFi.disconnect();
  // WiFi.mode(WIFI_STA);
  // WiFi.begin(ssid.c_str(), password.c_str());
  // while (WiFi.status() != WL_CONNECTED) delay(1500);
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);

  audio.setVolume(21);                                           // default 0...21
  audio.connecttoFS(SD, "/Alan Walker - Golden Gate 2016.mp3");  // SD
  //  audio.connecttoFS(SD_MMC, "/test.wav"); // SD_MMC
  //  audio.connecttoFS(SPIFFS, "/test.wav"); // SPIFFS

  //  audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de"); // Google TTS
}

void loop() {
  audio.loop();
  vTaskDelay(1);
}

// optional
void audio_info(const char *info) {
  Serial.print("info        ");
  Serial.println(info);
}
void audio_id3data(const char *info) {  //id3 metadata
  Serial.print("id3data     ");
  Serial.println(info);
}
void audio_eof_mp3(const char *info) {  //end of file
  Serial.print("eof_mp3     ");
  Serial.println(info);
}
void audio_showstation(const char *info) {
  Serial.print("station     ");
  Serial.println(info);
}
void audio_showstreamtitle(const char *info) {
  Serial.print("streamtitle ");
  Serial.println(info);
}
void audio_bitrate(const char *info) {
  Serial.print("bitrate     ");
  Serial.println(info);
}
void audio_commercial(const char *info) {  //duration in sec
  Serial.print("commercial  ");
  Serial.println(info);
}
void audio_icyurl(const char *info) {  //homepage
  Serial.print("icyurl      ");
  Serial.println(info);
}
void audio_lasthost(const char *info) {  //stream URL played
  Serial.print("lasthost    ");
  Serial.println(info);
}
void audio_eof_speech(const char *info) {
  Serial.print("eof_speech  ");
  Serial.println(info);
}

result:

13:59:19.869 -> info        buffers freed, free Heap: 183704 bytes
13:59:19.869 -> info        Reading file: "/Alan Walker - Golden Gate 2016.mp3"
13:59:19.869 -> info        MP3Decoder has been initialized, free Heap: 155772 bytes , free stack 6032 DWORDs
13:59:19.913 -> info        Content-Length: 6731007
13:59:19.946 -> info        ID3 framesSize: 204448
13:59:19.946 -> info        ID3 version: 2.3
13:59:19.946 -> info        ID3 normal frames
13:59:19.946 -> id3data     SettingsForEncoding: Lavf58.76.100
13:59:19.946 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
13:59:19.946 -> 
13:59:19.946 -> Core  0 register dump:
13:59:19.946 -> PC      : 0x40081454  PS      : 0x00060033  A0      : 0x80083d58  A1      : 0x3ffc08bc  
13:59:19.946 -> A2      : 0x3ffba79c  A3      : 0x00000000  A4      : 0x356c422b  A5      : 0x00060023  
13:59:19.980 -> A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x00100000  A9      : 0x3ffc08ac  
13:59:19.980 -> A10     : 0x00000001  A11     : 0x0000007a  A12     : 0xb33fffff  A13     : 0x3f4317a8  
13:59:19.980 -> A14     : 0x00000003  A15     : 0x00060823  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
13:59:19.980 -> EXCVADDR: 0x00100010  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
13:59:20.021 -> 
13:59:20.021 -> 
13:59:20.021 -> Backtrace: 0x40081451:0x3ffc08bc |<-CORRUPTED
13:59:20.021 -> 
13:59:20.021 -> 
13:59:20.021 -> 
13:59:20.021 -> 
13:59:20.021 -> ELF file SHA256: 25132685ec674d1c

DifficultyToCreatName, 2025-01-15:
https://github.com/DifficultyToCreatName/payDemo/blob/master/Alan%20Walker%20-%20Golden%20Gate%202016.mp3

DifficultyToCreatName, 2025-01-15:
Dear developer, while playing music, ESP32 keeps restarting and the music link is as shown above. Although using the official instance, it cannot play normally

DifficultyToCreatName, 2025-01-15:
and i use esp32-wroom-32

schreibfaul1, 2025-01-15:
and this link?
https://raw.githubusercontent.com/DifficultyToCreatName/payDemo/master/Alan%20Walker%20-%20Golden%20Gate%202016.mp3

DifficultyToCreatName, 2025-01-15:
> and this link? https://raw.githubusercontent.com/DifficultyToCreatName/payDemo/master/Alan%20Walker%20-%20Golden%20Gate%202016.mp3

yes,

Crazymanmusings, 2025-01-16:
I've been getting this error as well on a s3-n16r8

20:32:38.511 -> Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
20:32:38.576 -> 
20:32:38.576 -> Core  0 register dump:
20:32:38.576 -> PC      : 0x4202f910  PS      : 0x00060630  A0      : 0x8202fb4b  A1      : 0x3fc9f5b0  
20:32:38.576 -> A2      : 0x3c2204fc  A3      : 0x3c22202c  A4      : 0x00000000  A5      : 0x00000000  
20:32:38.576 -> A6      : 0x5a82799a  A7      : 0x007d0732  A8      : 0x00001018  A9      : 0x973c0b24  
20:32:38.576 -> A10     : 0xfd5c7a1a  A11     : 0x00001100  A12     : 0x00000006  A13     : 0x00266739  
20:32:38.610 -> A14     : 0x3c224144  A15     : 0x3c131754  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
20:32:38.610 -> EXCVADDR: 0x973c0b44  LBEG    : 0x4202f7aa  LEND    : 0x4202f808  LCOUNT  : 0x00000000  
20:32:38.610 -> 
20:32:38.610 -> 
20:32:38.610 -> Backtrace: 0x4202f90d:0x3fc9f5b0 0x4202fb48:0x3fc9f5e0 0x4202fe59:0x3fc9f610 0x4201009a:0x3fc9f650 0x4201097b:0x3fc9f6b0 0x42010b39:0x3fc9f6d0 0x4037f8aa:0x3fc9f6f0
20:32:38.610 -> 
20:32:38.610 -> 
20:32:38.610 -> 
20:32:38.610 -> 
20:32:38.610 -> ELF file SHA256: 77dad3a23

DifficultyToCreatName, 2025-01-17:
@schreibfaul1

Crazymanmusings, 2025-01-17:
I think I fixed my problem by putting the audio.loop in a task on core 1. Will let it run for a while to verify.

deltecent, 2025-01-21:
How did you do that? I am trying to stream over internet and my ESP32-S3 reboots after a few seconds.

If I add a `delay(100)` in the loop, it doesn't seem to crash.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376e49
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x109c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb50
load:0x403cc700,len:0x2fe4
entry 0x403c98ac
Setup
Connecting to WiFi
WiFi connected
info        PSRAM not found, inputBufferSize: 13951 bytes
info        buffers freed, free Heap: 198540 bytes
info        connect to: "audio-edge-w4d68.yul.o.radiomast.io" on port 443 path "/8a1ae985-ae96-4108-9b6c-331200e337a9"
info        SSL has been established in 454 ms, free Heap: 151756 bytes
Connect to host
info        MP3Decoder has been initialized, free Heap: 124304 bytes , free stack 4244 DWORDs
info        stream ready
info        syncword found at pos 0
info        MPEG-2.5, Layer I
info        Channels: 1
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 128000
info        StreamTitle=''
CORRUPT HEAP: Bad head at 0x3fcbb5dc. Expected 0xabba1234 got 0x0725b099

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)

Backtrace: 0x40376816:0x3fcebc10 0x4037ca09:0x3fcebc30 0x40382511:0x3fcebc50 0x403816ad:0x3fcebd80 0x403776fb:0x3fcebda0 0x40379099:0x3fcebdc0 0x42087d6e:0x3fcebde0 0x42087d9a:0x3fcebe00 0x42079bcc:0x3fcebe20 0x42031530:0x3fcebe40 0x420307f9:0x3fcebe60 0x42030a3d:0x3fcebe80 0x42012c8d:0x3fcebea0 0x42013d79:0x3fcebec0 0x42003142:0x3fcebee0 0x42035aa0:0x3fcebf00 0x4037f45a:0x3fcebf20

ELF file SHA256: e87159b2434760a7

Rebooting...
```
```
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"
#include "HardwareSerial.h"

#define I2S_DOUT            9
#define I2S_BCLK            3
#define I2S_LRC             1

Audio audio;
String ssid =    "ALPHA";
String password = "****";
void setup() {
  Serial.begin(115200);
  Serial.println("Setup");
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  Serial.println("Connecting to WiFi");
  WiFi.begin(ssid.c_str(), password.c_str());
  while (WiFi.status() != WL_CONNECTED)
    delay(1500);
  Serial.println("WiFi connected");
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(5);
  audio.connecttohost("https://audio-edge-w4d68.yul.o.radiomast.io/8a1ae985-ae96-4108-9b6c-331200e337a9");
  Serial.println("Connect to host");

  Serial.print("setup() running on core ");
  Serial.println(xPortGetCoreID());
}

void loop()
{
  audio.loop();
// delay(100);
}

void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
```

Crazymanmusings, 2025-01-21:
> How did you do that? I am trying to stream over internet and my ESP32-S3 reboots after a few seconds.
> 
> If I add a `delay(100)` in the loop, it doesn't seem to crash.

I'm just fumbling around in the dark. All this is kind of new to me. 
I'm not sure exactly which S3 you have but I have found that how it is set up makes a big difference.
I would stay off pin 3 if I had other pins open. I try to avoid pins 0, 3, 35, 36, 37, 45, and 46.
I found that even flash mode made a difference. below is how i set up the S3 that seems to work well.

[image]

closed 2025-01-22 by DifficultyToCreatName