schreibfaul1/ESP32-audioI2S #953: White noise

FatherMarco1971, 2025-01-16:
Using release 3.1.0 there is a strong white noise that start as soon Audio.init is called.
Everithing work fine using 3.0.12 so ithink is a sort of bug introduced after 3.012 release.

Anybody with similar problem?

FatherMarco1971, 2025-01-16:
Found the cause.
The problem is due to the line:

`m_i2s_std_cfg.clk_cfg.mclk_multiple  = I2S_MCLK_MULTIPLE_128;`

That was modified in:

`m_i2s_std_cfg.clk_cfg.mclk_multiple  = I2S_MCLK_MULTIPLE_512; `

I'm usning a devkit a1 swith ES3288.

Have you any idea why this cause the issue and how to solve without branche the library?

schreibfaul1, 2025-01-16:
Hello FatherMarco,
I can't say anything about the ES3288, I assume that it has an MCLK pin. Most DACs get the clock from BCLK and it doesn't matter what value _i2s_std_cfg.clk_cfg.mclk_multiple_ has. I think the value has been 512 for a long time, now it is I2S_MCLK_MULTIPLE_128