schreibfaul1/ESP32-audioI2S #956: After playing the internet radio for a while, there was no sound left

1359450511, 2025-01-23:
After listening to the network radio normally for a period of time, there is no sound anymore. The console will print the following information

[750006][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1399
[750014][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[750266][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[750273][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[750349][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[750356][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[750780][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[750787][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[750796][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1401
[750805][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[750860][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[750867][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[751393][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[751401][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[751410][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[751417][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[751426][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[751433][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[751594][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[751601][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[751770][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[751777][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[752007][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1399
[752015][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[752264][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[752271][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[752279][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[752286][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753438][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[753445][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753453][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[753460][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753469][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1401
[753479][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753488][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[753495][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753502][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[753509][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753540][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[753547][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753582][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[753589][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753681][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[753688][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[753867][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 1400
[753874][I][Audio.cpp:7639] chunkedDataTransfer(): chunksize 0
[754053][I][Audio.cpp:7639] chunkedDataTrans

Marcin-Fenger, 2025-01-23:
stream address, lib version, processor, framework version?

schreibfaul1, 2025-01-24:
It's possible that there is still an error. "chunkedDataTransfer" only have a few radio streams" Please paste the URL here.

1359450511, 2025-01-26:
https://lhttp.qtfm.cn/live/2803/64k.mp3