schreibfaul1/ESP32-audioI2S #957: CORRUPT HEAP

Negyszersult, 2025-01-25:
CORRUPT HEAP: Bad head at 0x3ffe854c. Expected 0xabba1234 got 0x57b78b16
assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)

I want to make an internet radio, but I get this error every minute and my ESP32 keeps rebooting. 

https://pastebin.com/e103yWDs - here is my code.

I'm using and ESP-Wroom-32 without PSRAM and the partition scheme is set to "Huge APP".
Any suggestions?

Thanks.

nstepanets, 2025-10-07:
I encountered a similar problem when switching stations (ESP32-vs1053_ext library).

e.g. 
http://stream.revma.ihrhls.com/zc6834/hls.m3u8
then
http://as-hls-ww-live.akamaized.net/pool_01505109/live/ww/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio%3d96000.norewind.m3u8

It seems the problem occurred after adding the m_lastM3U8host variable (it is not cleared when changing stations).

Negyszersult, 2025-10-07:
I didn't solved yet, if you will find any solution please notify me.

nstepanets, 2025-10-07:
To test, I tried adding a line to setDefaults() and it seems to have fixed the problem:
if(m_lastM3U8host){free(m_lastM3U8host); m_lastM3U8host = NULL;}

But in this case the problem may be something else.