schreibfaul1/ESP32-audioI2S #961: lost connection of internet streams

aber68, 2025-01-29:
hello, 

after playing internet radio for some time i got following result ...

[image]

ESP32 is still on an running, but no reconnection happens. is there a possibility to catch this state and do something like a restart?

best regards
andreas

mifarelight, 2025-01-31:
Depending on the existing code framework, here is my suggestion.

Step 1: declaring a weak function in the header file "Audio.h" like:
```c
extern __attribute__((weak)) void audio_host_unvalid(const char*);
```

Step 2: Modify the code block that prompts "Hostaddress is not valid" in "Audio.cpp" (line 479) to:
```c
    if(!startsWith(h_host, "http")) {
      AUDIO_INFO("Hostaddress is not valid");
      
      if(audio_host_unvalid) audio_host_unvalid("Reconnection is needed");
      
      stopSong();
      goto exit;
    }
```

Step 3: In the main loop file (e.g. main.cpp) define the audio_host_unvalid function, in which you change the flag "audio_connected":
```c
void audio_host_unvalid(const char *info){
  Serial.print(info);

  audio_connected = false;
}
```

Step 4: The last thing to do is to check the "audio_connected" flag in the main loop and reconnect if it is false.
```c
      if (!audio_connected)
      {
        station_connected = audio.connecttohost(yourStationLink);
      }
```

mifarelight, 2025-01-31:
You can try another simple way. Check for the running flag ("m_f_running") by calling "isRunning()" function.
Because "stopSong()" function clear the "m_f_running" flag.

aber68, 2025-01-31:
thx, i will give both a try ...

aber68, 2025-02-05:
change of audio.setConnectionTimeout(4000, 8000) helped a little, so the "Hostaddress is not valid" hasn't occurred since then, but in regulary time i got following (taking around 1 sec):

info		StreamTitle='Kelly Clarkson - Because of you'
**info		slow stream, dropouts are possible
info		slow stream, dropouts are possible
info		slow stream, dropouts are possible
info		slow stream, dropouts are possible
info		Stream lost -> try new connection
info		buffers freed, free Heap: 105760 bytes
info		connect to: "ors-sn08.ors-shoutcast.at" on port 80 path "/wie-q1a"
info		Connection has been established in 33 ms, free Heap: 106508 bytes
info		MP3Decoder has been initialized, free Heap: 75164 bytes , free stack 3092 DWORDs
info		stream ready
info		syncword found at pos 0
info		MPEG-2.5, Layer I
info		Channels: 2
info		SampleRate: 48000
info		BitsPerSample: 16
info		BitRate: 128000**
info		StreamTitle='Kelly Clarkson - Because of you'

PS: i use AI-Thinker ESP32 CAM having PSRAM

closed 2026-04-02 by github-actions[bot]