schreibfaul1/ESP32-audioI2S #963: mpd  dash protocol

lidense, 2025-02-03:
Hi, many stations like BBC are turning to more recent streaming protocols (dash) that require apps.
In particular this:
http://a.files.bbci.co.uk/ms6/live/3441A116-B12E-4D2F-ACA8-C1984642FA4B/audio/simulcast/dash/nonuk/pc_hd_abr_v2/cf/bbc_radio_scotland_fm.mpd
which doesn't seem to work with ESP32-audioI2S, any ideas?
Thanks

schreibfaul1, 2025-02-04:
Dash is rarely used in the audio sector; it is often used for video streaming. The question is whether it is worth the effort.

lidense, 2025-02-05:
The link above is for audio (radio Scotland).

lidense, 2025-02-05:
..apparently it is more and more used by BBC to limit listening through their BBC Sounds app.

schreibfaul1, 2025-02-05:
When I call up the BBC Scotland stream I get this:
````
<?xml version="1.0" encoding="utf-8"?>
<MPD xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns="urn:mpeg:dash:schema:mpd:2011" xmlns:dvb="urn:dvb:dash:dash-extensions:2014-1"
  xsi:schemaLocation="urn:mpeg:dash:schema:mpd:2011 http://standards.iso.org/ittf/PubliclyAvailableStandards/MPEG-DASH_schema_files/DASH-MPD.xsd"
  type="dynamic" availabilityStartTime="1969-12-31T23:59:44Z"
  minimumUpdatePeriod="PT6H" timeShiftBufferDepth="PT6H" maxSegmentDuration="PT7S" minBufferTime="PT3.200S"
  profiles="urn:dvb:dash:profile:dvb-dash:2014,urn:dvb:dash:profile:dvb-dash:isoff-ext-live:2014"
  publishTime="2025-01-22T12:36:41">

    <UTCTiming schemeIdUri="urn:mpeg:dash:utc:http-iso:2014" value="http://time.akamai.com/?iso" />

    <BaseURL dvb:priority="1" dvb:weight="1" serviceLocation="cf">http://as-dash-ww.live.cf.md.bbci.co.uk/pool_43322914/live/ww/bbc_radio_scotland_fm/bbc_radio_scotland_fm.isml/dash/</BaseURL>

    <Period id="1" start="PT0S">
    <AdaptationSet group="1" contentType="audio" lang="en" minBandwidth="48000" maxBandwidth="96000"
                   segmentAlignment="true" audioSamplingRate="48000" mimeType="audio/mp4" codecs="mp4a.40.5" startWithSAP="1">
        <AudioChannelConfiguration schemeIdUri="urn:mpeg:dash:23003:3:audio_channel_configuration:2011" value="2"/>
        <Role schemeIdUri="urn:mpeg:dash:role:2011" value="main"/>
        <SegmentTemplate timescale="48000" initialization="bbc_radio_scotland_fm-$RepresentationID$.dash"
                       media="bbc_radio_scotland_fm-$RepresentationID$-$Number$.m4s" startNumber="1" duration="307200"/>
        <Representation id="audio=48000" bandwidth="48000"/>
        <Representation id="audio=96000" bandwidth="96000"/>
    </AdaptationSet>
  
  </Period>
</MPD>
````

and that is what is put together:
_http://as-dash-ww.live.cf.md.bbci.co.uk/pool_43322914/live/ww/bbc_radio_scotland_fm/bbc_radio_scotland_fm.isml/dash/bbc_radio_scotland_fm-audio=96000.dash_

confusing

lidense, 2025-02-05:
If you do:
mplayer http://a.files.bbci.co.uk/ms6/live/3441A116-B12E-4D2F-ACA8-C1984642FA4B/audio/simulcast/dash/nonuk/pc_hd_abr_v2/cf/bbc_radio_scotland_fm.mpd
or 
vlc http://a.files.bbci.co.uk/ms6/live/3441A116-B12E-4D2F-ACA8-C1984642FA4B/audio/simulcast/dash/nonuk/pc_hd_abr_v2/cf/bbc_radio_scotland_fm.mpd

You will hear the stream!

lidense, 2025-02-05:
..if you open the station in the bbc sounds app:

https://www.bbc.co.uk/sounds/play/live/bbc_radio_scotland_fm

and "inspect" the network calls you see that it fetches segments like this:

https://as-dash-ww.live.cf.md.bbci.co.uk/pool_43322914/live/ww/bbc_radio_scotland_fm/bbc_radio_scotland_fm.isml/dash/bbc_radio_scotland_fm-audio=96000-271686524.m4s

lidense, 2025-02-05:
[network inspection.pdf](https://github.com/user-attachments/files/18679993/network.inspection.pdf)

Galaxy-Man, 2026-09-02:
This works fo rme, I'm using audio library 3.4.0
volosR internet radio sketch.

// workflow for BBC DASH → HLS MPD files
// http://nossl.fmstream.org/index.php?s=bbc
// download mpd, cut an dpast content into copilot. Ask copilot 'Find the corresponding HLS stream URL'
// copy the 96000 url to the stations array