schreibfaul1/ESP32-audioI2S #964: 2 different errors after random time

martinius96, 2025-02-03:
Hello there,
i am totally new in audio stuff and first time I am using I2S bus. I am surprised how huge the app is with minimum code for webradio with ESP32-audioI2S library. I had to use 3 MB HUGE APP partition, as it wasnt able to fit into standard size with 1,2MB OTA partitions each. I hoped for better stability :( That will not be good for my application where I would like to use combination of SPIFFS .mp3 files to play or webradio. For the app it will be critical to not randomly restarts, as it should be used within the escape room, where people will be solving problems.

From my testing I see, mp3 streams are running, but OGG or AAC stream does not work. AAC streams will reset the board right after I use them. That is not a bit deal, but I am facking other errors at the moment, so let's go step by step. OGG streams are not played when I try, but it will not restart the board. So only .mp3 streams are running but will reset the board after random time from seconds to minutes.

Hope my description helps.

**Hardware:** 

- MCU: Wemos LoLin32 Lite ESP32 (standard WROOM-32)
- Amplifier: MAX98357 module
- Speaker: 2W 8Ω 20x30x6.8mm (2030 Cavity Speaker Type B)

**Software:**

- Arduino IDE: 1.8.19
- Arduino Core for ESP32: 3.1.0
- ESP32-audioI2S library: 3.0.12 (latest stable release)

**Wiring ESP32 to MAX98357:**

- 3V3 Vcc (often I see in online tutorials 5V, but based on datasheet MAX98357, voltage from 2,5V is fine)
- GND GND
- D25 DIN
- D26 LRC
- D27 BCLK
- Note: GAIN is not connected.

I am using basically the minimum code for web radio to which I have integrated volume and stream URL swithing via Serial interface input.

At the moment I am facing 3 errors. First error is related to MP3 decoding and looks like it is fully related to the library only. It occurs from seconds to X minutes. Other error is  "Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled" that also happened during stream was running. This happened also after the first error and when wifi connected, so it again resets and then was running for XY seconds or minutes before again there was that mp3 decoding problem.

Also if I change URL of the stream to .aac one (ex. http://audio.lumen.sk/aac20.aac), it will reset the ESP32 due the error Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled. SEE, **CORE 0 instead of CORE 1 in that case**

So basically the first 2 error happened when nothing else was called than audio.loop() in the void loop(){ }. So I think these problems can be related to the library only. **Program implementation at the bottom of this issue.**

**Error 1:**
```
assert failed: int32_t HybridTransform(int32_t*, int32_t*, int32_t (*)[32], SideInfoSub_t*, BlockCount_t*) C:\Users\Martin_PC\Documents\Arduino\libraries\ESP32-audioI2S-master\src\mp3_decoder\mp3_dec

Backtrace: 0x400824b1:0x3ffc4e00 0x4008c8c9:0x3ffc4e20 0x40092afe:0x3ffc4e40 0x400f06a5:0x3ffc4f70 0x400f08fb:0x3ffc5000 0x400f0fd8:0x3ffc5050 0x400dd322:0x3ffc5090 0x400ddc0b:0x3ffc50f0 0x400dddc1:0x3ffc5110 0x4008d572:0x3ffc5130

ELF file SHA256: 843bd3604

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8
```

**Error 2:**
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401ad279  PS      : 0x00060230  A0      : 0x80182dc0  A1      : 0x3ffb1ff0  
A2      : 0x3ffedfc0  A3      : 0x00060023  A4      : 0x00060020  A5      : 0x3f4029e8  
A6      : 0x00000000  A7      : 0x0000abab  A8      : 0x853cf409  A9      : 0x729fa562  
A10     : 0x3ffbe1f4  A11     : 0xffffffff  A12     : 0x830e3884  A13     : 0x00060023  
A14     : 0xb33fffff  A15     : 0x0000abab  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x853cf40d  LBEG    : 0x40089c41  LEND    : 0x40089c63  LCOUNT  : 0x00000000  

Backtrace: 0x401ad276:0x3ffb1ff0 0x40182dbd:0x3ffb2010 0x4013eacf:0x3ffb2030 0x401a70a1:0x3ffb2050 0x40144c3d:0x3ffb2070 0x40119900:0x3ffb2090 0x401157ae:0x3ffb20b0 0x40116288:0x3ffb20f0 0x40116352:0x3ffb2140 0x400d4436:0x3ffb2160 0x400d4526:0x3ffb2180 0x400d47a6:0x3ffb21a0 0x400e230f:0x3ffb21d0 0x400e32ad:0x3ffb21f0 0x400d2e8e:0x3ffb2210 0x40101ad0:0x3ffb2270 0x4008d572:0x3ffb2290

ELF file SHA256: 843bd3604

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8

```
**Error 3 (when I changed stream to .aac one):**
```
20:46:49.300 -> 192.168.1.54
20:51:54.875 -> Stream changed to: http://audio.lumen.sk/aac20.aac
20:51:57.049 -> Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
20:51:57.049 -> 
20:51:57.049 -> Core  0 register dump:
20:51:57.049 -> PC      : 0x400ea79c  PS      : 0x00060930  A0      : 0x800ea849  A1      : 0x3ffc4f30  
20:51:57.083 -> A2      : 0x00000000  A3      : 0x3ffc4fec  A4      : 0x3fff9a84  A5      : 0x3ffc4f7f  
20:51:57.083 -> A6      : 0x3fff2864  A7      : 0x00000000  A8      : 0x800ea799  A9      : 0x3ffc4f10  
20:51:57.083 -> A10     : 0x00000000  A11     : 0x00000004  A12     : 0xbaad5678  A13     : 0xbaad5678  
20:51:57.083 -> A14     : 0x00000004  A15     : 0x3ffc4fec  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
20:51:57.133 -> EXCVADDR: 0x00000004  LBEG    : 0x40089d44  LEND    : 0x40089d4f  LCOUNT  : 0x00000000  
20:51:57.133 -> 
20:51:57.133 -> 
20:51:57.133 -> Backtrace: 0x400ea799:0x3ffc4f30 0x400ea846:0x3ffc4f70 0x400ea955:0x3ffc4fa0 0x400eaba2:0x3ffc4fc0 0x400ead4d:0x3ffc5040 0x400e3553:0x3ffc5060 0x400dd43d:0x3ffc5090 0x400ddc37:0x3ffc50f0 0x400ddded:0x3ffc5110 0x4008d572:0x3ffc5130
20:51:57.133 -> 
20:51:57.133 -> 
20:51:57.133 -> 
20:51:57.133 -> 
20:51:57.133 -> ELF file SHA256: 76208dba3
20:51:57.133 -> 
20:51:57.269 -> Rebooting...
20:51:57.269 -> ets Jul 29 2019 12:21:46
20:51:57.316 -> 
20:51:57.316 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:51:57.316 -> configsip: 0, SPIWP:0xee
20:51:57.316 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:51:57.316 -> mode:DIO, clock div:1
20:51:57.316 -> load:0x3fff0030,len:4916
20:51:57.316 -> load:0x40078000,len:16436
20:51:57.316 -> load:0x40080400,len:4
20:51:57.316 -> ho 8 tail 4 room 4
20:51:57.316 -> load:0x40080404,len:3524
20:51:57.316 -> entry 0x400805b8
20:51:57.862 -> ..
20:51:58.882 -> WiFi connected!
20:51:58.882 -> 192.168.1.54

```

**Actual program implementation where I am getting these problems. There is used stream of Slovak national radio with 128 kbps mp3:**
```
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"

#define I2S_DOUT  25
#define I2S_BCLK  27
#define I2S_LRC   26

Audio audio;
String ssid = "YOUR SSID";
String password = "YOUR PASSWORD WIFI";
int volume = 5;

void setup() {
  Serial.begin(115200);
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid.c_str(), password.c_str());
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("\nWiFi connected!");
  Serial.println(WiFi.localIP());

  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(volume);
  audio.connecttohost("http://icecast.stv.livebox.sk/slovensko_128.mp3");
}

void loop() {
  audio.loop();
  if (Serial.available()) {
    String input = Serial.readStringUntil('\n');
    input.trim();

    if (input == "+") {
      if (volume < 21) {
        volume++;
        audio.setVolume(volume);
        Serial.println("Volume: " + String(volume));
      } else {
        Serial.println("Volume already at max (21)");
      }
    } else if (input == "-") {
      if (volume > 0) {
        volume--;
        audio.setVolume(volume);
        Serial.println("Volume: " + String(volume));
      } else {
        Serial.println("Volume already at min (0)");
      }
    } else if (input == "0") {
      volume = 0;
      audio.setVolume(volume);
      Serial.println("Hlasitosť: " + String(volume));
    } else if (input.startsWith("http")) {
      audio.stopSong();
      audio.connecttohost(input.c_str());
      Serial.println("Stream changed to: " + input);
    } else {
      Serial.println("Invalid input. Use + or - or 0 or URL address");
    }
  }
}
```

martinius96, 2025-02-03:
Man I think the problems start after you change the VOLUME. i changed mp3 stream and then after 3 and half minutes I was playing with volume control and corrupt heap.

21:06:25.223 -> Stream changed to: https://playerservices.streamtheworld.com/api/livestream-redirect/AFNP_SBO.mp3
21:09:59.344 -> Volume: 6
21:09:59.892 -> Volume: 7
21:10:00.363 -> Volume: 8
21:10:00.812 -> Volume: 9
21:10:02.325 -> Volume: 8
21:10:02.681 -> Volume: 7
21:10:02.948 -> Volume: 6
21:10:31.861 -> CORRUPT HEAP: Bad head at 0x3ffe854c. Expected 0xabba1234 got 0xe8ffffdf
21:10:31.861 -> 
21:10:31.861 -> assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)

schreibfaul1, 2025-02-03:
This looks like a lack of memory. I suspect that your board has no PSRAM. Then your programme has to share the little RAM memory with the library . If PSRAM is recognised, all memory, except that of the decoders, is moved there. Errors such as "CORRUPT HEAP" or "Store Probibited" then also disappear

mifarelight, 2025-02-04:
If you use a ESP32 module without PSRAM, the lack of memory will always pop up at some point to interrupt you.

I have started using a module with 2M PSRAM earlier, but in the default espressif32 arduino library, mbedtls is pre-compiled to use internal RAM. Under this condition, when I try to have multiple tasks with SSL communication requirements, the allocation memory error prompt will pop up and tell me that I can't continue.

I was planning to compile the ESP32 arduino library myself, but after two days of trying, I am now planning to switch to the ESP-IDF platform.

martinius96, 2025-03-09:
U have switched to ESP32-CAM at the moment that has PSRAM onboard.
I was worried, if it will work, as this module does not have hardware I2S pins there. I have defined my own and it is running.

closed 2025-03-09 by martinius96