schreibfaul1/ESP32-audioI2S #966: reading from fatfs

hammerheaddown, 2025-02-14:
I have looked all over and i cant seem to find a way to use this instead of SD or spiffs

im using a  ESP32-S3-DevKit N16R8 Development Board

i have it using spiffs now but when i listen back to it  it just goes super slow and robotic...  but i know it sounds fine cause i send it to deepgram.com api and it can do the speech to text. 

also i swap and record to sd and it sounds fine too.. 

im hoping the fatfs will be better,  but  i have read it is better for data than spiffs,  but i am a noobie lol

schreibfaul1, 2025-02-14:
Hello, this library does not recognise SPIFFS, SD or FFAT. These are objects that are transferred.
I think a special partition table is required for FFAT.
8 bit is only available for WAV. Here is an example, 10s and 440Hz:  

[8Khz_8bit_2ch.wav.txt](https://github.com/user-attachments/files/18797011/8Khz_8bit_2ch.wav.txt)

hammerheaddown, 2025-02-14:
for now i am using this:

```
void playrecording() { // Modified to play only speech2txt.wav from SPIFFS

    Serial.println("Free Heap before listen to recording: " + String(ESP.getFreeHeap()));
    delay(10);

    String filename = "speech2txt.wav"; // Hardcoded filename for SPIFFS
    String fullPath = "/" + filename; // SPIFFS path is just filename in root

    char charArray[fullPath.length() + 1]; // Create a char array
    fullPath.toCharArray(charArray, fullPath.length() + 1); // Convert String to char array

    **audio.connecttoFS(SPIFFS, charArray); // Use SPIFFS and the char array**
    while (audio.isRunning()) {
        audio.loop();
    }
}
```

and it reads from spiffs  to play the audio file

so maybe i worded the question wrong?   in my partition  i want to use the  16mb   3mb/9.9 FATFS

so i was going to try and store the   audio file on the  9.9mb available on FATFS

 **audio.connecttoFS(SPIFFS, charArray); // Use SPIFFS and the char array**

this is where i am wondering how to play it if i use FATFS

hope that makes more sense,   silly how to us it makes sense  till someone else reads it lol

closed 2025-03-31 by github-actions[bot]