schreibfaul1/ESP32-audioI2S #967: On esp32duino v3.1.2,error

ChinaBurn, 2025-02-14:
My coed：
`#define I2S_DOUT 13
#define I2S_BCLK 12
#define I2S_LRC 11
Audio AudioPlayer(I2S_NUM_1);

void InitAudio() {
  AudioPlayer.setVolumeSteps(255);
  AudioPlayer.setVolume(255);
  AudioPlayer.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
}

`

error：E (358) gdma: gdma_register_tx_event_callbacks(464): user contex

jsc69, 2025-02-17:
Same here. It appears after an update of arduino-esp32 v3.1.2 from Expressif.
With the newest version 3.1.3 error still occurs.

Kay-Seljeseth, 2025-03-14:
Same error seen. Do use ESP32 S3 Wroom2 32M FLASH/8M PSRAM. Once enabling PSRAM "**OPI PSRAM**" and Flash Mode "**DIO 80MHz**" the same error occurs. Works ok for non-PSRAM playback, but also seeing issue below with the MP3Decoder. Works in general for single playbacks, so pins and cables are okay I guess.

Tested the PSRAM with test program and it works fine for simple data storage.

// I2S (external DAC) pins
#define I2S_DOUT      5  // DIN connection
#define I2S_BCLK       2  // Bit clock
#define I2S_LRC       42 // Left Right Clock
Audio audio;

### PSRAM issue:
E (421) gdma: gdma_register_tx_event_callbacks(464): user contexE (1566) i2s_common: i2s_channel_disable(1116): the channel has not been enabled yet
E (1566) i2s_std: i2s_channel_reconfig_std_gpio(357): Invalid state, I2S should be disabled before reconfiguring the gpio
E (1573) i2s_common: i2s_channel_enable(1090): the channel has already enabled or not initialized
info        PSRAM found, inputBufferSize: 638965 bytes

### Error playback (non PSRAM):
Happens after 2-4 songs played from SPI SD:
audio_info: buffers freed, free Heap: 37876 bytes
audio_info: Reading file: "/ONJ_Land_64.mp3"
**audio_info: The MP3Decoder could not be initialized**
audio_info: Closing audio file "ONJ_Land_64.mp3"

schreibfaul1, 2025-03-15:
Please do not use the Arduino versions 
3.1.2
3.1.3
3.2.0-RC1
if I2S is used together with PSRAM.

The bug has been fixed in 3.2.0-RC2.

Kay-Seljeseth, 2025-03-15:
Was able to use PSRAM after installing RC2 as mentioned.

10:47:50.180 -> info        PSRAM found, inputBufferSize: 638965 bytes
10:47:50.180 -> info        buffers freed, free Heap: 209060 bytes
10:47:50.180 -> info        Reading file: "/CCR32B_32.mp3"
10:47:50.180 -> info        MP3Decoder has been initialized, free Heap: 204540 bytes , free stack 2500 DWORDs
10:47:51.682 -> info        Content-Length: 1265663

Thank you!!!!!!

closed 2025-04-29 by github-actions[bot]

pie19988, 2026-02-11:
Hello, I am having this issue and I cannot find a fix. Has anyone found any fix for this?