schreibfaul1/ESP32-audioI2S #968: Some radio stream play sound of low quality

mifarelight, 2025-02-15:
Recently, the sound quality of some online radio stations I often listen to (e.g. https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio.m3u8) has deteriorated, and the sound effect is very similar to the previous reflection #941 of BBC radio (now using another live broadcast method). "The sound quality is very good for about 1 second at the beginning of successful connection, but then it gets worse (there is always sound distortion). I checked the log and there is not any abnormal prompt."

Today I tried to capture a section of I2S signal waveform with a logic analyzer, and I found that the output of the radio station with deteriorated sound quality has a blank groove of several milliseconds to more than ten milliseconds, and it is very frequent.

I don’t know what causes this. Is it because of insufficient decoding ability? The sound quality of these online radio stations has always been great, and so far, they still play well with VLC player.

The attached picture is a short clip of the I2S signal of a radio station with poor sound quality.The attached picture is a short clip of the I2S signal of a radio station with poor sound quality that I captured. The continuous signal is the BCLK signal, while the intermittent signal is the DOUT signal (the LRCK signal is not shown here, but I think it should not affect the description of the problem).

[image]

schreibfaul1, 2025-02-16:
The link https://radiopull.radiofoshan.com.cn/live/1400820947_BSID_46_audio.m3u8 does not work for me. Also not with the VLC player. In some cases WiFi is the bottleneck. You can try to increase the RX buffers in LWIP/TCP. If you have an ESP32-S3 you can increase PSRAM to 120MHz.

mifarelight, 2025-02-17:
1. The reason why the radio station link is invalid for you may be that some access policies of the website are restricted (like limit access out of China). This radio station can also be played through the browser page. The website link is: [radio_46](https://www.radiofoshan.com.cn/stationplayer.html?id=46)
2. Before reporting the problem, I have tried to connect via wired mode (ESP32-S3 + W5500), and the effect is the same as Wi-Fi, which should rule out the possibility that Wi-Fi causes the problem. However, this radio station can be played normally on my ESP32-S3 + PCM5102A hardware some time ago, whether via wired or wireless.
3. For lwIP buffer and PSRAM frequency setting, I need to check the relevant options of ESP32-S3 menuconfig before making further attempts.

schreibfaul1, 2025-02-17:
Yes, the m3u8 link seems to be blocked. I found an mp3 link on the website that works.
https://1500022859.vod-qcloud.com/43a6caeavodtranscq1500022859/3eb3acdf1397757905378556919/v.f1010.mp3?t=67bc1815&us=67b2dd95&sign=7ed122ac51adda10b3462a1536bc343c

It is possible that the m3u8 stream contains video content. In this case, the WiFi bandwidth would not be sufficient.

mifarelight, 2025-02-17:
The MP3 link you just provided works! I couldn't find any "vod_qcloud" information on that web site even though.

I tested the M3U8 link by using wired hardware (ESP32S3 + W5500) before. It has enogh bandwidth I think.

mifarelight, 2025-02-20:
I tried increasing the SPIRAM SPEED and now the playback is smoother.
```
...
CONFIG_SPIRAM_SPEED_80M=y
# CONFIG_SPIRAM_SPEED_40M is not set
...
```

closed 2025-02-20 by mifarelight