schreibfaul1/ESP32-audioI2S #970: Need support for outputting sound with PCM5102A and ESP32-S3.

dangtin306, 2025-02-16:
[image]

I need help getting sound output. I’m using the ESP32-S3-WROOM-1-N16R8 with Arduino IDE and Arduino-ESP32. I wired and programmed it exactly as shown in the tutorial, but the speaker produces no sound. When I remove the SCK pin, I only hear a faint hissing noise. I also tried switching to a 5V power supply and changing the default I2S pins:
#define I2S_DOUT 9
#define I2S_BCLK 3
#define I2S_LRC  1
but still nothing.

mifarelight, 2025-02-16:
First, check whether the hardware connection corresponds to the software definition.

From your picture, it seems that the GND connection is not seen. There are two blue wires and two yellow wires.

When using the breadboard test, it is recommended to use short connecting wires of DIFFERENT colors.

In fact, one end of the wire is led out from the ESP32 module, and the other end can be directly connected to PCM5102A.
To reduce the occurrence of connection errors and poor contact.

schreibfaul1, 2025-02-16:
sometimes the solder bridges are missing
https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/docs/PCM5102A.png

renatoianhez, 2025-03-05:
Hi! I have the same board. 
I connect a 10K resistor between VIN and SCK and XSMT. (in this picture I haven't soldered the XSMT yet, because I was testing it)

[image]

cniedzi, 2025-03-13:
> I need help getting sound output.
> 
> [image]
> 
> I’m using the ESP32-S3-WROOM-1-N16R8 connect PCM5102A with Arduino IDE and Arduino-ESP32. I wired and programmed it exactly as shown in the tutorial, but the speaker produces no sound. When I remove the SCK pin, I only hear a faint hissing noise. I also tried switching to a 5V power supply and modifying the default I2S pins #define I2S_DOUT 9 #define I2S_BCLK 3 #define I2S_LRC 1 I even updated the Arduino-ESP32 version from 3.0.7 to 3.1.11, but nothing worked. I also enabled verbose debugging and didn't see any error messages.

Can you show the bottom side of your PCM5102A?

schreibfaul1, 2025-03-14:
The bridges H1L ... H4L must be set. 
https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/docs/PCM5102A.png

dangtin306, 2025-03-15:
> > Tôi cần trợ giúp để có được âm thanh đầu ra.
> > [image]
> > Tôi đang sử dụng ESP32-S3-WROOM-1-N16R8 kết nối PCM5102A với Arduino IDE và Arduino-ESP32. Tôi đã nối dây và lập trình chính xác như trong hướng dẫn, nhưng loa không phát ra âm thanh. Khi tôi tháo chân SCK, tôi chỉ nghe thấy tiếng rít yếu ớt. Tôi cũng đã thử chuyển sang nguồn điện 5V và sửa đổi các chân I2S mặc định #define I2S_DOUT 9 #define I2S_BCLK 3 #define I2S_LRC 1 Tôi thậm chí đã cập nhật phiên bản Arduino-ESP32 từ 3.0.7 lên 3.1.11, nhưng không có gì hiệu quả. Tôi cũng đã bật gỡ lỗi chi tiết và không thấy bất kỳ thông báo lỗi nào.
> 
> Bạn có thể cho xem mặt dưới của PCM5102A không? Các jumper H1L - H4L phải được hàn đúng cách.

hi

dangtin306, 2025-03-15:
> Bạn có thể cho xem mặt dưới của PCM5102A không? Các jumper H1L - H4L phải được hàn đúng cách.

[image]
So I need to solder the H1L - H4L jumpers, right?

dangtin306, 2025-03-15:
> Can you show the bottom side of your PCM5102A? Jumpers H1L - H4L must be soldered appropriately.

[image]

So I need to solder the H1L - H4L jumpers, right?

schreibfaul1, 2025-03-15:
[image]

cniedzi, 2025-03-15:
> So I need to solder the H1L - H4L jumpers, right?

Yes, as in the Wolle's image.

Guerre450, 2025-04-01:
I've been having  the same issue and I've been following this thread. I've done the soldering jumper part, however the only thing I hear is static wether there is music playing or not. Are my pieces broken or am I doing something wrong? 

[image]

Guerre450, 2025-04-02:
Found the solution to my problem, turns out my  pins were acting as ground because of the power supply which was different than my esp32. Switched the Vin to the 3.3v on the esp32 and it now works no issues

closed 2025-05-17 by github-actions[bot]

hfzrfa, 2026-11-16:
> Hi! I have the same board. I connect a 10K resistor between VIN and SCK and XSMT. (in this picture I haven't soldered the XSMT yet, because I was testing it)
> 
> [image]

it's work thanks