schreibfaul1/ESP32-audioI2S #971: Is it possible to play an MP3 file which is represented as a byte array on RAM?

tanmanh0707, 2025-02-16:
Hello,

First of all, I would like to say thank you for your amazing library. I have a question:

As the title says, I have a small MP3 file, then I extract it to a byte array stored in a variable like
`const uint8_t mp3_data[] = {...};`

I would like to play that data something like
`audio.connecttoram(mp3_data);`

Is it possible?

schreibfaul1, 2025-02-16:
This will not work. Functions such as open(), position(), seek(), read() are required.
But you can set up an FFat partition and copy an mp3 file from the SD card to the FFat, for example.

tanmanh0707, 2025-02-18:
> This will not work. Functions such as open(), position(), seek(), read() are required. But you can set up an FFat partition and copy an mp3 file from the SD card to the FFat, for example.

Thank you very much for your response. There are also 2 other questions that I could not find a proper answer.
1) When I play an MP3 file from SD, it causes a loud pop at beginning, after that the playback is perfectly smooth. Is there a way prevent that noise such as fade in or something?
2) Even when I play a small mp3 file, duration is less than 500ms, it also takes more than 1s to hear the sound, causing a delay. Is it caused by the delay of SD reading?

closed 2025-04-04 by github-actions[bot]