schreibfaul1/ESP32-audioI2S #975: I2S not initialized after installing the latest version of the library

cniedzi, 2025-02-20:
After installing the latest version of the library, I get the following error just after `audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);` command:
  
```
E (11889) i2s_common: i2s_channel_disable(1116): the channel has not been enabled yet
E (11890) i2s_std: i2s_channel_reconfig_std_gpio(357): Invalid state, I2S should be disabled before reconfiguring the gpio
E (11897) i2s_common: i2s_channel_enable(1090): the channel has already enabled or not initialized
```

Should we add any command or config the latest version to work?

cniedzi, 2025-02-20:
False alarm :) The errors happens after upgrading ESP core to 3.1.3.

schreibfaul1, 2025-02-20:
Arduino 3.1.3 then probably uses a newer IDF which probably changes something in the I2S area. I'll have a look at this soon.

Zhentao-Lin, 2025-02-21:
I've also encountered this issue. It works fine for me in version 3.1.1, but not in versions 3.1.2 and 3.1.3.

cniedzi, 2025-02-21:
> I've also encountered this issue. It works fine for me in version 3.1.1, but not in versions 3.1.2 and 3.1.3.

confirm

Marcin-Fenger, 2025-02-21:
3.1.3  works fine as previous

cniedzi, 2025-02-21:
> 3.1.3 works fine as previous

what ESP?

Marcin-Fenger, 2025-02-21:
ESP32 WROVER n16r8

cniedzi, 2025-02-21:
My ESP32-S3-WROOM-1-N16R8 doesn't work.

Marcin-Fenger, 2025-02-21:
I use v3.1.0a so far, but it does not matter i think so

Zhentao-Lin, 2025-02-22:
> I use v3.1.0a so far, but it does not matter i think so

3.1.0 and 3.1.1 is based on 5.3.2, while 5.1.2 and 5.1.3 are based on 5.3. Why would you think they're not contributing to the problem? Did you compare their driver codes?
https://github.com/espressif/arduino-esp32/releases

schreibfaul1, 2025-02-22:
In the Arduino IDE I have the error neither in the Arduio version 3.1.2 nor in the version 3.1.3

If I enter the following in PlatformIO: platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip ; Arduino 3.1.3 / IDF 5.3.2

no DMA memory is made available for I2S. There is nothing I can do.
Set `framework = arduino, espidf` and configure it by yourself.

cniedzi, 2025-02-22:
Does it work for you in Arduino IDE 3.12 & 3.1.3?

> In the Arduino IDE I have the error neither in the Arduio version 3.1.2 nor in the version 3.1.3
>
> If I enter the following in PlatformIO: platform = [...]

cniedzi, 2025-02-22:
Could you share some Arduino IDE settings to try to make it working on my end?

cniedzi, 2025-02-22:
Changes regarding I2S in Arduino 3.1.2:

```
I2S
Fix(i2s): Check if pin is used before clearing bus by @P-R-O-C-H-Y in https://github.com/espressif/arduino-esp32/pull/10833
Fix(i2s): Add missing initializer for I2S CLK config by @me-no-dev in https://github.com/espressif/arduino-esp32/pull/10963
```

They might be the reason...

schreibfaul1, 2025-02-23:
Hello @cniedzi,
thank you for your research and your contribution.
Both pull requests
 https://github.com/espressif/arduino-esp32/pull/10833
 https://github.com/espressif/arduino-esp32/pull/10963
are most likely not the cause and do not solve our problem.

In PlatformIO, I2S from Arduino 3.1.2 does not receive DMA memory (I don't know the reason, I have no idea what happens internally), but you can do it yourself.
Please take the platformio.ini
https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/platformio.ini
 (from our radio application) and add the appropriate sdkconfig and it will work.

And I think it's a bug in PlatformIO or PioArduino? because I2S has nothing to do with Arduino, it's a question of implementation in the IDF.

schreibfaul1, 2025-02-23:
I have opened an issue
https://github.com/pioarduino/platform-espressif32/issues/120
Perhaps the experts know more :-))

cniedzi, 2025-02-23:
> Hello [@cniedzi](https://github.com/cniedzi), thank you for your research and your contribution. Both pull requests [espressif/arduino-esp32#10833](https://github.com/espressif/arduino-esp32/pull/10833) [espressif/arduino-esp32#10963](https://github.com/espressif/arduino-esp32/pull/10963) are most likely not the cause and do not solve our problem.
> 
> In PlatformIO, I2S from Arduino 3.1.2 does not receive DMA memory (I don't know the reason, I have no idea what happens internally), but you can do it yourself. Please take the platformio.ini https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/platformio.ini (from our radio application) and add the appropriate sdkconfig and it will work.
> 
> And I think it's a bug in PlatformIO or PioArduino? because I2S has nothing to do with Arduino, it's a question of implementation in the IDF.

Thanks @schreibfaul1. I've never used PlatformIO, but maybe it's good time to learn :)

sivar2311, 2025-02-23:
If this is working in the ArduinoIDE but not in PlatformIO some settings must be different.
Please show a screenshot of your ArduinoIDE settings as well as the used platformio.ini and the used board manifest.

sivar2311, 2025-02-23:
For me, i have the same results in ArduinoIDE and PlatformIO using an ESP32-S3-DevKitC1-N16R8:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
E (20) gdma: gdma_register_tx_event_callbacks(464): user context not in internal RAM
E (48) i2s_common: i2s_init_dma_intr(734): Register tx callback failed
E (48) i2s_std: i2s_channel_init_std_mode(236): initialize dma interrupt failed
```

But it seems to work when disabling the PSRAM!
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
```

I think the bug is somewhere related to the PSRAM, but not to PlatformIO or pioarduino (which are just the build system)

Therefore the error should be reported to Espressif.

cniedzi, 2025-02-23:
> For me, i have the same results in ArduinoIDE and PlatformIO using an ESP32-S3-DevKitC1-N16R8:
>
> ```
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT) [...]

I own the same board as yours. When I disable PSRAM, it still crashes...

sivar2311, 2025-02-23:
I don't see a crash here....

cniedzi, 2025-02-23:
> I don't see a crash here....

btw. what board do you choose in Arduino IDE?

sivar2311, 2025-02-23:
[image]

And here is my platformio.ini:
```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.extra_flags = 
  -DBOARD_HAS_PSRAM
```

These error messages disapear when turning off the PSRAM:
```
E (20) gdma: gdma_register_tx_event_callbacks(464): user context not in internal RAM
E (48) i2s_common: i2s_init_dma_intr(734): Register tx callback failed
E (48) i2s_std: i2s_channel_init_std_mode(236): initialize dma interrupt failed
```

cniedzi, 2025-02-23:
@sivar2311 you're right. it works without PSRAM...

```
ArduinoIDE
ESP32S3 Dev Module

Flash Size: 16mB (128Mb)
Partition scheme: Huge App (3MB No OTA/1MB SPIFFS)
PSRAM: Disabled

Works OK
**************************************************

Flash Size: 4mB (32Mb)
Partition scheme: Huge App (3MB No OTA/1MB SPIFFS)
PSRAM: OPI PSRAM

Doesn't work!
**************************************************
```

PS. Strange, today it isn't crushing without PSRAM (while yesterday it was)...

cniedzi, 2025-02-23:
https://github.com/espressif/arduino-esp32/issues/11004

copych, 2025-02-23:
I confirm the issue for ESP32-S3, Arduino IDE 2, core 3.1.2 and 3.1.3. PSRAM ON.

schreibfaul1, 2025-02-23:
My guess is that if PSRAM is found, the I2S buffers are allocated there, but DMA is not switched on.

Marcin-Fenger, 2025-02-24:
> 3.1.0 and 3.1.1 is based on 5.3.2, while 5.1.2 and 5.1.3 are based on 5.3. Why would you think they're not contributing to the problem? Did you compare their driver codes?

I have written about audio my current version of library:

*  Version 3.1.0a
 *  Updated on: Jan 16.2025
 *  Author: Wolle (schreibfaul1)
 
used in PlatformIO with:

platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip

And it is working now on ESP32 WROVER n16r8 . It is clear now?

sivar2311, 2025-02-24:
> Version 3.1.0a

Afaik there is no Version 3.1.0**a** 
There have been a few release candiates... RC1, RC2 etc...

`https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip` is not 3.1.0! 
This is 3.1.3 - See [this list](https://gist.github.com/sivar2311/5125011eaaf859eff90c4b502e1408b1)

> And it is working now on ESP32 WROVER n16r8 . 

What about the PSRAM in your project? (enabled / disabled) ?

>  Is it clear now?

No, see the points above.

Marcin-Fenger, 2025-02-24:
Please read carefully. Talk about audio.cpp version... I use very latest vesrion 3.1.3 IDF 5.3.2. Simply click and check it out https://github.com/pioarduino/platform-espressif32/releases/

sivar2311, 2025-02-24:
> Please read carefully

Sorry, my fault, 
I got confused by the versions - sorry.

What about your PSRAM settings? 
Is PSRAM enabled or disabled in your working project?

sivar2311, 2025-02-24:
> #define CONFIG_I2S_ISR_IRAM_SAFE 1

Sounds reasonable per [documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2s.html#iram-safe):

> Places driver object into DRAM (in case it is linked to PSRAM by accident).

@cniedzi & @copych 
So the solution for PSRAM enabled boards on PlatformIO based projects should be

```
build_flags = 
  -D CONFIG_I2S_ISR_IRAM_SAFE=1
```

sivar2311, 2025-02-24:
@cniedzi 

You might need to edit the files directly - See https://github.com/schreibfaul1/ESP32-audioI2S/issues/975#issuecomment-2677737489 since the ArduinoIDE does not have the hybrid compile feature.

schreibfaul1, 2025-02-25:
If Arduino is used as a component in PlatformIO, it works. Also with Arduino V3.1.3
https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF

cniedzi, 2025-03-04:
> If Arduino is used as a component in PlatformIO, it works. Also with Arduino V3.1.3 https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF

Confirm :)

`----------------------------------
ESP32 Chip: ESP32-S3
Arduino Version: 3.2.0
ESP-IDF Version: 5.4.0
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM SIZE 8388608
----------------------------------

audio_info: MP3Decoder has been initialized, free Heap: 189007 bytes , free stack 4408 DWORDs
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: MPEG-2.5, Layer I
audio_info: Channels: 2
audio_info: SampleRate: 44100
audio_info: BitsPerSample: 16
audio_info: BitRate: 256000
audio_info: StreamTitle='Fancy - Flames Of Love (Extended Version)'`

schreibfaul1, 2025-03-05:
Thanks for the feedback.
With the Arduino IDE it will probably be nothing. It is not clear to me with which values ​​Arduino is compiled. It's just that I2S DMA no longer works.

me-no-dev, 2025-03-07:
Also reported to ESP-IDF, because this is definitely a bug there. https://github.com/espressif/esp-idf/issues/15533

cniedzi, 2025-03-12:
Looks like it works @3.2.0 RC2