schreibfaul1/ESP32-audioI2S #979: Problema witha stream

FatherMarco1971, 2025-02-26:
Two very similar stram,on work one not..
This

http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8

Work fine for some times and then stop in a strange mode is like the audio loop start a loop see below.

```
INFO:PSRAM found, inputBufferSize: 638965 bytes
INFO:buffers freed, free Heap: 110500 bytes
INFO:connect to: "streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net" on port 80 path "/radiodeejay/radiodeejay/play1.m3u8"
INFO:Connection has been established in 72 ms, free Heap: 109852 bytes
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015160.ts"
INFO:AACDecoder has been initialized, free Heap: 107264 bytes , free stack 5404 DWORDs
INFO:stream ready
INFO:buffer filled in 2836 ms
INFO:syncword found at pos 0
INFO:Channels: 2
INFO:SampleRate: 48000
INFO:BitsPerSample: 16
INFO:BitRate: 131437
INFO:AAC HeaderFormat: ADTS
INFO:Spectral band replication: upsampled SBR
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015161.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015162.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015163.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015164.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015165.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015166.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015167.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015168.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/20250224T144458_p1s_000015169.ts"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
INFO:connect to: "http://streamcdnb1-4c4b867c89244861ac216426883d1ad0.msvdn.net/radiodeejay/radiodeejay/play1.m3u8"
```
This work like a charm:
http://streamcdnf25-4c4b867c89244861ac216426883d1ad0.msvdn.net/webradio/deejay80/live.m3u8

```
INFO:buffers freed, free Heap: 111784 bytes
INFO:connect to: "streamcdnf25-4c4b867c89244861ac216426883d1ad0.msvdn.net" on port 80 path "/webradio/deejay80/live.m3u8"
INFO:Connection has been established in 55 ms, free Heap: 111520 bytes
INFO:media sequence number: 17185613
INFO:connect to: "http://streamcdnf25-4c4b867c89244861ac216426883d1ad0.msvdn.net/webradio/deejay80/026_17185613.ts"
INFO:AACDecoder has been initialized, free Heap: 107188 bytes , free stack 5356 DWORDs
INFO:stream ready
INFO:buffer filled in 2936 ms
INFO:syncword found at pos 0
INFO:Channels: 2
INFO:SampleRate: 48000
INFO:BitsPerSample: 16
INFO:BitRate: 140250
INFO:AAC HeaderFormat: ADTS
```
this in the first stream:
```
INFO:Spectral band replication: upsampled SBR
```

could be the problem?

schreibfaul1, 2025-02-26:
Hello @FatherMarco1971,
SBR is a bit problematic. It looks like the buffer is running empty. If you are using an ESP32-S3 in VS Code with PlatformIO, try this template.
https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF
TCP-RX buffers are slightly larger and PSRAM runs at 120MHz. This means that the stream runs well for me.

P.S. I found another error in the first stream. Please reload the audio library.

FatherMarco1971, 2025-02-26:
_P.S. I found another error in the first stream. Please reload the audio library._
This seems not change the behavior, after some times thing goes wrong.

Tomorrow i'll see ESP32_Arduino_ESPIDF and i'ldo a tentative to use it.

Thanks for the support!

cniedzi, 2025-03-04:
I also have problems with playing AAC files. The sound is intermittent.

`audio_info: redirect to new host "https://ic2.smcdn.pl/2380-1.aac"
audio_info: buffers freed, free Heap: 190259 bytes
audio_info: connect to: "ic2.smcdn.pl" on port 443 path "/2380-1.aac"
audio_info: SSL has been established in 675 ms, free Heap: 162807 bytes
audio_info: chunked data transfer
audio_info: AACDecoder has been initialized, free Heap: 162415 bytes , free stack 3088 DWORDs
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: StreamTitle='Radio Eska - Hity Na Czasie'
audio_info: Channels: 2
audio_info: SampleRate: 48000
audio_info: BitsPerSample: 16
audio_info: BitRate: 59062
audio_info: AAC HeaderFormat: ADTS
audio_info: Spectral band replication: upsampled SBR`

My sdkkconfig.defaults file contains:
`CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_WS2812_LED_ENABLE is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_LWIP_IRAM_OPTIMIZATION=y
# CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC is not set
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
CONFIG_LWIP_MAX_ACTIVE_TCP=512
CONFIG_LWIP_MAX_LISTENING_TCP=512
CONFIG_LWIP_TCP_HIGH_SPEED_RETRANSMISSION=y
CONFIG_LWIP_TCP_MSS=1460
CONFIG_LWIP_TCP_TMR_INTERVAL=200
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=8192
CONFIG_LWIP_TCP_WND_DEFAULT=32768
CONFIG_LWIP_TCP_RECVMBOX_SIZE=32`

Shouls I add something?

cniedzi, 2025-03-05:
> It works with your SDKConfig ESP32S3 file, but there is only 6MB of PSRAM detected with `Serial.printf("PSRAM SIZE %d\n", ESP.getPsramSize());`

The following content of the file `sdkconfig.defaults` is necessary to correctly play AAC and detect a full size of the PSRAM:

```
CONFIG_AUTOSTART_ARDUINO=y
CONFIG_FREERTOS_HZ=1000

#Logs
CONFIG_BOOTLOADER_LOG_LEVEL_ERROR=y
CONFIG_LOG_DEFAULT_LEVEL_ERROR=y

CONFIG_IDF_EXPERIMENTAL_FEATURES=y

# Serial flasher config
CONFIG_ESPTOOLPY_FLASHFREQ_120M=y
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y

# Compiler options
CONFIG_COMPILER_OPTIMIZATION_PERF=y

# PSRAM
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_SPEED_120M=y

# ESP System Settings
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y

# TCP
CONFIG_LWIP_MAX_ACTIVE_TCP=1024
CONFIG_LWIP_MAX_LISTENING_TCP=1024
CONFIG_LWIP_TCP_HIGH_SPEED_RETRANSMISSION=y
CONFIG_LWIP_TCP_MAXRTX=12
CONFIG_LWIP_TCP_SYNMAXRTX=6
CONFIG_LWIP_TCP_MSS=1436
CONFIG_LWIP_TCP_TMR_INTERVAL=200
CONFIG_LWIP_TCP_MSL=60000
CONFIG_LWIP_TCP_FIN_WAIT_TIMEOUT=20000
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=8192
CONFIG_LWIP_TCP_WND_DEFAULT=32768
CONFIG_LWIP_TCP_RECVMBOX_SIZE=32
CONFIG_LWIP_TCP_ACCEPTMBOX_SIZE=6
CONFIG_LWIP_TCP_QUEUE_OOSEQ=y
CONFIG_LWIP_TCP_OOSEQ_TIMEOUT=4
CONFIG_LWIP_TCP_OOSEQ_MAX_PBUFS=12
CONFIG_LWIP_TCP_SACK_OUT=y
CONFIG_LWIP_TCP_OVERSIZE_QUARTER_MSS=y
CONFIG_LWIP_WND_SCALE=y
CONFIG_LWIP_TCP_RCV_SCALE=0
CONFIG_LWIP_TCP_RTO_TIME=1500
CONFIG_LWIP_IRAM_OPTIMIZATION=y

# mbedTLS
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
```

schreibfaul1, 2025-03-05:
I think the PSRAM size cannot be set in menuconfig, it is recognised automatically.
Although 6MB would be unusual.

FatherMarco1971, 2025-03-06:
I'm fghting to learn menuconfig and arduino as component usage.
I think to be close to do it and give a feedback.
Thanks @cniedzi for your feedback on SDKCONFIG.

cniedzi, 2025-03-06:
> I'm fghting to learn menuconfig and arduino as component usage. I think to be close to do it and give a feedback. Thanks [@cniedzi](https://github.com/cniedzi) for your feedback on SDKCONFIG.

I'm also learning Platformio :) What I've already found, it is prone to errors/instabilities due to menuconfig settings. I'm still getting many issues with AAC streams (ex. sync word not found, restars, etc.) in Platformio, while it works like a charm in Arduino IDE... (unfortunately they broke PSRAM staff in the latest ESP32 cores).
Btw. My outcomes on SDKCONFIG are the results of elimination unnecessary items, rather than conscious settings :)

schreibfaul1, 2025-03-12:
The Media Sequence Number was not found with more than two leading zeros. This should now work.

Example: ..../radiodeejay/20250224T144458_p1s_**000134911**.ts
is the real number 134911 (was previously 0)

Marco1971Repo, 2025-03-14:
[image]

Thanks @schreibfaul1 It seems to work!!
It still using Arduino precompiled!!
So probably the real problem was the zero filling.

schreibfaul1, 2025-03-14:
Hi @Marco1971Repo,
the cause was really the leading zeros.

P.S.
[IMG_20250314_184648.jpg (view on web)](https://github.com/user-attachments/assets/59313cfa-1772-4944-b769-3dd76dfbe0e2)
You have built a nice display layout.

Marco1971Repo, 2025-03-15:
Well i' don't know why my post is deleted anyway you read it.
Anyway i said that with the last fix related to zero leading it seems to work using arduino precompiled library.

And that i tried everything but compiling with the optimization bring my board to a total instability.

[LCDWebRadio](https://github.com/user-attachments/assets/59313cfa-1772-4944-b769-3dd76dfbe0e2)

Marco1971Repo, 2025-03-15:
Something is still working bad...
https://dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/playlist_audio.m3u8

This stream start well but after few moment:
'''
1:50:44.001 > INFO:ID3 version: 2.4
21:50:44.006 > INFO:ID3 normal frames
21:50:44.095 > INFO:Audio-Length: 67046
21:50:46.865 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/media-**ur748eh1d_b192000_227197786**.aac"
......
21:50:49.873 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/chunklist_b192000.m3u8"
21:50:49.923 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/media-**ur748eh1d_b192000_227197786**.aac"
'''
end then:
'''
21:50:49.923 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/media-ur748eh1d_b192000_227197786.aac"
....................
21:50:52.887 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/media-ur748eh1d_b192000_227197787.aac"
........
21:50:55.894 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/media-ur748eh1d_b192000_227197788.aac"
.................
21:50:58.897 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/media-ur748eh1d_b192000_227197789.aac"
.................
21:51:01.905 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/chunklist_b192000.m3u8"
21:51:01.959 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/chunklist_b192000.m3u8"
21:51:03.980 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/chunklist_b192000.m3u8"
21:51:04.039 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/chunklist_b192000.m3u8"
21:51:04.094 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/chunklist_b192000.m3u8"
21:51:04.154 > INFO:connect to: "https://StreamCdnG16-dd782ed59e2a4e86aabf6fc508674b59.msvdn.net/live/S97044836/tbbP8T1ZRPBL/chunklist_b192000.m3u8"
'''
Something is wrong in the management of m3u8 playlist.
I'm quite sure is there the problem but i do not find the root cause.

schreibfaul1, 2025-03-16:
The MediaSequenceNumber was not recognised correctly. Previously it was always in the line after #EXTINF: but here it is different, there is #MY-USER-CHUNK-DATA in between. 

[image]

Marco1971Repo, 2025-03-16:
@schreibfaul1 it works !!!!!!!!!!!!!!!!!
A big thank you!!!

closed 2025-03-17 by Marco1971Repo