schreibfaul1/ESP32-audioI2S #984: Choppy and draggy sound after last update.

hevet, 2025-03-07:
After this update https://github.com/schreibfaul1/ESP32-audioI2S/commit/210378a62adb223682f4c32ed5073597258036c5 some stations in aac have interrupted and dragged sound.

Examples of stations:

http://ic1.smcdn.pl/2030-1.aac
http://ic1.smcdn.pl/5990-1.aac

cniedzi, 2025-03-07:
See here https://github.com/schreibfaul1/ESP32-audioI2S/issues/979

hevet, 2025-03-07:
@cniedzi I saw this topic, but after the last change there is this problem, I uploaded the previous commit and they play fine. The problem is the last commit.

cniedzi, 2025-03-07:
> [@cniedzi](https://github.com/cniedzi) I saw this topic, but after the last change there is this problem, I uploaded the previous commit and they play fine. The problem is the last commit.

on Arduino IDE or Platformio?

hevet, 2025-03-07:
On the Platformio. I use Miniwebradio..

closed 2025-04-21 by github-actions[bot]