schreibfaul1/ESP32-audioI2S #990: Compilation after upgrade not possible (Audio.cpp/.h)

cdmaxy, 2025-03-16:
Dear Schreiblaul1,

After updating all libraries and packages, I'm getting errors in the Audio.cpp/.h library. Unfortunately, I haven't been able to resolve the issue in the last two days.

Use the example:
[https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/ESP32_ES8388/ESP32_ES8388.ino](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/ESP32_ES8388/ESP32_ES8388.ino)

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

platform_packages =

lib_deps =
https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9486
https://github.com/schreibfaul1/ESP32-audioI2S

lib_ignore =
ESP32syncTCP

The error is as follows:

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Modules
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency graph
|-- ES8388 @ 0.0.1+sha.3047786
|-- ESP32-audioI2S @ 3.0.13+sha.18fef51
|-- FS @ 2.0.0
|-- SD@2.0.0
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Building .pio\build\esp32dev\bootloader.bin
Generating partitions .pio\build\esp32dev\partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32dev\lib1cd\Wire\Wire.cpp.o
Compiling .pio\build\esp32dev\libe33\ES8388\ES8388.cpp.o
Compiling .pio\build\esp32dev\lib3f0\FS\FS.cpp.o
Compiling .pio\build\esp32dev\lib3f0\FS\vfs_api.cpp.o
Compiling .pio\build\esp32dev\lib762\FFat\FFat.cpp.o
Compiling .pio\build\esp32dev\lib078\SPI\SPI.cpp.o
Compiling .pio\build\esp32dev\lib601\SD\SD.cpp.o
Archiving .pio\build\esp32dev\lib1cd\libWire.a
Compiling .pio\build\esp32dev\lib601\SD\sd_diskio.cpp.o
Archiving .pio\build\esp32dev\libe33\libES8388.a
Compiling .pio\build\esp32dev\lib601\SD\sd_diskio_crc.c.o
Compiling .pio\build\esp32dev\lib2d7\SD_MMC\SD_MMC.cpp.o
Compiling .pio\build\esp32dev\lib84e\SPIFFS\SPIFFS.cpp.o
Archiving .pio\build\esp32dev\lib762\libFFat.a
Archiving .pio\build\esp32dev\lib3f0\libFS.a
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFi.cpp.o
Archiving .pio\build\esp32dev\lib078\libSPI.a
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFiMulti.cpp.o
Archiving .pio\build\esp32dev\lib601\libSD.a
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFiSTA.cpp.o
Archiving .pio\build\esp32dev\lib2d7\libSD_MMC.a
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFiScan.cpp.o
Archiving .pio\build\esp32dev\lib84e\libSPIFFS.a
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32dev\lib34f\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32dev\libd11\WiFiClientSecure\WiFiClientSecure.cpp.o
Compiling .pio\build\esp32dev\libd11\WiFiClientSecure\esp_crt_bundle.c.o
Compiling .pio\build\esp32dev\libd11\WiFiClientSecure\ssl_client.cpp.o
Compiling .pio\build\esp32dev\lib841\ESP32-audioI2S\Audio.cpp.o
Compiling .pio\build\esp32dev\lib841\ESP32-audioI2S\aac_decoder\aac_decoder.cpp.o
Compiling .pio\build\esp32dev\lib841\ESP32-audioI2S\aac_decoder\libfaad\neaacdec.cpp.o
Compiling .pio\build\esp32dev\lib841\ESP32-audioI2S\flac_decoder\flac_decoder.cpp.o
Compiling .pio\build\esp32dev\lib841\ESP32-audioI2S\mp3_decoder\mp3_decoder.cpp.o
Archiving .pio\build\esp32dev\lib34f\libWiFi.a
Compiling .pio\build\esp32dev\lib841\ESP32-audioI2S\opus_decoder\celt.cpp.o
Compiling .pio\build\esp32dev\lib841\ESP32-audioI2S\opus_decoder\opus_decoder.cpp.o
Archiving .pio\build\esp32dev\libd11\libWiFiClientSecure.a
Compiling .pio\build\esp32dev\lib841\ESP32-audioI2S\opus_decoder\silk.cpp.o
.pio/libdeps/esp32dev/ESP32-audioI2S/src/Audio.cpp: In constructor 'Audio::Audio(uint8_t)':
.pio/libdeps/esp32dev/ESP32-audioI2S/src/Audio.cpp:182:5: error: 'm_i2s_chan_cfg' was not declared in this scope
 m_i2s_chan_cfg.id = (i2s_port_t)m_i2s_num; // I2S_NUM_AUTO, I2S_NUM_0, I2S_NUM_1
 ^~~~~~~~~~~~~~
.pio/libdeps/esp32dev/ESP32-audioI2S/src/Audio.cpp:182:5: note: suggested alternative: 'm_i2s_config'
 m_i2s_chan_cfg.id = (i2s_port_t)m_i2s_num; // I2S_NUM_AUTO, I2S_NUM_0, I2S_NUM_1
 ^~~~~~~~~~~~~~
 m_i2s_config
.pio/libdeps/esp32dev/ESP32-audioI2S/src/Audio.cpp:183:36: error: 'I2S_ROLE_MASTER' was not declared in this scope
 m_i2s_chan_cfg.role = I2S_ROLE_MASTER; // I2S controller master role, bclk and lrc signal will be set to output

Thanks in advance
Regards
cdmaxy

schreibfaul1, 2025-03-16:
Please try this instead of :  `platform = espressif32`

`platform = https://github.com/pioarduino/platform-espressif32/archive/refs/tags/54.03.20-rc2.zip ; Arduino 3.2.0 IDF 5.4.0`

Alexandros444, 2025-04-29:
> Please try this instead of : `platform = espressif32`
> 
> `platform = https://github.com/pioarduino/platform-espressif32/archive/refs/tags/54.03.20-rc2.zip ; Arduino 3.2.0 IDF 5.4.0`

I was searching houres for a solution lol
(I got errors from the Audio.h in Platformio, something about the includes WiFiClientSecure and Spiffs not beeing found.)

Thank you SOO much!
(Why was this neccessary thoo?)

Kind Regards, Alex