schreibfaul1/ESP32-audioI2S #992: always reboot on ESP32

makerawei, 2025-03-17:
here is my setup code snippet  
```C++

#define I2S_DOUT  7
#define I2S_BCLK  8
#define I2S_LRC   9

  debugPrintln("\nwifi connect success");
  size_t freeHeap = ESP.getFreeHeap();
  Serial.printf("Free Heap: %d bytes\n", freeHeap);
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  debugPrintln("setPinout ok");
  audio.setVolume(21); // default 0...21
  debugPrintln("setVolume ok");
```
The program crashed when calling `audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);`  
Here is my output log  
```
17:19:03.329 -> wifi connect success
17:19:03.329 -> Free Heap: 252504 bytes
17:19:04.215 -> ets Jul 29 2019 12:21:46
17:19:04.215 -> 
17:19:04.215 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:19:04.215 -> configsip: 0, SPIWP:0xee
17:19:04.215 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:19:04.215 -> mode:DIO, clock div:1
17:19:04.215 -> load:0x3fff0030,len:1344
17:19:04.215 -> load:0x40078000,len:13964
17:19:04.215 -> load:0x40080400,len:3600
17:19:04.263 -> entry 0x400805f0
17:19:04.593 -> 
17:19:04.593 -> wifi connect success
17:19:04.593 -> Free Heap: 252504 bytes
17:19:05.481 -> ets Jul 29 2019 12:21:46
17:19:05.481 -> 
17:19:05.481 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:19:05.481 -> configsip: 0, SPIWP:0xee
17:19:05.481 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:19:05.481 -> mode:DIO, clock div:1
17:19:05.481 -> load:0x3fff0030,len:1344
17:19:05.481 -> load:0x40078000,len:13964
17:19:05.481 -> load:0x40080400,len:3600
17:19:05.528 -> entry 0x400805f0

```

schreibfaul1, 2025-03-17:
This means that this GPIO does not exist. If you are using an ESP32 try 17, 25, 26

makerawei, 2025-03-17:
> This means that this GPIO does not exist. If you are using an ESP32 try 17, 25, 26

I mixed up the models of two development boards, but the error disappeared after switching to other GPIOs. Thank you so much!

closed 2025-03-18 by makerawei