schreibfaul1/ESP32-audioI2S #996: Succesive mp3 files

lidense, 2025-03-19:
I would like to read successive mp3 short files but I have the impression that audio.connectFS cannot open successive files one after the other .
Please advise.

mifarelight, 2025-03-20:
1. Create an audio files link array;
2. Switch (audio.connectFS()) to another file in "audio_eof_mp3()" callback.

lidense, 2025-03-29:
I have done as you suggested (see below)

void audio_eof_mp3(const char* info) {
 
   log_w("%s and size of list is: %d", info, stringList.size());
   if(stringList.size()>0){
   audio.connecttoFS(SD,stringList.pop().c_str());
     }
   else
   audio.connecttoFS(SD, "/2024_12_31_15_48_43_clean4_mono.mp3");
}

but I stil get an error: 
"48243][E][Audio.cpp:4162] playSample(): Can't stuff any more in I2S..."

Please advise

mifarelight, 2025-03-29:
I suggest you to refer to another project by `Schreibfaul1`:
[ESP32-MiniWebRadio](https://github.com/schreibfaul1/ESP32-MiniWebRadio)

Focus on the usage of vector "_PLS_content" in it.

closed 2025-05-16 by github-actions[bot]