schreibfaul1/ESP32-audioI2S #999: MP3 file ends prematurely if setFilePos() has been called.

AntoineLep, 2025-03-21:
Hello,

I noticed a problem when playing mp3 tracks from my SD card.
When I simply play my file from the begining to the end, everything is OK

```
[381820][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 357 / 362
[382819][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 358 / 362
[383828][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 359 / 362
[384827][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 360 / 362
[385816][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 361 / 362
[386805][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 362 / 362
[387314][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Closing audio file "9782075089951_01.mp3"
[387325][I][006-audio-events.cpp:18] audio_eof_mp3(): [AUDIO:MP3 EOF] 9782075089951_01.mp3
[387334][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] End of file "9782075089951_01.mp3"
[387353][I][main.cpp:269] updateAudioInfo(): [MP3] Now Stopped
[387360][I][021-player-command-loop.cpp:193] playingMainStateJob(): [NAV] Play next
```

If I come back to this file and do a setFilePos() on it, It'll stop before the end.

```
[416263][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 353 / 362
[417262][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 354 / 362
[418251][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 355 / 362
[419270][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 356 / 362
[420269][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 357 / 362
[421258][I][main.cpp:260] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 358 / 362
[421387][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Closing audio file "9782075089951_01.mp3"
[421398][I][006-audio-events.cpp:18] audio_eof_mp3(): [AUDIO:MP3 EOF] 9782075089951_01.mp3
[421407][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] End of file "9782075089951_01.mp3"
[421427][I][main.cpp:269] updateAudioInfo(): [MP3] Now Stopped
[421434][I][021-player-command-loop.cpp:193] playingMainStateJob(): [NAV] Play next
[421442][I][main.cpp:357] playNextFile(): Going to the next file
```

After doing some research on several files, I found that the missing duration is proportional to the ID3 framesSize.
We can calculate it in advance

```
400265][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Content-Length: 11719766
[400274][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] ID3 framesSize: 139790
[400283][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] ID3 version: 2.3
[400301][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] ID3 normal frames
[400973][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Audio-Length: 11579976
```

The file being 11719766 long for 362 seconds
ID3 framesSize: 139790 represent almost 4 sec of data and the duraction of the missing part at the end is around 4 seconds as well

I've tried this with huge ID3 framesSize and I can have up to 45 missing seconds on the file after doing a setFilePos().
The calculation above is systematically verified.

So I think there is a problem where the setFilePos() somehow remove the consideration of the audio start offset based on the ID3 frame size.

Could you please have a look at it ?

Here are 2 example files:
- 45 sec cut at the end: https://storage.staytuned.io/media/audiobooks/7473c1f0-18f6-4a40-82f3-213bea9add55/ed790cc8-8327-4ad2-9d1d-8365c227ca86/3328140024906_02.mp3?context=audioSpeaker&appuserKey=60147fe2-d0d9-485a-b45e-d1eb4eea01e0
- 4 sec cut at the end: https://storage.staytuned.io/media/audiobooks/7473c1f0-18f6-4a40-82f3-213bea9add55/fe271dc2-4a14-4575-abed-ee9422be6b7f/9782075089951_01.mp3?context=audioSpeaker&appuserKey=60147fe2-d0d9-485a-b45e-d1eb4eea01e0

Hope you can help :)

Best regards

schreibfaul1, 2025-03-22:
Hi @AntoineLep ,
`setFilePos()` is only intended for debugging and sets the read pointer to the desired position.
Please try` setAudioPlayPosition() `to change the position within the audio file.

Best regards

AntoineLep, 2025-03-24:
He @schreibfaul1,

Thanks for having a look. At first, I was using `setAudioPlayPosition()` but the exact same problem occurs. The last seconds of the file were being cut off based on the size of the ID3 frames.

Furthermore, the behaviour is quite strange as `getCurrentTime()` returns completely incorrect values ​​for 10 to 15 seconds (rewinding at first, then stabilizing and finally going back upward).

As well, I can't rely on `setAudioPlayPosition()` since I'm not being placed at the correct position. In this example I run `audio.setAudioPlayPosition(1667)` and the audio time is set to approximately 1783.

I've set it back to test and the problem occurs again with `setAudioPlayPosition()`. See the logs below:

```
[ 76373][I][main.cpp:396] backward(): [NAV] Backward to track 3 at 1667 seconds
[ 78709][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Closing audio file "3328140024906_04.mp3"
[ 78722][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] buffers freed, free Heap: 67112 bytes
[ 78739][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Reading file: "/audiobooks/ed790cc8-8327-4ad2-9d1d-8365c227ca86/3328140024906_03.mp3"
[ 78762][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] MP3Decoder has been initialized, free Heap: 37948 bytes , free stack 1052 DWORDs
[ 79573][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Content-Length: 48210729
[ 79581][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] ID3 framesSize: 1096908
[ 79590][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] ID3 version: 2.4
[ 79608][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] ID3 normal frames
[ 80361][I][001-common.h:666] logMemoryConsumption(): [HARD] -- HEAP --  Free: 37432 / 237628 | Minimum free: 35588
[ 80372][I][001-common.h:667] logMemoryConsumption(): [HARD] -- PSRAM -- Free: 3368364 / 4178720
[ 81178][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Audio-Length: 47113821
[ 81197][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] stream ready
[ 81198][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] syncword found at pos 0
[ 81220][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] MPEG-2.5, Layer I
[ 81229][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] Channels: 2
[ 81236][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] SampleRate: 44100
[ 81245][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] BitsPerSample: 16
[ 81253][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] BitRate: 192000
[ 82311][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1 / 1963

--> setAudioPlayPosition(1667) is done here !!

[ 82391][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1667 / 1963
[ 83153][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 2296 / 1963
[ 83660][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1908 / 1963
[ 84183][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1876 / 1963
[ 84706][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1854 / 1963
[ 85228][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1839 / 1963
[ 85750][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1828 / 1963
[ 86273][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1820 / 1963
[ 86796][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1813 / 1963
[ 87318][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1808 / 1963
[ 87840][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1804 / 1963
[ 88363][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1801 / 1963
[ 88886][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1798 / 1963
[ 89408][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1796 / 1963
[ 89930][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1794 / 1963
[ 90457][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1792 / 1963
[ 90980][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1791 / 1963
[ 91493][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1789 / 1963
[ 92015][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1788 / 1963
[ 92537][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1787 / 1963
[ 93584][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1786 / 1963
[ 94107][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1785 / 1963
[ 95153][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1784 / 1963
[ 98283][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1783 / 1963
[100382][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1784 / 1963
[104035][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1785 / 1963
[106128][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1786 / 1963
[108210][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1787 / 1963
[109785][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1788 / 1963
.
.
.
.
[246141][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1915 / 1963
[247188][I][main.cpp:223] updateAudioInfo(): [MP3] Track 3 / 20 | Audio time: 1916 / 1963
[... cut ...]
[250185][I][006-audio-events.cpp:5] audio_info(): [AUDIO:INFO] End of file "3328140024906_03.mp3"
```

I looked at how `setAudioPlayPosition()` works and my approach was pretty similar except I rely on the current position to file size ratio after substracting the ID3 frames size to determine the duration. Which avoids doing calculations based on an average bitrate that doesn't seem to work in my case. But anyway the problem is more in the `setFilePos()` side than on `setAudioPlayPosition()`.

Any clue ?

AntoineLep, 2025-04-08:
Hello @schreibfaul1 ,

Have you had time to take a look at the issue ?

Best regards,

Antoine

AntoineLep, 2025-04-22:
Hello @schreibfaul1,
I switched to this rev: https://github.com/schreibfaul1/ESP32-audioI2S/commit/567abd22c07286ddab01134a8e7f1189f9701758 and I don't have the problem anymore.
So something must have changed in between this rev and the current one.

Hope it helps

AntoineLep, 2025-04-23:
Hello @schreibfaul1,

I looked through the intermediate commits and found that the problem occurs since this commit: https://github.com/schreibfaul1/ESP32-audioI2S/commit/3efb3d6c28c4bd63c5ca46c5b08e3211da92846e

Regards

AntoineLep, 2025-04-23:
Indeed that fix the cut at the end of the audio but now the audio eof is not triggered anymore.

AntoineLep, 2025-04-29:
Hi again @schreibfaul1,

I've added some logs for you to see what happens

The log format is: 
```
[MP3] currentFilePosition  (audio.getFilePos()  - audio.inBufferFilled()) / audio.getFileSize()
```

The audio.inBufferFilled() value never decreases below a given value at the end (which fluctuacte around 836 and 964 on the 3 files I tested this on)

```
[ 27965][I][main.cpp:278] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 361 / 362
[ 27984][I][main.cpp:224] updateAudioInfo(): [MP3] 11672118 (11719766 - 47648) / 11719766
[ 28003][I][main.cpp:224] updateAudioInfo(): [MP3] 11672954 (11719766 - 46812) / 11719766
[ 28022][I][main.cpp:224] updateAudioInfo(): [MP3] 11672954 (11719766 - 46812) / 11719766
.
.
.
.
[ 28953][I][main.cpp:224] updateAudioInfo(): [MP3] 11703047 (11719766 - 16719) / 11719766
[ 28972][I][main.cpp:224] updateAudioInfo(): [MP3] 11703883 (11719766 - 15883) / 11719766
[ 28981][I][main.cpp:278] updateAudioInfo(): [MP3] Track 1 / 16 | Audio time: 362 / 362
[ 29000][I][main.cpp:224] updateAudioInfo(): [MP3] 11704719 (11719766 - 15047) / 11719766
[ 29019][I][main.cpp:224] updateAudioInfo(): [MP3] 11705555 (11719766 - 14211) / 11719766
[ 29038][I][main.cpp:224] updateAudioInfo(): [MP3] 11706391 (11719766 - 13375) / 11719766
[ 29057][I][main.cpp:224] updateAudioInfo(): [MP3] 11706391 (11719766 - 13375) / 11719766
[ 29076][I][main.cpp:224] updateAudioInfo(): [MP3] 11707227 (11719766 - 12539) / 11719766
[ 29095][I][main.cpp:224] updateAudioInfo(): [MP3] 11707227 (11719766 - 12539) / 11719766
[ 29114][I][main.cpp:224] updateAudioInfo(): [MP3] 11708063 (11719766 - 11703) / 11719766
[ 29133][I][main.cpp:224] updateAudioInfo(): [MP3] 11708899 (11719766 - 10867) / 11719766
[ 29152][I][main.cpp:224] updateAudioInfo(): [MP3] 11709735 (11719766 - 10031) / 11719766
[ 29171][I][main.cpp:224] updateAudioInfo(): [MP3] 11710571 (11719766 - 9195) / 11719766
[ 29190][I][main.cpp:224] updateAudioInfo(): [MP3] 11710571 (11719766 - 9195) / 11719766
[ 29209][I][main.cpp:224] updateAudioInfo(): [MP3] 11711407 (11719766 - 8359) / 11719766
[ 29228][I][main.cpp:224] updateAudioInfo(): [MP3] 11712243 (11719766 - 7523) / 11719766
[ 29247][I][main.cpp:224] updateAudioInfo(): [MP3] 11712243 (11719766 - 7523) / 11719766
[ 29266][I][main.cpp:224] updateAudioInfo(): [MP3] 11713078 (11719766 - 6688) / 11719766
[ 29285][I][main.cpp:224] updateAudioInfo(): [MP3] 11713078 (11719766 - 6688) / 11719766
[ 29304][I][main.cpp:224] updateAudioInfo(): [MP3] 11713914 (11719766 - 5852) / 11719766
[ 29323][I][main.cpp:224] updateAudioInfo(): [MP3] 11714750 (11719766 - 5016) / 11719766
[ 29342][I][main.cpp:224] updateAudioInfo(): [MP3] 11715586 (11719766 - 4180) / 11719766
[ 29361][I][main.cpp:224] updateAudioInfo(): [MP3] 11716422 (11719766 - 3344) / 11719766
[ 29380][I][main.cpp:224] updateAudioInfo(): [MP3] 11716422 (11719766 - 3344) / 11719766
[ 29399][I][main.cpp:224] updateAudioInfo(): [MP3] 11717258 (11719766 - 2508) / 11719766
[ 29418][I][main.cpp:224] updateAudioInfo(): [MP3] 11718094 (11719766 - 1672) / 11719766
[ 29437][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29456][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29475][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29494][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29513][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29532][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29551][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29570][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29589][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29608][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29627][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29646][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29665][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29684][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29703][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
[ 29722][I][main.cpp:224] updateAudioInfo(): [MP3] 11718930 (11719766 - 836) / 11719766
```

I manage to workaround this by detecting when the inBufferSize() value doesn't change anymore while being below a given low value (used 4096) to call stopSong() manually but this is not ideal

Hope this helps.

Regards

closed 2025-06-13 by github-actions[bot]