schreibfaul1/ESP32-audioI2S #1004: TFT_eSPI display freezes when using ESP32-audioI2S

wailay, 2025-03-27:
Hello,

When using the `ESP32-audioI2S `library to play audio on my ESP32 board, the TFT display controlled by the `TFT_eSPI` library freezes and stops updating. It appears that the audio playback is blocking or interfering with the display updates.

TFT display works perfectly when used alone and the issue only occurs when both libraries are used together.

I am not entirely sure if this is the correct 

```cpp
#include <Audio.h> // from ESP32-audioI2S
#include <TFT_eSPI.h>

Audio audio; // <-- this line here makes the TFT display freeze
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sprite = TFT_eSprite(&tft);

void setup() {
  tft.init();
  sprite.createSprite(240, 240);
  tft.fillScreen(TFT_BLACK);

  audio.setPinout(14, 12, 13); // Example I2S pins
  audio.connecttoFS(SPIFFS, "/test.mp3"); // Example audio file
  audio.setVolume(10);
}

int pos = 0;
void loop() {
  audio.loop();

  // This works, but freezes when defining the `Audio audio;` class
  tft.startWrite();
  sprite.fillEllipse(pos, 121, 10, 10, TFT_BLACK);
  pos = (pos + 1) % 222;
  sprite.fillEllipse(pos, 121, 10, 10, TFT_RED);
  tft.endWrite();
}
```

it seems like there is some interference with the way i2s is initialized. If the `Audio audio;` class definition is removed, the TFT works as intended.

I have tried also a dummy I2S initialization similar to what the Audio constructor is doing

```cpp
#include "driver/gpio.h"
#include "driver/i2s_std.h"

void setupI2SEsp() {
    i2s_chan_handle_t m_i2s_tx_handle;
    i2s_chan_config_t m_i2s_chan_cfg;  // stores I2S channel values
    i2s_std_config_t m_i2s_std_cfg;

    m_i2s_chan_cfg.id =
        (i2s_port_t)I2S_NUM_0;  // I2S_NUM_AUTO, I2S_NUM_0, I2S_NUM_1
    m_i2s_chan_cfg.role =
        I2S_ROLE_MASTER;  // I2S controller master role, bclk and lrc signal
                          // will be set to output
    m_i2s_chan_cfg.dma_desc_num = 32;    // number of DMA buffer
    m_i2s_chan_cfg.dma_frame_num = 256;  // I2S frame number in one DMA buffer.
    m_i2s_chan_cfg.auto_clear =
        true;  // i2s will always send zero automatically if no data to send
    i2s_new_channel(&m_i2s_chan_cfg, &m_i2s_tx_handle, NULL);

    m_i2s_std_cfg.slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(
        I2S_DATA_BIT_WIDTH_16BIT,
        I2S_SLOT_MODE_STEREO);  // Set to enable bit shift in Philips mode
    m_i2s_std_cfg.gpio_cfg.bclk =
        I2S_GPIO_UNUSED;  // BCLK, Assignment in setPinout()
    m_i2s_std_cfg.gpio_cfg.din = I2S_GPIO_UNUSED;  // not used
    m_i2s_std_cfg.gpio_cfg.dout =
        I2S_GPIO_UNUSED;  // DOUT, Assignment in setPinout()
    m_i2s_std_cfg.gpio_cfg.mclk =
        I2S_GPIO_UNUSED;  // MCLK, Assignment in setPinout()
    m_i2s_std_cfg.gpio_cfg.ws =
        I2S_GPIO_UNUSED;  // LRC,  Assignment in setPinout()
    m_i2s_std_cfg.gpio_cfg.invert_flags.mclk_inv = false;
    m_i2s_std_cfg.gpio_cfg.invert_flags.bclk_inv = false;
    m_i2s_std_cfg.gpio_cfg.invert_flags.ws_inv = false;
    m_i2s_std_cfg.clk_cfg.sample_rate_hz = 44100;
    m_i2s_std_cfg.clk_cfg.clk_src =
        I2S_CLK_SRC_DEFAULT;  // Select PLL_F160M as the default source clock
    m_i2s_std_cfg.clk_cfg.mclk_multiple =
        I2S_MCLK_MULTIPLE_128;  // mclk = sample_rate * 256
    i2s_channel_init_std_mode(m_i2s_tx_handle, &m_i2s_std_cfg);
    // I2Sstart(m_i2s_num);
    i2s_channel_enable(m_i2s_tx_handle);

    // m_sampleRate = 44100;
}

void setup() {
  tft.init();
  sprite.createSprite(240, 240);
  tft.fillScreen(TFT_BLACK);
 
  setupI2SEsp();

  //audio.setPinout(14, 12, 13); // Example I2S pins
  //audio.connecttoFS(SPIFFS, "/test.mp3"); // Example audio file
  //audio.setVolume(10);
}

```
Getting this error:

```
E (6171) i2s_common: i2s_new_channel(872): intr_priority should be within 0~7
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

Setting the `intr_priority` to 0 which is the default for the i2s_chan_config_t. The I2S is correclty initialized and the TFT display works fine

Logs:
```
D (6171) i2s_common: tx channel is registered on I2S0 successfully
D (6173) i2s_common: DMA malloc info: dma_desc_num = 32, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 1024
D (6181) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 28 [mclk] 5644800 Hz [bdiv] 4 [bclk] 1411200 Hz
D (6191) i2s_std: The tx channel on I2S0 has been initialized to STD mode successfully
D (6199) i2s_common: i2s tx channel enabled
```

Not sure what's the issue here? Why is I2S in this library interfering with the SPI in another library.
Note that I also have another device connected to the SPI and is working fine.

wailay, 2025-03-27:
Could it be because the I2S initialisation happens way too early and should be ideally done in the `setup` routine?

wailay, 2025-03-27:
@Pako2 yeah I am using `tft.startWrite()` and `tft.endWrite()` because I am sharing the SPI port with another device.
Which ESP32 model are you using?

@BoltSwith 

Thanks, I will share the SRAM usage shortly.

As for cause 2. the audio task in this library is already running in Core 0 which is the `PRO_CPU`, the rest of the application logic (TFT) is running on core 1 (`APP_CPU`) so the audio compute is already running in a different core.

wailay, 2025-03-28:
@BoltSwith 

SRAM status with TFT running only
```cpp
I (48251) MAIN: -------------------------------
I (48251) MAIN: Free Heap: 146916 bytes
I (48251) MAIN: Min Free Heap: 146916 bytes
I (48254) MAIN: Internal Free SRAM: 180864 bytes
```

With the audio running

```cpp
I (48502) MAIN: -------------------------------
I (48502) MAIN: Free Heap: 145620 bytes
I (48503) MAIN: Min Free Heap: 145620 bytes
I (48506) MAIN: Internal Free SRAM: 177636 bytes
```

Seems normal to me

Pako2, 2025-03-28:
@wailay 
I'm using the basic ESP32 model (ESP32-WROWER-E). You can actually see that in the photos on my project page.
When I call ESP.getChipModel() and ESP.getChipRevision(), I get the response "ESP32-D0WD-V3 Rev 3".

BoltSwith, 2025-03-28:
@wailay 

From the look of things, you have enough SRAM to run both libraries simultaneously.
For the sake of further troubleshooting, attempt changing the i2sPort used during `Audio audio` initialization from port_0 to port_1

I guess you can edit the source files of your ESP32-audioI2S library. If yes, navigate to the Audio.h file, and edit the Audio constructor.

`Audio(bool internalDAC = false, uint8_t channelEnabled = 3, uint8_t i2sPort = I2S_NUM_1); `

Set the i2sPort parameter to I2S_NUM_1 (if I2S_NUM_0 was previously assigned)

Test your code.............

If problem persists, navigate to the Audio.cpp file and set the number of DMA Buffer to 16 (32 is previously assigned)

```
    // -------- I2S configuration -------------------------------------------------------------------------------------------
#if ESP_IDF_VERSION_MAJOR == 5
    m_i2s_chan_cfg.id            = (i2s_port_t)m_i2s_num;  // I2S_NUM_AUTO, I2S_NUM_0, I2S_NUM_1
    m_i2s_chan_cfg.role          = I2S_ROLE_MASTER;        // I2S controller master role, bclk and lrc signal will be set to output
    m_i2s_chan_cfg.dma_desc_num  = 16;                     // number of DMA buffer
    m_i2s_chan_cfg.dma_frame_num = 256;                    // I2S frame number in one DMA buffer.
    m_i2s_chan_cfg.auto_clear    = true;                   // i2s will always send zero automatically if no data to send
    i2s_new_channel(&m_i2s_chan_cfg, &m_i2s_tx_handle, NULL);
```

Test your code Again........

Finally, make sure there are no pins overlap between the TFT_eSPI and ESP32-audioI2S libraries.

Confirm your pins assignments in your code:

`audio.setPinout(14, 12, 13); `

Wish your luck......................!!!!

wailay, 2025-03-28:
@BoltSwith I have pretty much tried everything you have suggested 😭 

At least I have the confirmation from @Pako2 that the libraries should both work fine together.

wailay, 2025-03-30:
I think I have finally identified why the TFT screen freezes.
I have created a new project and retried from scratch and was able to successfully get both the audio and the screen to work together.

However right when I use sprites for the TFT display, the screen freeze

```cpp

void audioTask(void *params) {
  Serial.println("Audio task started");
  while (1) {
      audio.loop();
      vTaskDelay(1);
  }
}

void setup() {
   tft.init();
    tft.setRotation(0);
    sprite.createSprite(240, 240); 
    tft.fillScreen(TFT_BLACK);

    audio.setPinout(22, 33, 21);
    audio.setVolume(14);  // default 0...21
    audio.connecttohost("radio-url");

    BaseType_t createdTask = xTaskCreatePinnedToCore(audioTask, "audioTask", 4096, NULL, 2, NULL, 0);

    Serial.println("Task created: " + String(createdTask));
}
int pos = 0;
void loop() {

    // sprite.fillEllipse(pos, 121, 10, 10, TFT_BLACK); <-- The screen works fine unless I use sprites
    tft.fillEllipse(pos, 121, 10, 10, TFT_BLACK);
    pos = (pos + 1) % 222;
    // sprite.fillEllipse(pos, 121, 10, 10, TFT_RED);
    tft.fillEllipse(pos, 121, 10, 10, TFT_RED);

    // sprite.pushSprite(0, 0);

    vTaskDelay(1);
}
```

@Pako2 I see you are also using sprites in your project. Not sure what could be going wrong here?

wailay, 2025-03-30:
Closing as it's not an issue with this library

closed 2025-03-30 by wailay

Pako2, 2025-03-31:
@wailay
> I see you are also using sprites in your project. Not sure what could be going wrong here?

Unfortunately, I have no idea what causes this.