schreibfaul1/ESP32-audioI2S #1008: Needed to make changes in Audio.cpp

TroKhon, 2025-04-02:
After experiencing resets due to memory overrun, I tried to solve the problem.

Error message was:

[Audio.cpp:125] bytesWritten(): m_writePtr 1009183168, m_endPtr 1009182826 Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

So I changed to code in Audio.cpp in the accoring lines. Now reading:

void AudioBuffer::bytesWritten(size_t bw) {
    if(!bw) return;
    m_writePtr += bw;
    if (m_writePtr >= m_endPtr) {
        log_e("Buffer Overflow m_writePtr! Resetting m_writePtr. mBuffer. bw = %i",bw);
        m_writePtr = m_buffer;  // Back to the begin of the buffer
    }
    m_f_isEmpty = false;
}

At least this helped, now I have no longer resets..... but maybe this was only a unique problem in my environment - I am using some of the very good YB-ESP32-S3-AMP boards. ArduinoIDE settings with Flash 8MB,  Partition scheme 8M,  QSPI PSRAM enabled. But not sure that I have chosen the right place with Audio.cpp to solve the problem .... on the other hand it helped...

anp59, 2025-04-06:
Hi TroKhon ,
if you use the example **Play-All-MP3-Files-on-SD** from yellobyte (https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/blob/main/boards/YB-ESP32-S3-AMP/examples/PlatformIO/Test-SdFatFS/src/main.cpp), 
the line `vTaskDelay(1);`  must be added to the loop() function. 
These example dates back to Arduino version 2.x.x and must be adapted. Changing the code in audio.cpp should not be the solution.
The example https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/tree/main/boards/YB-ESP32-S3-AMP/examples/PlatformIO/Test-SdFatFS is newer and already contains this line.

Please also note the information from @schreibfaul1 in issue #887

yellobyte, 2025-04-06:
anp59 actually refers to example **Test-SdFatFS** (https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/tree/main/boards/YB-ESP32-S3-AMP/examples/PlatformIO/Test-SdFatFS) !
His mentioned example **Play-All-MP3-Files-on-SD** (https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/tree/main/boards/YB-ESP32-S3-AMP/examples/PlatformIO/Play-All-MP3-Files-on-SD) works as well though.

yellobyte, 2025-04-07:
All examples on https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/tree/main/boards/YB-ESP32-S3-AMP/examples/PlatformIO/Play-All-MP3-Files-on-SD have been updated, build with Arduino ESP32 v3.1.2/ESP-audioI2S v3.0.13 and successfully tested. No crashes/core panics at all.

TroKhon, 2025-04-07:
Yep, that did the trick. As I wrote, I was not sure if I tackled the problem in the right way. I downloaded the examples quite some time ago......

closed 2025-04-07 by TroKhon