schreibfaul1/ESP32-audioI2S #1011: How enable sound Speaker for ESP32-2432S028 CYD

ultranano, 2025-04-08:
Hi ALL!

i Want to use the library for using with ESP32-2432S028 CYD (Cheap Yellow Display)

It's possible to use the internal speaker output as indicated on the image below?

[image]

Thanks in advance

Andrea

schreibfaul1, 2025-04-08:
I think that will work. There are no problems with this display https://www.makerfabs.com/sunton-esp32-s3-7-inch-tn-display-with-touch.html, then it will also work with the "little brother". However, there is only one channel and the output is mono.

schreibfaul1, 2025-04-08:
The connection between ESP32 and DAC is hard-wired on the board. Use the schematic or the manufacturer's documentation to determine the pins and enter them here:
`audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);`
`DOUT` on the ESP32 is `DIN` on the DAC
Everything else is as in the example sketch.

nodoubtman, 2025-04-19:
Here you go:

#define BCLK  4 // are using Set to I2S pin
#define LRC  16 // are using Set to I2S pin
#define DIN  17 // are using Set to I2S pin 

:-)
Marc.

enjoyneering, 2025-04-22:
As far as I know, the current branch does not support the internal DAC of the ESP32. You will have to use an external I2S DAC, such as the PCM5102.

schreibfaul1, 2025-04-22:
Exactly, since Arduino V3 (IDF V5...) I2S is no longer supported for the internal DAC.
There is another option using DMA, but we are not doing that here.

closed 2025-06-06 by github-actions[bot]