schreibfaul1/ESP32-audioI2S #1013: Choppy audio for HLS stream .m3u8 stations when using "better WiFi speed"i sdkconfig and espidf, works correctly when using only Arduino.

amishv, 2025-04-12:
I tried Better_WiFi_throughput example on esp32-wrover with 8MB PSRAM as there were a lot of packets drop even for 128K streams. The update made mp3 steams work better, however the audio from  HLS streams got choppy.

The stream tested were 
https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/playlist.m3u8"
https://air.pc.cdn.bitgravity.com/air/live/pbaudio059/playlist.m3u8.

These streams work flawless when only Arduino framework is selected.

------------------------------------------
Chip Info:

  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 1.00
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
 
INTERNAL Memory Info:

  Total Size        :   313007 B ( 305.7 KB)
  Free Bytes        :   195255 B ( 190.7 KB)
  Allocated Bytes   :   113376 B ( 110.7 KB)
  Minimum Free Bytes:   190995 B ( 186.5 KB)
  Largest Free Block:   110592 B ( 108.0 KB)

SPIRAM Memory Info:

  Total Size        :  4194304 B (4096.0 KB)
  Free Bytes        :  4190912 B (4092.7 KB)
  Allocated Bytes   :     1184 B (   1.2 KB)
  Minimum Free Bytes:  4190912 B (4092.7 KB)
  Largest Free Block:  4128768 B (4032.0 KB)
  Bus Mode          : QSPI

Flash Info:

  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : DIO

Board Info: 

  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Core Debug Level  : 5
  Arduino Runs Core : 1
  Arduino Events on : 1
  CDC On Boot       : 0

schreibfaul1, 2025-04-13:
The example "Better_WiFi_throughput" comes from the time of Arduino V2. With Arduino V3 (IDF V5xx) the WiFi throughput is better. If you are looking for a template for PlatformIO you can test this: https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF

amishv, 2025-04-13:
Thanks @schreibfaul1 . The cocktail of versions was causing this issue.
Closing this issue as completed.

closed 2025-04-13 by amishv