schreibfaul1/ESP32-audioI2S #1014: Audio-jumps and not detecting the end of songs (mp3)

Joe91, 2025-04-14:
As mentioned here:
https://forum.espuino.de/t/musiktitel-huepft-waehrend-des-abspielens-vor-spielt-weiter-bis-zum-titelende-und-stoppt-dann/3597/29

Audio issues, that don't happen on rev 567abd2, but on all the alter versions:

Here some error-log created with the latest commit on master (d7b88374d14b5ec4074626c40618a81c7f35b08c):
```
N [28319] Modus: Spiele alle Tracks (sortiert) des Ordners '/Example! Folder'
I [28329] Sorting files using case-insensitive natural sorting
N [28332] Neue Playlist mit 13 Titel(n) empfangen
D [28342] Free heap: 103448
I [28400] info        : PSRAM found, inputBufferSize: 630773 bytes
I [28402] info        : buffers freed, free Heap: 103448 bytes
I [28411] info        : Reading file: "/Example! Folder/01-test, test, test, test.mp3"
I [28421] info        : MP3Decoder has been initialized, free Heap: 78696 bytes , free stack 3184 DWORDs
N [28422] '/Example! Folder/01-test, test, test, test.mp3' wird abgespielt (1 von 13)
I [28898] info        : Content-Length: 4614719
I [28899] info        : ID3 framesSize: 4096
I [28899] info        : ID3 version: 2.3
I [28900] info        : ID3 normal frames
I [28919] id3data     : Artist: Test
I [28923] id3data     : Album: Test
I [28927] id3data     : Title: test, test, test, test
I [28931] id3data     : Year: 2015
I [28947] id3data     : Track: 1/13
I [28951] id3data     : ContentType: Undefined
I [28955] id3data     : Length (ms): 196200
I [28959] info        : Audio-Length: 4610623
I [28960] info        : stream ready
I [28961] info        : syncword found at pos 0
I [28971] info        : MPEG-2.5, Layer I
I [28972] info        : Channels: 2
I [28973] info        : SampleRate: 44100
I [28973] info        : BitsPerSample: 16
I [28984] info        : BitRate: 128000
D [32059] ws[/ws][1] connect
D [33066] no cover image for SD-card audio
D [60003] RSSI: -43 dBm
I [77756] Neue Lautstärke empfangen via Queue: 1
I [78316] Neue Lautstärke empfangen via Queue: 2
I [82826] info        : MP3 decode error -6 : INVALID_FRAMEHEADER
I [82828] info        : syncword found at pos 238
I [82828] info        : syncword found at pos 0
I [82844] info        : MP3 decode error -9 : INVALID_HUFFCODES
I [82846] info        : syncword found at pos 521
I [82846] info        : syncword found at pos 0
I [82870] info        : MPEG-2.5, Layer I
I [82872] info        : Channels: 2
I [82872] info        : SampleRate: 44100
I [82872] info        : BitsPerSample: 16
I [82884] info        : BitRate: 224000
I [127361] info        : MP3 decode error -6 : INVALID_FRAMEHEADER
I [127362] info        : syncword found at pos 227
I [127363] info        : syncword found at pos 0
I [127382] info        : MPEG-2.5, Layer I
I [127383] info        : Channels: 2
I [127383] info        : SampleRate: 44100
I [127384] info        : BitsPerSample: 16
I [127394] info        : BitRate: 192000
```

Will stay there forever...
Not specific to this file, a lot of files are affected.

schreibfaul1, 2025-04-15:
This is a strange thing. Basically two processes are running, these are:
1) audio.loop(); reads the data from the SD card or from Wifi and writes it to a buffer.
2) internal task; fetches the data from the buffer, decodes it and feeds I2S
If the buffer is well filled, several seconds can be bridged without interruptions. audio.loop(); is therefore not time-critical. If audio.loop() and the internal task are running on different cores, they can access the buffer at the same time without collisions occurring. 

Example;
 -D ARDUINO_RUNNING_CORE=1
audio.setAudioTaskCore(0);

For large projects, the LOOP_STACK (Arduino loop) may be too small (8KB is standard), then:
SET_LOOP_TASK_STACK_SIZE(10 * 1024);

It looks like audio.loop() is interrupted during execution. I have no idea why. It reads from the SD card but does not write to the buffer.
The size of the audio block is known; if not enough data is written to the buffer, the end of a local file cannot be recognised.

Unfortunately, I have no idea what the exact cause is, but if it is due to this library I would certainly have already received many similar error messages.

Joe91, 2025-04-15:
So what would you suggest to further debug this? In this System I use a quite slow SD-card. May this affect this? 
I know there are a lot of changes all the time, but might there be any reason why there is no issue at all when going back to this old version?

schreibfaul1, 2025-04-15:
I don't think it will be the SD card. Your project uses a separate task for audio.loop() I think.
You should look at this task. It could be omitted to save resources.

Joe91, 2025-04-15:
Will have a look at this. Still strange why changing only this library can get rid of the issues...

Joe91, 2025-04-16:
Sorry, to bother you again, but what would you suggest us to change on the ESPUINO-Project to use the audio-lib as intended?
Do you have an example you recommand for the structure?

Thanks in advance and best regards
Joe

Joe91, 2025-04-17:
For Debugging and more flexibilty in huge projects: 
would it be possible to create a setter for the internal task-priority, just how it is already existing to select the core of this task?

dronecz, 2025-04-19:
> Unfortunately, I have no idea what the exact cause is, but if it is due to this library I would certainly have already received many similar error messages.

I just started to work with this library, and I can confirm that the sound volume is jumping up and down. I used the same library (3.0.13) with ESP32 (no PSRAM) and ESP32-S3 (version N4R2), and both boards have this issue.

Joe91, 2025-04-23:
For me it seems fixed now, since the buffers were changed. Thanks a lot for this! If it still occours (still testing) then we can still change them back completely. Will close this issue then.

Joe91, 2025-05-16:
Sadly it's still happening for me even after we got rid of the Audio-Task in our project and use the latest version of the libary.
Are there more thing we can do, to further narrow the reason down and to maybe solve this finally?
Is it just too much other stuff going on?
Is there a maximum time, that must not be exceeded between two Audio-loop-calls for example?

closed 2025-06-30 by github-actions[bot]