schreibfaul1/ESP32-audioI2S #1018: HLS  .m3u8 MP3 (non AAC) streams are not decoded.

amishv, 2025-04-20:
Stations like https://air.pc.cdn.bitgravity.com/air/live/pbaudio047/playlist.m3u8 that play mp3 audio instead of AAC are not supported.
where as atation w
from VLC 
Codec: MPEG Audio layer 1/2 (mpga)

where as supported station like https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/playlist.m3u8 have codec as ADTS.

further debugging in audio.cpp the codec is always selected as CODEC_AAC = 3 indifferent to the stream.

in  Audio::ts_parsePacket () we need to add check for MP3 stream as well and handle the stream.

if(streamType == 0x0F || streamType == 0x11) {
                        if(m_f_Log) log_i("AAC PID discover");
                        pidOfAAC = elementaryPID;
}
else if(streamType == 0x04 ) {

Still trying to understand the flow will update the finding and hopefully a solution.

schreibfaul1, 2025-04-20:
Hi @amishv ,
this is rare, I know HLS streams are not always AAC, MP3 is also part of the spec, the difference here is:

 https://air.pc.cdn.bitgravity.com/air/live/pbaudio047/playlist.m3u8
`#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:BANDWIDTH=127240,CODECS="mp4a.40.34"
chunklist.m3u8`

https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/playlist.m3u8
`#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:BANDWIDTH=95150,CODECS="mp4a.40.2"
chunklist.m3u8`

where `mp4a.40.34` refers to MP3.
I will adjust this in the next few days

Best regards
Wolle

amishv, 2025-04-20:
@schreibfaul1 , thanks for looking into this. I added some debugs in the code  to find out stream type 0x04 was not implemented yet.

lasthost    https://air.pc.cdn.bitgravity.com/air/live/pbaudio047/playlist.m3u8
[  6050][E][Audio.cpp:4037] parseContentType():  ********* content type = application/vnd.apple.mpegurl initial codec value = 0, m_expected =0
[  6063][E][Audio.cpp:4146] parseContentType():  returning ********* content type = application/vnd.apple.mpegurl returning codec value = 0, m_expected =0
info        content-length: 95
[  6081][I][Audio.cpp:3930] parseHttpResponseHeader(): now parse playlist
[  6108][I][Audio.cpp:2923] m3u8redirection(): codeString   and codec = 0
[  6117][I][Audio.cpp:2923] m3u8redirection():  codeString   and codec = 0
[  6125][I][Audio.cpp:2923] m3u8redirection():  codeString CODECS="mp4a.40.34"  and codec = 2
[  6135][I][Audio.cpp:2923] m3u8redirection():  codeString   and codec = 2
info        buffers freed, free Heap: 173559 bytes
info        connect to: "air.pc.cdn.bitgravity.com" on port 443 path "/air/live/pbaudio047/chunklist.m3u8"
info        SSL has been established in 755 ms, free Heap: 173331 bytes
lasthost    https://air.pc.cdn.bitgravity.com/air/live/pbaudio047/chunklist.m3u8
[  7068][E][Audio.cpp:4037] parseContentType():   content type = application/vnd.apple.mpegurl initial codec value = 0, m_expected =0
[  7081][E][Audio.cpp:4146] parseContentType():  returning  content type = application/vnd.apple.mpegurl returning codec value = 0, m_expected =0
info        content-length: 196
[  7099][I][Audio.cpp:3930] parseHttpResponseHeader(): now parse playlist
info        media sequence number: 2223
[  7128][I][Audio.cpp:2835] parsePlaylist_M3U8(): now playing https://air.pc.cdn.bitgravity.com/air/live/pbaudio047/media_2223.ts
info        connect to: "https://air.pc.cdn.bitgravity.com/air/live/pbaudio047/media_2223.ts"
[  7145][E][Audio.cpp:692] httpPrint():  Extention = /air/live/pbaudio047/media_2223.ts
[  7160][E][Audio.cpp:711] httpPrint():  Extention = /air/live/pbaudio047/media_2223.ts  m_expectedPlsFmt=4 
[  7324][E][Audio.cpp:4037] parseContentType():   content type = video/mp2t initial codec value = 3, m_expected =0
[  7336][I][Audio.cpp:4085] parseContentType(): ContentType video/mp2t, format is aac
[  7344][E][Audio.cpp:4146] parseContentType():  returning  content type = video/mp2t returning codec value = 3, m_expected =0
info        content-length: 87232
[  7360][I][Audio.cpp:3943] initializeDecoder():  in initializeDecoder codec = 3
info        AACDecoder has been initialized, free Heap: 173047 bytes , free stack 4072 DWORDs
[  7371][I][Audio.cpp:3925] parseHttpResponseHeader(): Switch to DATA, metaint is 0
[  7397][I][Audio.cpp:2190] process_m3u8_ID3_Header(): m3u8 file has no mp3 tag
[  7415][I][Audio.cpp:5592] ts_parsePacket(): PMT
[  7421][I][Audio.cpp:5595] ts_parsePacket(): Section Length: 55
[  7427][I][Audio.cpp:5597] ts_parsePacket(): Program Info Length: 17
[  7435][I][Audio.cpp:5602] ts_parsePacket(): Stream Type: 0x15 Elementary PID: 0x0102
[  7443][I][Audio.cpp:5614] ts_parsePacket(): ES Info Length: 0x000F
[  7450][I][Audio.cpp:5602] ts_parsePacket(): **Stream Type: 0x04** Elementary PID: 0x0101
[  7459][I][Audio.cpp:5609] ts_parsePacket(): MP3 PID discover
[  7468][I][Audio.cpp:5614] ts_parsePacket(): ES Info Length: 0x0000

Hope this helps.

schreibfaul1, 2025-04-20:
I have changed a few things.
The MP3 decoder is now initialised correctly and the StreamType for mp3 is actually **0x04**.
Only a few people know that, I think you're deep in the subject.

But unfortunately it still doesn't work correctly. In the first **ts** after the **m3u8** something is still being transmitted that I have to filter out. This prevents the MP3 decoder from remaining synchronised. I still have no idea what this is. In general, there is an ID3 header at the beginning.
But that's enough for today. Thanks for your help!

amishv, 2025-04-20:
It seems that the stream is recognised correctly 

[ 6125][I][Audio.cpp:2923] m3u8redirection(): codeString CODECS="mp4a.40.34" and codec = 2
[ 6135][I][Audio.cpp:2923] m3u8redirection(): codeString and codec = 2

but the codec changes to AAC in parseContentType and AAC encoder is initialized.

7160][E][Audio.cpp:711] httpPrint(): Extention = /air/live/pbaudio047/media_2223.ts m_expectedPlsFmt=4
[ 7324][E][Audio.cpp:4037] parseContentType(): content type = video/mp2t initial codec value = 3, m_expected =0
[ 7336][I][Audio.cpp:4085] parseContentType(): ContentType video/mp2t, format is aac
[ 7344][E][Audio.cpp:4146] parseContentType(): returning content type = video/mp2t returning codec value = 3, m_expected =0
info content-length: 87232
[ 7360][I][Audio.cpp:3943] initializeDecoder(): in initializeDecoder codec = 3
info AACDecoder has been initialized, free Heap: 173047 bytes , free stack 4072 DWORDs

Can we select the codec in m3u8redirection() and not change it for m3u8?

I tried to change the codec based on stream type 
`if(streamType == 0x0F || streamType == 0x11) {
if(m_f_Log) log_i("AAC PID discover");
pidOfAAC = elementaryPID;
}
else if(streamType == 0x04 ) {
if(m_f_Log) log_i("MP# PID discover");
pidOfMP3 = elementaryPID;`

but need to understand the code structure better to implement pidOfMP3, will try something next weekend.
Have a nice weekend and Happy Easter!

schreibfaul1, 2025-04-21:
Now it works

amishv, 2025-04-21:
Thanks Wolle for a quick turnaround and creating an excellent library.

I was trying to get the codec selected based on the EXT-X-STREAM-INF only, but would have taken time to understand the code flow.

Thanks again!

Closing as completed.

closed 2025-04-21 by amishv