schreibfaul1/ESP32-audioI2S #1020: .isRunning() issue - falling back to 'false' delayed

kaloprojects, 2025-04-25:
Hey Wolle,

i love your **openai_speech()** call, using in all my projects (e.g. latest ESP32-Voice-ChatGPT).
audio_play.openai_speech(OPENAI_KEY, "tts-1", LLM_Feedback.c_str(), Voice, "aac", "1");    

One issue i struggle since a longer time, not a big deal/issue more and more annoying, its related to your function **.isRunning()**:
I use this is.Running for multiple use cases (it is a great feature !) .. 2 examples how i use often:

Use case 1:  e.g. triggering my RGB leds on status, example:

```
if (audio_play.isRunning())      
{  // led MAGENTA when Open AI voice is speaking
   led_RGB(LOW,HIGH,LOW);  
}    
else  
{   // led GREEN (default) when ready for next user request
    // issue: the led is still magenta for 2 secs (even TTS is done/finished)
    led_RGB(HIGH,LOW,HIGH); 
}  

```
or Usecase 2 - playing 2 IS2 audio events in sequence:

```
// 1st playing an Open AI voice (TTS):
audio_play.openai_speech(OPENAI_KEY, "tts-1", WELCOME_MSG, FAVORITE_VOICE_TTS, "aac", "1");   
while (audio_play.isRunning())
{ audio_play.loop();   // <- here we want wait until done, works perfect ..
}   
// 2nd then we play another Audio event, e.g. a wav file
// .. unfortunatly 2 secs delay before wav starts (because is.Running above becomes false delayed)
audio_play.connecttoFS( SD, "welcome.wav" );  

```

Means: .isRunning() returns 'true' far too long (about 2 seconds), maybe because you added a latency by purpose (i don't know) ? 

It's an issue because my workflow breaks (led are not in sync, or user has to wait before next action starts)
Q: Do you have any workaround (e.g. any eof pointer i could check when I2S stream is done) any alternative or add-on to .isRunning() ?

All idea would help (even some tricky workaround would be fine)

thanks in prepare, i love your library !

Kalo

kaloprojects, 2025-04-25:
additional info, not sure this helps. I added 
```
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
```
to see what happens when audio TTS is done (via openai_speech(..)), but isRunning() is still true. That's what is printed in the last 2 seconds (after audio was finished correctly but isRunning still true):
```
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        End of Stream.
``` 
not sure this helps ;)

schreibfaul1, 2025-04-26:
Unfortunately I can't contribute much to the problem, an OpenAI subscription would be required.
The `openai_speech()` function was written by our users.

What strikes me is this:

[image]

I think `ST_WEBSTREAM` should be replaced by `ST_WEBFILE` so that the contntLenght is taken into account.

kaloprojects, 2025-04-26:
Great, thank you for the idea @schreibfaul1 !

@akdeb: Hi ;) .. Seen that you did some[ instructions](https://github.com/schreibfaul1/ESP32-audioI2S/pull/1003) add-ons to openai(), so great ! .. do you also have a chance to add & test this `ST_WEBFILE` replacement ? I am not experienced in 'merging user mods', that's why asking. otherwise i will test on my local copy.

schreibfaul1, 2025-04-26:
Great, I've changed it.
I wish you every success with your project!

kaloprojects, 2025-04-27:
Hey @schreibfaul1  Wolle,
thank you so much .. for the fast code merge and you wishes to my projects ;) .. appreciated !

I immediately tested your updated library, unfortunately bad news ;( .. the replacement with `ST_WEBFILE` did not solve it, it results in the situation that openai_speech() does no longer work at all (no audio at all). I checked the info messages .. now this happens:
```
info        buffers freed, free Heap: 179036 bytes
info        Connect to: "api.openai.com"
info        SSL has been established in 948 ms, free Heap: 136820 bytes
info        AACDecoder has been initialized, free Heap: 135940 bytes , free stack 2464 DWORDs
```
then no progress, no audio. So sorry to say .. but i feel you should merge back to original `ST_WEBTREAM`. Sorry for making this effort again for you.

Hmm. since there is another root cause why the original code does not detect the end of audio streaming. Maybe @akdeb has any idea  what has to be modified in original `.isRunning()` or `openai_speech()` to detect stream end in time (without 2 sec delay)?

Wolle, maybe i will use a work around until some one finding the root cause and a final solution: I will add an additional flag in my main loop() which will set to false triggered by info "slow stream, dropouts are possible" from 'audio_info(const char *info)', using this own flag in addition with `.isRunning`. I know, a dirty hack in my code .. but should work because this 'slow stream..' info typically never appears during regular playing, so i could use it as end maker. I will test, at least for now i have no better idea, maybe works.

@akdeb : btw, just an info to your 'instruction' update. I could test it .. it works great ! ;). What i also learned, the older (default) model _'tts-1'_ still works, but instructions are much more realistic if using the model _'gpt-4o-mini-tts'_, right ?. Great instruction to test is e.g. "you are whispering". One issue which you might think about, before your add-on is merged in final library ?: Maybe adding the new openai_speech() as an overloaded function .. because now it has 7 parameter (in past 6) .. so it is not downward compatible .. and some user might struggle with their existing code. I also had to modify the call on several lines in my code to get Wolle library with your additional instruction parameter running. Just as thought ;)
  
meanwhile: thank you @schreibfaul1 Wolle, 
best wishes on success on _your_ library ! .. it is best and famous already !! :)

schreibfaul1, 2025-04-27:
I have reset it again.
`ST_WEBSTREAM` is intended for Internet radio and has no time limit,
`ST_WEBFILE` knows the length of the file and can thus determine the end.

Unfortunately, I don't know what OpenAI sends back.

Can you please activate this line:

[image]

The answer should look something like this:

[image]

Apparently the answer is somehow different.

kaloprojects, 2025-04-27:
i just activated the line (removed //). 
But i am using Arduino IDE, so not sure where i can find the result ? .. : | Your screenshot above its a log print, right ?

In case you mean the Serial Monitor, output (from 'void audio_info(const char *info)' ) .. this prints this:

```
info:   buffers freed, free Heap: 179332 bytes
info:   Connect to: "api.openai.com"
info:   SSL has been established in 960 ms, free Heap: 137316 bytes
info:   AACDecoder has been initialized, free Heap: 133876 bytes , free stack 2412 DWORDs
info:   stream ready
info:   syncword found at pos 0
info:   slow stream, dropouts are possible
info:   Channels: 1
info:   SampleRate: 24000
info:   BitsPerSample: 16
info:   BitRate: 59437
info:   AAC HeaderFormat: ADTS
info:   slow stream, dropouts are possible
info:   slow stream, dropouts are possible
info:   slow stream, dropouts are possible
info:   End of Stream.

```

schreibfaul1, 2025-04-27:
Yes, I use PlatformIO.
"Core Debug Level" in Arduino-IDE is probably set to None

[image]

kaloprojects, 2025-04-28:
Cool !, i have never seen this debug toggle in Arduino IDE.
Here the output when launching (and hearing audio) from OpenAI TTS:
```
.. [Hallo, wie geht es dir?]
OpenAI LLM> ..... [Hallo, KALO! Mir geht's super, danke der Nachfrage! Ich habe gerade einen neuen Witz über einen Kühlschrank gehört – aber das erzähl ich dir ein anderes Mal. Wie läuft's bei dir?]
OpenAI Voice = [onyx]
info:   buffers freed, free Heap: 179200 bytes
info:   Connect to: "api.openai.com"
info:   SSL has been established in 586 ms, free Heap: 137116 bytes
[ 27427][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: HTTP/1.1 200 OK
[ 27434][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: date: Mon, 28 Apr 2025 05:55:33 GMT
[ 27443][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: content-type: audio/aac
[ 27450][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: connection: close
[ 27457][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: access-control-expose-headers: X-Request-ID
[ 27467][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: openai-organization: user-k8xn0wqxkpwkwejbtcubg6jn
[ 27477][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: openai-processing-ms: 1077
[ 27484][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: openai-version: 2020-10-01
[ 27492][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: strict-transport-security: max-age=31536000; includeSubDomains; preload
[ 27504][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: via: envoy-router-75c4587d8-tnq8f
[ 27512][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: x-envoy-upstream-service-time: 1064
[ 27521][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: x-ratelimit-limit-requests: 500
[ 27529][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: x-ratelimit-remaining-requests: 499
[ 27538][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: x-ratelimit-reset-requests: 120ms
[ 27546][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: x-request-id: req_27c186e92e7f7a8379476c8455ca4664
[ 27556][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: cf-cache-status: DYNAMIC
[ 27564][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: set-cookie: __cf_bm=2Hcdcc5znFylvfsR4HRz1QFEpMOAKZ_C_487P10rF.k-1745819733-1.0.1.1-leRvgI8DkUpV6mHikRmFWOd0Q2A5GxTlOLM2i4jyvOY_zIw31CaLcjowlXDnY_JSTgdtvMGSCG1n.64quT6pD0C2o69vGM04AmfQWxAppUw; path=/; expires=Mon, 28-Apr-25 06:25:33 GMT; domain=.api.openai.com; HttpOnly; Secure; SameSite=None
[ 27595][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: x-content-type-options: nosniff
[ 27604][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: set-cookie: _cfuvid=gpWYPyXh4Rg8QA701a7qlqYa7IKfk9FPhfZd3pd5_GU-1745819733429-0.0.1.1-604800000; path=/; domain=.api.openai.com; HttpOnly; Secure; SameSite=None
[ 27623][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: server: cloudflare
[ 27630][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: cf-ray: 93744facd9629238-FRA
[ 27638][E][Audio.cpp:3747] parseHttpResponseHeader(): rhl: alt-svc: h3=":443"; ma=86400
info:   AACDecoder has been initialized, free Heap: 131884 bytes , free stack 2416 DWORDs
info:   stream ready
info:   syncword found at pos 0
info:   slow stream, dropouts are possible
info:   Channels: 1
info:   SampleRate: 24000
info:   BitsPerSample: 16
info:   BitRate: 59062
info:   AAC HeaderFormat: ADTS
info:   slow stream, dropouts are possible
info:   slow stream, dropouts are possible
info:   slow stream, dropouts are possible
info:   slow stream, dropouts are possible
info:   End of Stream.

```
thx for asking

schreibfaul1, 2025-04-28:
Thanks for your log.
I have simulated the HTTP ResponseHeader from OpenAI and I think it will work now.

There is neither `contentLength` nor `chunkSize`, but `connection: close` after the payload.
All data should then have been transmitted when the server disconnects.
And then it waits until all data has been decoded and output via I2S.

kaloprojects, 2025-04-28:
Hi Wolle,
first of all: You are a magic master ;) .. and awesome how much effort you spend for us !!! 

Don't kill me .. but i have bad news .. i immidiatly tested but now it 'crashes' on most Openai calls, or alternativelly it speaking single words then either stops or repeates the word endless. Here a typical output when it crashes:

```

== KALO ESP32 Voice ChatGPT (last update: April 28, 2025) ==================================

Sketch Size: 1600272
Total Heap:   309884
Free Heap:    220228
Flash Size:  8388608
Total PSRAM: 4194304
Free PSRAM:  4192124

> Connecting WLAN .. Done, device connected.

Free PSRAM after 2MB Malloc():  2164684

for your info @schreibfaul1: fyi: i am using PSRAM (4MB), malloced for own usage (for testing) 2MB. 
ESP32 Wrover Dev Kit.

============================================================================================
.... [Hallo, wie geht es dir?]
OpenAI LLM> ...... [Hallo KALO! Mir geht's blendend, danke! Ich habe gerade ein neues Rezept für Spaghetti mit Schokoladensauce ausprobiert – das war eine interessante Kombination! Und wie läuft's bei dir, mein Freund?]
OpenAI Voice = [onyx]

<<<- here the Openai_speech() call starts: 

info:   buffers freed, free Heap: 178132 bytes
info:   Connect to: "api.openai.com"
info:   SSL has been established in 568 ms, free Heap: 136008 bytes
info:   AACDecoder has been initialized, free Heap: 132456 bytes , free stack 2416 DWORDs
info:   Webfile: stream ready, buffer filled in 7 ms
info:   syncword found at pos 0
info:   Channels: 1
info:   SampleRate: 24000
info:   BitsPerSample: 16
info:   BitRate: 50062
info:   AAC HeaderFormat: ADTS
[ 38134][E][ssl_client.cpp:36] _handle_error(): [data_to_read():401]: (-29184) SSL - An invalid SSL record was received
[ 38134][E][ssl_client.cpp:36] _handle_error(): [data_to_read():401]: (-29184) SSL - An invalid SSL record was received
[ 38155][E][NetworkClientSecure.cpp:309] available(): Closing connection on failed available check
[ 38155][E][NetworkClientSecure.cpp:309] available(): Closing connection on failed available check
CORRUPT HEAP: Bad head at 0x3ffd1680. Expected 0xabba1234 got 0x3ffcd5bc

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)

Backtrace: 0x40082b99:0x3ffc4f00 0x4008ee7d:0x3ffc4f20 0x4009424a:0x3ffc4f40 0x400933d5:0x3ffc5070 0x40083cdb:0x3ffc5090 0x40086b45:0x3ffc50b0 0x4016ee1e:0x3ffc50d0 0x4016129e:0x3ffc50f0 0x4010ccce:0x3ffc5110 0x4010b955:0x3ffc5130 0x4010bc51:0x3ffc5150 0x4010bb1e:0x3ffc5180 0x401bfc05:0x3ffc51c0 0x400e547e:0x3ffc51f0 0x400e5655:0x3ffc5210 0x4009158a:0x3ffc5230

ELF file SHA256: 43e75699d96776e1

*r⸮⸮Pۨ⸮⸮⸮⸮⸮-⸮⸮⸮⸮-X⸮⸮⸮b&⸮n'nbf(⸮⸮⸮-⸮"⸮⸮d⸮cbv#⸮%⸮⸮f8⸮~⸮⸮r⸮-⸮@%VF-⸮dN8|⸮t⸮X⸮⸮⸮'p⸮,P⸮w⸮t⸮,⸮⸮-⸮ׯ⸮P⸮⸮'p⸮,P⸮⸮'p⸮,P⸮X~⸮-jq⸮|⸮)⸮EX⸮H~t⸮쭷,⸮⸮٫⸮⸮[!⸮(X⸮⸮챱⸮٫⸮⸮/⸮H~t⸮,⸮⸮⸮⸮⸮an ⸮ɪ⸮T⸮⸮⸮.⸮am⸮%a|⸮,⸮T⸮⸮"r [   454][I][esp32-hal-psram.c:82] psramInit(): [arduino-psram] PSRAM enabled
[   486][I][esp32-hal-psram.c:100] psramAddToHeap(): PSRAM added to the heap.
```
or not crashing .. but stopping after few words .. and this appears example):
```
nfo:   AACDecoder has been initialized, free Heap: 133592 bytes , free stack 3520 DWORDs
info:   Webfile: stream ready, buffer filled in 6 ms
info:   syncword found at pos 0
info:   Channels: 1
info:   SampleRate: 24000
info:   BitsPerSample: 16
info:   BitRate: 58687
info:   AAC HeaderFormat: ADTS
[  8991][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  9000][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  9010][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  9012][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  9027][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
...
[ 12243][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12252][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12261][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12270][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12276][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12289][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12298][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12306][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12316][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12325][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12334][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 12336][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
info:   End of webstream: "api.openai.com"
```

looks like there is any pointer /zero position / heap issue now, maybe you missed any tiny detail ?
Take care you time, i don not want to bore you with my issue !

Btw: i wished you would create an OpenAI API key (there are no costs! bc. the 10$ free preload is enough for many many months. Reason i mention: Your openai.speech() is awesome .. (in opposite to Google TTS: 1. no TTS length limit, 2. multilingual, menas no boring country parameter and multiple languages can mixed in one sentence !, 3. many human similar voices etc. etc). I do not want to advertise, just as info for you ;)

Thanks !! .. let me know if i should test anything more for you .. or any debug log_xy toggle needed ?

schreibfaul1, 2025-04-28:
Ooh, I tested with my Apache2 server on the home network and that's unencrypted.
I also have the error with SSL connections.
I cannot query the `SecureClient` from an external task, this leads to a crash.
Now I have used a flag as a semaphore, hopefully it will now also work with "https".

kaloprojects, 2025-04-29:
Hi Wolle, @schreibfaul1 

.. as mentioned earlier: **You are a MAGIC MASTER 💫 ! ... it works perfect now !!! :)**  👍 
Awesome, it works best ever .. even the response time (latency from call to audio start) feels better (no idea why) .. and the _.isRunning()_ stops immediately to false when Audio is done. Audio output works flawless, Fantastic !

You had to spend some effort in this .. so much thankful for your work Wolle, other folks will love this improvement for sure, the updated function **openai_speech() works best now**. 

Huge **THANK YOU Wolle !**, i will update my github [project](https://github.com/kaloprojects/KALO-ESP32-Voice-ChatGPT) code (using @akdeb additional instruction parameter update and a hint to the folks in my README to download YOUR latest zip library (3.1.0w - April 28, 2025 or later). And hey .. as mentioned, you should request an own Open AI API key for you, you will love this awesome TTS service with your function ;).

Giving you a final hint, not sure of interest .. with Debug Info activated there are some messages (not sure what 'fail on 0, errno: 9, "Bad file number' means). Posting this only in case you need details. It is _no issue at all_, as mentioned Audio streaming works perfect already, no noticeable issues any longer (and messages disappear if debug mode off)
```
info:   AACDecoder has been initialized, free Heap: 133316 bytes , free stack 3688 DWORDs
info:   Webfile: stream ready, buffer filled in 5 ms
info:   syncword found at pos 0
info:   Channels: 1
info:   SampleRate: 24000
info:   BitsPerSample: 16
info:   BitRate: 57750
info:   AAC HeaderFormat: ADTS
[  7549][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  7558][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  7566][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  7575][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  7584][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  7593][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"...
[ 11496][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 11505][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[ 11513][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
info:   End of webstream: "api.openai.com"
```
Wolle @schreibfaul1 : you made my day !, and the ESP32 TTS folks will love it for sure too.
Thank You ! (also in behalf of many other folks)

Kalo :blush:

schreibfaul1, 2025-04-29:
Hi Kalo,
the problem is that OpenAi does not provide the length of the payload. The payload is complete when OpenAi closes the TCP connection. 
The clientSecure no longer exists when the connection is closed. This causes the error message.
I have submitted a ticket to the experts, perhaps we will know more afterwards.

WiFiClientSecure, Object destroys itself even after the server separates the connection #11325 
https://github.com/espressif/arduino-esp32/issues/11325

Until then, I use a workaround to suppress the error message.

kaloprojects, 2025-04-29:
Got it, this makes sense .. so let's see their feedback. And for your AUDIO..H user it is not a problem at all, _openai.speech() works great for the user now already_ ! 😊 

btw.. posting here to you .. because i _also planned to ping the Espressif /arduino-esp32 folks_, as i found a issue yestarday (i would even say 'bug') with WifiClientSecure function **client.write**( const uint8_t *buf,  size_t size )

Background: Meanwhile i record my I2S audio into PSRAM (creating proper wav header etc.), sending this to Deepgram SST Server (via WifiClientSecure connect). The issue i found: It is NOT possible to send content of a buffer with more than (exactly) 16383 bytes (funny, no idea why they limit to a 14 bit value). As soon you send more it is skipped, and the STT server waiting for the rest forever. My workaround now: I break down my audio (typically 100 - 200 KB audio) into smaller chunks :|  .. urgly workaround:

```
  size_t blocksize =   16383;   // 14 value seems max. limit !!
  size_t chunk_start = 0;
  size_t result_bytes_sent = 0;
  
  while ( chunk_start < PSRAM_BUFFER_counter ) 
  { 
    // reduce remaining blocksize on final rest
    if ( (chunk_start + blocksize) > PSRAM_BUFFER_counter )   
    {  blocksize = PSRAM_BUFFER_counter - chunk_start;   // remaining rest
    }
    result_bytes_sent = client.write( PSRAM_BUFFER + chunk_start, blocksize );
    Serial.println( "Blocksize: " + (String) blocksize );  
    Serial.println( "Result:    " + (String) result_bytes_sent );  
    chunk_start += blocksize;
  }
```

This is a bug in my opinion, _makes no sense and a pain as it is time consuming_ (each write spend time in buffer cleanup and synchronizing). When i look to current ESP32 lib, the size is not limited: [.cpp/h](https://github.com/espressif/arduino-esp32/blob/master/libraries/NetworkClientSecure/src/NetworkClientSecure.h) for NetworkClientSecure (alias WifiClientSecure)i do not see any limits (bytes with size_t)

Are you aware that the buffersize in write is limited ? Or do i miss anything ?

schreibfaul1, 2025-04-30:
`fail on 0, errno: 9, "Bad file number` should not appear even if the TCP connection has been closed. I have reported this unusual message to Espressif.

The event:
````c++
void audio_eof_stream(const char* info) {
    .....
}
````
is called.

MrNaif2018, 2025-05-02:
Hi! I am not 100% sure that's the case, but I think some of the last commits like 69abb69ee821824d4f498aaf8b80dbf7fc0f5d0a and eb2026eacc942b696a45d0e09f8c3cf1d41594d3 may have regressed playing mp3 from a remote URL.
I was getting similar issues as in #1014 pretty frequently, but the reporter closed the issue because they "edited buffers". I tried editing buffers myself (DMA buffers) as well but it didn't help. But reverting back 6 commits to ae5f749d72d775019769bf21e1a300a29f16a9e5 and then changing DMA buffer size worked.
I just re-tested and without changing DMA buffer sizes it has occasional issues but more rare than with the last commits
So looks like there is a regression in the last 6 commits, probably the ones I've shared at the top of the comment which affects stream URL loading

For now I will use the following params:
https://github.com/MrNaif2018/ESP32-audioI2S/commit/83ff1fc57d9f9e058030786fdeb93fb63f554c85

closed 2025-06-16 by github-actions[bot]