schreibfaul1/ESP32-audioI2S #1024: Example: Simple webradio and TFT libraries are out of sync.

amishv, 2025-05-05:
in the webradio example the TFT call has 5 parameters

`tft.begin(TFT_CS, TFT_DC, SPI_MOSI, SPI_MISO, SPI_SCK);`

while the library "https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9486" requires six parameters

` tft.begin(TFT_CS, TFT_DC,/*VSPI*/ 2, SPI_MOSI, SPI_MISO, SPI_SCK);`

schreibfaul1, 2025-05-05:
Over time, several updates were necessary.

amishv, 2025-05-11:
Thanks for making the changes, though I cannot verify the .ino files but the changes are consistent with ones I made including in the function `tp_released()` that was causing an extra button with non existent array element. 

Also, the TFT and TP in the library ESP32-TFT-Library-ILI9486 seems to have a different orientation. I will raise an issue in that repository after root causing.

Thanks again for quick turnaround.

schreibfaul1, 2025-05-11:
The displays are not all identical.
This depends on the manufacturer.
`    tft.setRotation(3);
    tp.setRotation(3);`

Maybe it works: ` tp.setRotation(1);`

amishv, 2025-05-11:
looks like my screen oriented x=480 and y=320 while the  screen used in code seems to have x=320 and y=480.

Will make the changes in the local copy only.

Thanks for the help.

closed 2025-06-11 by amishv