schreibfaul1/ESP32-audioI2S #1026: Incorrect reading of broadcast title.

hevet, 2025-05-07:
On the radio about the address
http://stream.radiobaobab.pl:8000/radiobaobab.mp3
does not read the broadcast information correctly. It only reads
"Jan Chojnacki - Bielszy odcie&#324"
In foobar it reads the entire title of the broadcast, but also with the error @#324
"Jan Chojnacki - Bielszy odcie&#324 bluesa cz.1".
It seems that after reading the html character it breaks off the title.

schreibfaul1, 2025-05-07:
Hallo @hevet,
the sequence "&amp;#nnn" or "&amp;#nnnn" can actually be an HTML character.
Maybe I need a routine to convert to Unicode. I'll have a look at that.

schreibfaul1, 2025-05-07:
The stream titles are now searched for HTML characters and replaced with UTF-8 characters if necessary.

[image]

hevet, 2025-05-07:
@schreibfaul1 Great, works beautifully. Thank you.

hevet, 2025-05-15:
@schreibfaul1 There is another error, sometimes in titles it displays &amp; and also breaks off further text. Can you make a correction for this too?

e.g. it is Jack&amp;
and it should be Jack&Kate.

schreibfaul1, 2025-05-15:
There are actually other HTML characters, apart from those with 3 digits `&#nnn;`

These are:

[image]

and then `StreamTitle='Jack &amp; Kate';` would be displayed incorrectly.

I hope the error has now been rectified.