schreibfaul1/ESP32-audioI2S #1027: Important note

Dcmwrnd, 2025-05-08:
Hi schreibfaul1, this is a wonderful audio library for the ESP32.

My last ESP32-audioI2S library update was v3.1.0s on Arduino IDE v2.3.5 with Board Manager v3.1.3 on ESP32 Dev Module setting using ‘audio.connecttospeech’ and was working perfectly.

Recently I updated the ESP32-audioI2S library to v3.1.0y and compile my sketch again which is not modified. After the audio speech finished, the ESP32 hang. It did not trigger the ‘void audio_eof_speech(const char *info)’ after the speech ended.

Here is another problem, when I revert the audio library back to v3.1.0s, the problem still persist, EP32 hang. This is very strange because it was working before.

The problem is on the IDE, when compile the first time it will stored all the included libraries version number on the cache folder. Then I manually update the latest library but the ‘library.properties’ and ‘library.json’ file version number did not change and the IDE assume that there is no changes on the library. So when compile the second time the IDE use the library from the cache folder to save time. It is always faster when compile the second time. So, make sure you guys delete all the cache files after every library update manually before compile.

My cache folder is at:-
C:\Users\user name\AppData\Local\arduino\sketches\xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

Have to manually delete all the ‘xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx’ folders inside the …\sketches\ than re-compile your sketch and everything works again.

Regards,
David.

schreibfaul1, 2025-05-08:
Hello David,
thank you for your experience report with the Arduino IDE.
I would have thought that it would be enough to install the zip file with the board manager.

Alternatively, you could install VSCode with PlatformIO. Then you can simply clone the repository:
https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF/blob/master/platformio.ini

Regards,
Wolle

eyestoreoye, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

fivemeyestoreserver, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

ayestoro, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

eyestorefivemstore, 2025-05-10:
I'm impressed by the modularity of this codebase. Makes it very maintainable.

closed 2025-07-13 by github-actions[bot]