schreibfaul1/ESP32-audioI2S #1028: Error when compiling

mecukuryurt, 2025-05-11:
Hello, I was trying to play audio from a web radio stream. This is my code:
```cpp
#include "Arduino.h"
#include <WiFi.h>
#include <SD.h>
#include "Audio.h"

#define I2S_DOUT     14
#define I2S_BCLK      13
#define I2S_LRC        12
Audio audio;
String ssid =    "****";
String password = "****";
void setup() {
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid.c_str(), password.c_str());
  while (WiFi.status() != WL_CONNECTED)
  delay(1500);
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(100);
  audio.connecttohost("https://stream.radyo45lik.com:4545/stream");
}
void loop()
{
  audio.loop();
}
```
When I tried to compile my code, this error appeared:
```
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In constructor 'Audio::Audio(uint8_t)':
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:183:5: error: 'm_i2s_chan_cfg' was not declared in this scope
     m_i2s_chan_cfg.id            = (i2s_port_t)m_i2s_num;  // I2S_NUM_AUTO, I2S_NUM_0, I2S_NUM_1
     ^~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:183:5: note: suggested alternative: 'm_i2s_config'
     m_i2s_chan_cfg.id            = (i2s_port_t)m_i2s_num;  // I2S_NUM_AUTO, I2S_NUM_0, I2S_NUM_1
     ^~~~~~~~~~~~~~
     m_i2s_config
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:184:36: error: 'I2S_ROLE_MASTER' was not declared in this scope
     m_i2s_chan_cfg.role          = I2S_ROLE_MASTER;        // I2S controller master role, bclk and lrc signal will be set to output
                                    ^~~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:184:36: note: suggested alternative: 'I2S_MODE_MASTER'
     m_i2s_chan_cfg.role          = I2S_ROLE_MASTER;        // I2S controller master role, bclk and lrc signal will be set to output
                                    ^~~~~~~~~~~~~~~
                                    I2S_MODE_MASTER
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:188:39: error: 'm_i2s_tx_handle' was not declared in this scope
     i2s_new_channel(&m_i2s_chan_cfg, &m_i2s_tx_handle, NULL);
                                       ^~~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:188:39: note: suggested alternative: 'i2s_isr_handle_t'
     i2s_new_channel(&m_i2s_chan_cfg, &m_i2s_tx_handle, NULL);
                                       ^~~~~~~~~~~~~~~
                                       i2s_isr_handle_t
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:188:5: error: 'i2s_new_channel' was not declared in this scope
     i2s_new_channel(&m_i2s_chan_cfg, &m_i2s_tx_handle, NULL);
     ^~~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:188:5: note: suggested alternative: 'i2s_channel_t'
     i2s_new_channel(&m_i2s_chan_cfg, &m_i2s_tx_handle, NULL);
     ^~~~~~~~~~~~~~~
     i2s_channel_t
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:190:5: error: 'm_i2s_std_cfg' was not declared in this scope
     m_i2s_std_cfg.slot_cfg                = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO); // Set to enable bit shift in Philips mode
     ^~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:190:5: note: suggested alternative: 'm_i2s_config'
     m_i2s_std_cfg.slot_cfg                = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO); // Set to enable bit shift in Philips mode
     ^~~~~~~~~~~~~
     m_i2s_config
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:190:81: error: 'I2S_DATA_BIT_WIDTH_16BIT' was not declared in this scope
     m_i2s_std_cfg.slot_cfg                = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO); // Set to enable bit shift in Philips mode
                                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:190:81: note: suggested alternative: 'I2S_BITS_PER_CHAN_16BIT'
[... 311 lines cut ...]
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4903:53: error: 'm_i2s_std_cfg' was not declared in this scope
     i2s_channel_reconfig_std_slot(m_i2s_tx_handle, &m_i2s_std_cfg.slot_cfg);
                                                     ^~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4903:53: note: suggested alternative: 'm_i2s_config'
     i2s_channel_reconfig_std_slot(m_i2s_tx_handle, &m_i2s_std_cfg.slot_cfg);
                                                     ^~~~~~~~~~~~~
                                                     m_i2s_config
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4903:5: error: 'i2s_channel_reconfig_std_slot' was not declared in this scope
     i2s_channel_reconfig_std_slot(m_i2s_tx_handle, &m_i2s_std_cfg.slot_cfg);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4903:5: note: suggested alternative: 'i2s_channel_fmt_t'
     i2s_channel_reconfig_std_slot(m_i2s_tx_handle, &m_i2s_std_cfg.slot_cfg);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     i2s_channel_fmt_t
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4904:5: error: 'i2s_channel_enable' was not declared in this scope
     i2s_channel_enable(m_i2s_tx_handle);
     ^~~~~~~~~~~~~~~~~~
C:\Program Files (x86)\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4904:5: note: suggested alternative: 'm_f_channelEnabled'
     i2s_channel_enable(m_i2s_tx_handle);
     ^~~~~~~~~~~~~~~~~~
     m_f_channelEnabled

```
Can someone help me about this issue?
Thank you.

mecukuryurt, 2025-05-11:
I downgraded library version to 2.0.0 and it works now

schreibfaul1, 2025-05-11:
The compiler errors relate to I2S. Espressif has changed a lot in the I2S area since IDF V5.0. Please use a Arduino version V3.xx or higher