schreibfaul1/ESP32-audioI2S #1038: http://bestof80s.stream.laut.fm/best_of_80s seems to get out of sync

seife, 2025-05-23:
The http://bestof80s.stream.laut.fm/best_of_80s stream (mp3) does seem to get "out of sync".
Sometimes it just glitches from time to time, bu tsometimes it goes completely out of sync and ends in a constant stream of MP3 decode errors and "syncword found" sequences or even stops playing completely.

Other MP3 streams work without issues (e.g. http://www.dradio.de/streaming/dlf.m3u)
```
audio not running => connect!
00:00:02>info        ==> PSRAM found, inputBufferSize: 630773 bytes
00:00:02>info        ==> buffers freed, free Heap: 164164 bytes
00:00:02>info        ==> connect to: "bestof80s.stream.laut.fm" on port 80 path "/best_of_80s"
[  2606][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[  2676][D][NetworkManager.cpp:127] hostByName(): DNS found IPv4 80.255.8.200
00:00:02>info        ==> Connection has been established in 109 ms, free Heap: 163064 bytes
00:00:02>info        ==> chunked data transfer
00:00:02>station     Best of 80s
00:00:02>icyurl      http://laut.fm/best_of_80s
00:00:02>bitrate     128000
00:00:02>info        ==> MP3Decoder has been initialized, free Heap: 132116 bytes , free stack 5788 DWORDs
00:00:02>lasthost    http://bestof80s.stream.laut.fm/best_of_80s
00:00:02>info        ==> stream ready
00:00:02>info        ==> syncword found at pos 220
00:00:02>info        ==> syncword found at pos 0
00:00:02>info        ==> MPEG-2.5, Layer I
00:00:02>info        ==> Channels: 2
00:00:02>info        ==> SampleRate: 44100
00:00:02>info        ==> BitsPerSample: 16
00:00:02>info        ==> BitRate: 128000
00:00:02>info        ==> StreamTitle='www.edeka.de - Ein bisschen Verbraucherinformationen ...'
00:00:03>info        ==> StreamTitle='www.edeka.de - Ein bisschen Verbraucherinformationen ....'
00:00:03>info        ==> StreamTitle='www.edeka.de - Ein bisschen Verbraucherinformationen .....'
00:00:03>info        ==> StreamTitle='www.edeka.de - Ein bisschen Verbraucherinformationen ......'
00:00:03>info        ==> StreamTitle='www.edeka.de - ...  schon fast die Hälfte geschafft ...'
00:00:03>info        ==> StreamTitle='www.edeka.de - ...  zweite Halbzeit ...'
00:00:03>info        ==> StreamTitle='www.edeka.de - ...  gleich geht's los mit laut.fm/best_of_80s ...'
00:00:04>info        ==> StreamTitle='Die Doraus & die Marinas - Kleines Stubenmädchen'
00:00:23>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
00:00:23>info        ==> syncword found at pos 62
00:00:23>info        ==> syncword found at pos 0
[ 23026][D][mp3_decoder.cpp:1708] DecodeHuffmanPairs(): (tabType != invalidTab)
00:00:23>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
00:00:23>info        ==> syncword found at pos 204
00:00:23>info        ==> syncword found at pos 0
[ 23047][D][mp3_decoder.cpp:1708] DecodeHuffmanPairs(): (tabType != invalidTab)
00:00:23>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
00:00:23>info        ==> syncword found at pos 149
00:00:23>info        ==> syncword found at pos 0
00:00:23>info        ==> MPEG-2.5, Layer I
00:00:23>info        ==> Channels: 2
00:00:23>info        ==> SampleRate: 44100
00:00:23>info        ==> BitsPerSample: 16
00:00:23>info        ==> BitRate: 128000
00:00:23>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
[ 23093][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
[ 23105][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
00:00:23>info        ==> syncword found at pos 417
00:00:23>info        ==> syncword found at pos 0
00:00:23>info        ==> MPEG-2.5, Layer I
00:00:23>info        ==> Channels: 2
00:00:23>info        ==> SampleRate: 44100
00:00:23>info        ==> BitsPerSample: 16
00:00:23>info        ==> BitRate: 128000
00:00:42>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
00:00:42>info        ==> syncword found at pos 92
00:00:42>info        ==> syncword found at pos 0
00:00:42>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
[ 42370][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
00:00:42>info        ==> syncword found at pos 324
00:00:42>info        ==> syncword found at pos 0
00:00:42>info        ==> MPEG-2.5, Layer I
00:00:42>info        ==> Channels: 2
00:00:42>info        ==> SampleRate: 44100
00:00:42>info        ==> BitsPerSample: 16
00:00:42>info        ==> BitRate: 128000
```

seife, 2025-05-23:
I tested both with git master (`7e7b452`) and version 3.2.0 (`e008d56`), made no obvious difference.

schreibfaul1, 2025-05-24:
I'll have a look at that. It looks as if the error only occurs within a commercial break.

schreibfaul1, 2025-05-24:
Hi @seife, 
this has nothing to do with the adverts. Apparently there is a short pause between the chunks. When querying the chunk length, the WiFi client then returns -1.
I hope this no longer happens.

seife, 2025-05-24:
It is better but not totally fixed, I'm still getting:
```
00:00:02>info        ==> MP3Decoder has been initialized, free Heap: 131984 bytes , free stack 5788 DWORDs
00:00:02>lasthost    http://bestof80s.stream.laut.fm/best_of_80s
00:00:02>info        ==> stream ready
00:00:02>info        ==> syncword found at pos 0
00:00:02>info        ==> MPEG-2.5, Layer I
00:00:02>info        ==> Channels: 2
00:00:02>info        ==> SampleRate: 44100
00:00:02>info        ==> BitsPerSample: 16
00:00:02>info        ==> BitRate: 128000
00:00:02>info        ==> StreamTitle='Wham! - Freedom'
/control. Args:
00:02:00>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
[120859][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
00:02:00>info        ==> syncword found at pos 303
00:02:00>info        ==> syncword found at pos 0
00:02:00>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
00:02:00>info        ==> syncword found at pos 0
00:02:00>info        ==> MP3 decode error -6 : INVALID_FRAMEHEADER
[120893][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
00:02:00>info        ==> syncword found at pos 112
00:02:00>info        ==> syncword found at pos 0
00:02:00>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
00:02:00>info        ==> syncword found at pos 92
00:02:00>info        ==> syncword found at pos 0
00:02:00>info        ==> MP3 decode error -6 : INVALID_FRAMEHEADER
[120924][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
00:02:00>info        ==> syncword found at pos 116
00:02:00>info        ==> syncword found at pos 0
00:02:00>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
00:02:00>info        ==> syncword found at pos 77
00:02:00>info        ==> syncword found at pos 0
00:02:00>info        ==> MP3 decode error -9 : INVALID_HUFFCODES
[120962][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
[120973][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
00:02:00>info        ==> syncword found at pos 224
00:02:00>info        ==> syncword found at pos 0
00:02:00>info        ==> MP3 decode error -8 : INVALID_SCALEFACT
[120983][D][mp3_decoder.cpp:1219] MP3FindSyncWord(): wrong sampling rate
...
```
which eventually synchronized again:
```
[121319][D][mp3_decoder.cpp:1209] MP3FindSyncWord(): wrong bitrate index
00:02:01>info        ==> syncword found at pos 13
00:02:01>info        ==> syncword found at pos 0
00:02:01>info        ==> MPEG-2.5, Layer I
00:02:01>info        ==> Channels: 2
00:02:01>info        ==> SampleRate: 44100
00:02:01>info        ==> BitsPerSample: 16
00:02:01>info        ==> BitRate: 128000
00:04:51>info        ==> StreamTitle='Matt Bianco - More Than I Can Bear'
```
Much later it then failed compeletely (playback stopped):
```
00:19:31>info        ==> StreamTitle='Pat Benatar - We Belong'
00:19:45>info        ==> ID3 tag found, skip 45 bytes
00:23:14>info        ==> StreamTitle='Tina B. - Nothin's' Gonna Come Easy'
00:23:28>info        ==> ID3 tag found, skip 45 bytes
[1416742][W][Audio.cpp:3405] processWebStream(): chunk count error, expected: 0x0A, received: 0xFFFFFFFF
[1416755][W][Audio.cpp:3404] processWebStream(): chunk count error, expected: 0x0D, received: 0x84
[1416763][W][Audio.cpp:3405] processWebStream(): chunk count error, expected: 0x0A, received: 0x02
```
where the "14157xx" timestamp corresponds to about 00:23:36 so it is some seconds after the "ID3 tag found"

This is after fix commit number 2 already.

Now that I have an idea where to look, I'll also poke at it once I get some time.

seife, 2025-05-25:
I took a look and it seems it has to do with the input buffer never filling completely, and then something going wrong in chunked mode reading when no bytes are in `_client->available()`.
I did, however, not find the root cause yet.
It does not seem to happen if I do a `audio.setBufferSize(64*1024 + 4096*6);` to set the buffersize to 64k, which then gets filled completely. But I have not done long time tests yet.

schreibfaul1, 2025-05-25:
http://bestof80s.stream.laut.fm/best_of_80s is complicated and differs greatly from the "normal" streams.
Each new music track has its own ID3 header. I can now recognise this and can skip it.The ID3 header is usually only a few bytes long. During advertising it is larger and probably contains an image?

[image]

Something else comes up at the start, I still have no idea what it is ??????

The input buffer is rarely filled completely, this depends on the settings in the server. 

But the chunks and metadata are processed correctly. There will no longer be a crash.

seife, 2025-05-25:
I still had it "locking up" in the metadata parsing; Audio::readMetadata().
It got called with "maxBytes = 1", because 95 of 96 metadata bytes were already read (metalen = 96; pos_ml = 95), but then it did only read 0 bytes, because of the "maxBytes - 1" in https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L5888 and never progressed.

I have now changed this to
```
     if(metalen < m_chbufSize) {
-        uint16_t a = _client->readBytes(&m_chbuf[pos_ml], min((uint16_t)(metalen - pos_ml), (uint16_t)(maxBytes - 1)));
+        uint16_t a = _client->readBytes(&m_chbuf[pos_ml], min((uint16_t)(metalen - pos_ml), (uint16_t)(maxBytes)));
         res += a;
         pos_ml += a;
     }
```

This is running now in a longterm test, it survived already longer than before.
Note this obviously affects all streams on https://laut.fm/, so you can choose another one which suits your taste better for testing ;-)

schreibfaul1, 2025-05-25:
This may be the cause of the problem. This station transmits metadata with a size of 1 byte. And with `maxBytes - 1` nothing is read at all. I have adopted your suggestion together with other changes.

Best regards
Wolle

seife, 2025-05-26:
This is much better now -- it ran for more than 8 hours, then failed with a WiFi-Problem but that's quite surely totally unrelated ;-)

seife, 2025-05-27:
Actually, the laut.fm streams are apparently not 100% correct, they contain errors when splicing in the advertisements or maybe even when changing songs.

I get this when playing them with ffplay:
```
Input #0, mp3, from 'http://stream.laut.fm/1000oldies':  0B f=0/0   
  Metadata:
    artist          : Verbraucherinformationen
    track           : 1
    date            : 2025
    comment         : br-128_sr-44100
    album           : "laut_transcoder__v0.1.0__2025-05-25_22:10:12
    title           : lautfm
    encoder         : Lavf60.16.100
    icy-name        : 1000 Oldies
    icy-url         : http://laut.fm/1000oldies
    icy-description : Der Soundtrack unseres Lebens. 60er 70er 80er 90er.
    icy-genre       : Rock
    icy-pub         : 1
    icy-sr          : 44100
    icy-audio-info  : ice-channels=2;ice-samplerate=44100;ice-bitrate=128
    icy-br          : 128
    StreamTitle     : www.edeka.de - Ein bisschen Verbraucherinformationen ...
    StreamUrl       : 0
  Duration: N/A, start: 0.000000, bitrate: 128 kb/s
  Stream #0:0: Audio: mp3, 44100 Hz, stereo, fltp, 128 kb/s
[mp3float @ 0x7f0dec010300] overread, skip -6 enddists: -5 -5=0/0   
[mp3float @ 0x7f0dec010300] Header missing    0KB sq=    0B f=0/0   
[mp3float @ 0x7f0dec010300] Header missing    0KB sq=    0B f=0/0   
[mp3float @ 0x7f0dec010300] Header missing    0KB sq=    0B f=0/0   
[mp3float @ 0x7f0dec010300] Header missing    0KB sq=    0B f=0/0   
[mp3float @ 0x7f0dec010300] overread, skip -6 enddists: -5 -5
[mp3float @ 0x7f0dec010300] Header missing    0KB sq=    0B f=0/0   
```
and with the "mpv" player:
```
> mpv http://stream.laut.fm/1000oldies
● Audio  --aid=1  (mp3 2ch 44100 Hz 128 kbps)
AO: [pipewire] 44100Hz stereo 2ch floatp
File tags:
 icy-title: Clout - Substitute
[ffmpeg/audio] mp3float: Header missing
Error decoding audio.
File tags:
 icy-title: 1000 Oldies - Spenden 2021
[ffmpeg/audio] mp3float: Header missing
Error decoding audio.
File tags:
 icy-title: 1000 Oldies - Web
[ffmpeg/audio] mp3float: Header missing
Error decoding audio.
File tags:
 icy-title: Hot Chocolate - Girl Crazy
[ffmpeg/audio] mp3float: Header missing
Error decoding audio.
File tags:
 icy-title: www.edeka.de - Ein bisschen Verbraucherinformationen ...
File tags:
 icy-title: www.edeka.de - Ein bisschen Verbraucherinformationen ....
File tags:
 icy-title: www.edeka.de - Ein bisschen Verbraucherinformationen .....
File tags:
 icy-title: www.edeka.de - Ein bisschen Verbraucherinformationen ......
[ffmpeg/audio] mp3float: Header missing
Error decoding audio.
File tags:
 icy-title: Beatles - Here Comes The Sun
[ffmpeg/audio] mp3float: Header missing
Error decoding audio.
File tags:
 icy-title: Daniel Boone - Beautiful Sunday
A: 00:07:42 / 00:07:57 (97%) Cache: 14s/531KB
Exiting... (Quit)
```
So it looks like something is not 100% correct in their streaming software.

The good thing: ESP32-audioI2S is handing this fine now, maybe these streams can serve as a test case for its error recovery capabilities ;-)

closed 2025-07-11 by github-actions[bot]