schreibfaul1/ESP32-audioI2S #1040: ESP32 reboot using ‘audio.connecttohost’

Dcmwrnd, 2025-05-25:
Hi to all,

For you guys out there, if you are using ‘audio.connecttohost’ for internet radio on your ESP32 4MB Dev Module without PSRAM and reboot itself very often or once in a while like every hour or hours. Try this out on your Arduino IDE:-

void loop() {
    audio.loop();
    vTaskDelay(1);
}

Set the ‘vTaskDelay();’ between 3 and 20. Currently I’m using ‘vTaskDelay(10);’ and voila! The ESP32 did not reboot the whole day. I even tested up to ‘vTaskDelay(30);’ and it still works.

Regards,
David

enjoyneering, 2025-05-28:
Have you tried increasing the connection timeout?

https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.h#L146C10-L146C76

Dcmwrnd, 2025-05-29:
Thanks for your recommendation and I will give it a try and see what happen.
Regards.

closed 2025-07-13 by github-actions[bot]