schreibfaul1/ESP32-audioI2S #1045: [BUG]

tv4you2016, 2025-05-29:
Hello,
I have been listening to a radio station (https://stream-icy.bauermedia.pt/cidade.mp3) for around 2 consecutive days without turning off my device, today on the 2nd I came across this error: 
 SSL - Memory allocation failed, from now on I cannot play any station that is https 
I use a esp 32-wroverie with psram.

```
[0;37m[86164028][V][ssl_client.cpp:364] stop_ssl_socket(): Cleaning SSL connection.[0m
[0;31m[86164038][E][AudioPlayer.h:169] audio_info(): Audio Info: buffers freed, free Heap: 66908 bytes
[0m
[0;31m[86164048][E][AudioPlayer.h:169] audio_info(): Audio Info: connect to: "stream-icy.bauermedia.pt" on port 443 path "/cidade.mp3"
[0m
[0;36m[86164123][D][NetworkManager.cpp:127] hostByName(): DNS found IPv4 91.193.245.115[0m
[0;37m[86164133][V][ssl_client.cpp:68] start_ssl_client(): Free internal heap before TLS 67144[0m
[0;37m[86164141][V][ssl_client.cpp:75] start_ssl_client(): Starting socket (domain 2)[0m
[0;37m[86164234][V][ssl_client.cpp:170] start_ssl_client(): Seeding the random number generator[0m
[0;37m[86164245][V][ssl_client.cpp:178] start_ssl_client(): Setting up the SSL/TLS structure...[0m
[0;36m[86164254][D][ssl_client.cpp:196] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE![0m
[0;37m[86164264][V][ssl_client.cpp:292] start_ssl_client(): Setting hostname for TLS session...[0m
[0;31m[86164273][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():302]: (-32512) SSL - Memory allocation failed[0m
[0;32m[86164284][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.[0m
[0;31m[86164293][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -32512[0m
[0;37m[86164302][V][ssl_client.cpp:364] stop_ssl_socket(): Cleaning SSL connection.[0m
[0;31m[86164312][E][AudioPlayer.h:169] audio_info(): Audio Info: Request https://stream-icy.bauermedia.pt/cidade.mp3 failed!
[0m
[0;31m[86164324][E][AudioPlayer.h:162] audio_showstreamtitle(): streamtitle :  [0m
```

[MY_sdkconfig .txt](https://github.com/user-attachments/files/20501033/MY_sdkconfig.txt)

schreibfaul1, 2025-05-30:
MBEDTLS (SSL) consumes between 40 and 50KB. It is better to move this to PSRAM.

[image]

tv4you2016, 2025-06-05:
@schreibfaul1 
Thanks for the tip.. 
To date, it is working correctly

closed 2025-06-05 by tv4you2016