schreibfaul1/ESP32-audioI2S #1046: ESP32-WROOM crashing at the start of music playback

InviseDivine, 2025-05-29:
Hello schreibfaul1,
I'm trying to use PCM5102a with ESP32-WROOM and I'm getting an ESP error.  Tried playing music stream and wav file from link, also i work in Arduino IDE
``` 
info        audioI2S Version 3.2.0
info        PSRAM not found, inputBufferSize: 13951 bytes
info        buffers freed, free Heap: 171124 bytes
info        connect to: "s2-webradio.rockantenne.de" on port 443 path "/alternative/stream/mp3"
info        SSL has been established in 560 ms, free Heap: 129584 bytes
bitrate     128000
station     ROCK ANTENNE Alternative
icyurl      https://www.rockantenne.de
info        MP3Decoder has been initialized, free Heap: 101100 bytes , free stack 3976 DWORDs
lasthost    https://s2-webradio.rockantenne.de/alternative/stream/mp3
info        stream ready
info        syncword found at pos 0
info        MPEG-2.5, Layer I
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008f8c1  PS      : 0x00060f33  A0      : 0x8008d5fd  A1      : 0x3ffc4cb0  
A2      : 0x3ffc80c0  A3      : 0x3ffc4f04  A4      : 0xb33fffff  A5      : 0x3f430d70  
A6      : 0x000000fd  A7      : 0x3ffb6330  A8      : 0x3ffb818c  A9      : 0x00000000  
A10     : 0x0000104b  A11     : 0x00000000  A12     : 0x3ffb4fac  A13     : 0x3f430df0  
A14     : 0xffffffff  A15     : 0x3ffaf3d4  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40089e4c  LEND    : 0x40089e62  LCOUNT  : 0x00000000  

Backtrace: 0x4008f8be:0x3ffc4cb0 0x4008d5fa:0x3ffc4cd0 0x4008de64:0x3ffc4cf0 0x4008d106:0x3ffc4d10 0x4010790f:0x3ffc4d50 0x40081573:0x3ffc4d90 0x400dca66:0x3ffc4dd0 0x400dcfa6:0x3ffc4e30 0x400dd159:0x3ffc4e50 0x4008f046:0x3ffc4e70
```

schreibfaul1, 2025-05-30:
Difficult to say, LoadProbibited indicates a memory problem. PSRAM should be cativised for SSL, the internal memory is usually not sufficient.

InviseDivine, 2025-05-30:
Thank you for reply, i tried to use streaming without SSL and one time I had an error ```assert failed: spinlock_release spinlock.h:160 (core_id == lock->owner)```, now im getting this error 
```info        audioI2S Version 3.2.0
info        PSRAM not found, inputBufferSize: 13951 bytes
info        buffers freed, free Heap: 171096 bytes
info        connect to: "19273.live.streamtheworld.com" on port 80 path "/2CA.mp3"
info        Connection has been established in 3093 ms, free Heap: 169488 bytes
icyurl      http://2ca.com.au
bitrate     64000
info        MP3Decoder has been initialized, free Heap: 140340 bytes , free stack 6288 DWORDs
lasthost    http://19273.live.streamtheworld.com/2CA.mp3
info        stream ready
info        syncword found at pos 0
info        MPEG-1, Layer I
info        Channels: 2
info        SampleRate: 22050
info        BitsPerSample: 16
info        BitRate: 64000
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008f995  PS      : 0x00060933  A0      : 0x8008d6d1  A1      : 0x3ffc4df0  
A2      : 0x3ffc80c0  A3      : 0x3ffc4f04  A4      : 0xb33fffff  A5      : 0x3f435c2c  
A6      : 0x00000240  A7      : 0x3ffcdd08  A8      : 0x3ffb818c  A9      : 0x00000000  
A10     : 0x00002d58  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00060923  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40089f20  LEND    : 0x40089f36  LCOUNT  : 0x00000000  

Backtrace: 0x4008f992:0x3ffc4df0 0x4008d6ce:0x3ffc4e10 0x4008e69f:0x3ffc4e30 0x400df08d:0x3ffc4e50 0x4008f11a:0x3ffc4e70
```

closed 2025-07-14 by github-actions[bot]