schreibfaul1/ESP32-audioI2S #1048: Version # in json header not updated -> Arduino IDE shows 'wrong' version in Library manager

kaloprojects, 2025-05-31:
Hi @schreibfaul1 , 
found a tiny issue which makes some confusion in Arduino IDE library manager for many user.  One of my Voice assistant user have found this detail, Thanks to @chhorisberger! 

Here the issue / screenshot to Arduino IDE library manager: 

[image]

[image]

So in my examples you see **3.0.13 installed (or older 2.0.0)**, _both are wrong!_ .. in real I currently installed your latest 3.2.0i (not 3.0.13) .. and i never had a older 2.0.0..

Reason might be simple .. just seen that your file `library.json` are often not updated (inside the zips), which causes this confusion, 
2 examples:

Example 1: your latest **3.2.0i** zip has wrong entry 3.0.13

[image]

Example 2: the old **3.0.12f** zip has wrong entry 2.0.0 

[image]

I know you don't use Arduino, that' the reason i am posting for you .. so maybe double check in future your `library.json`  always, to avoid that folks are confused on there cuiurrent installed version ;)

Thx !

schreibfaul1, 2025-05-31:
That's right, I didn't bother with the library.json file. I didn't realise that this is used in the Arduino IDE and I did it manually for the last two releases.  I will develop a Python script that updates the version automatically.

kaloprojects, 2025-06-01:
Hi @schreibfaul1 .. just a thought came in my mind, you might know best if this is a good idea or not:

In my latest projects i am using function which c_hanged in syntax between versions_, best example: `audio.openai_speech(..)` 

In past (AUDIO.H version < 3.1.0u) the function had 6 parameter, since 3.1.0u it has 7 (you might remember, @akdeb added a new 4th parameter 'voice_instruction). Unfortunately without 'overloading', means the compiler produces ERROR if the code is not manually updated from user. (depending installed AUDIO.H version) 

So my question: Could it make sense if you add any #define ESP32-AUDIOI2S-MASTER  "versionxy" .. which could be checked from any sketch ... (maybe won't help because even any 'if-then' might still produce compiler errors? .. or is there any other trick to overcome the missed overloading for backwards compatibility ? 

You have deeper skills than me in coding, so i was just wondering about any ideas ;)

schreibfaul1, 2025-06-01:
The fact that the version can be read out has not been around for very long, only since 29 Apr. 25.
You could do it this way:

````c++
// Events from audioI2S library
void audio_info(const char* info) {
    if(strncmp("audioI2S Version ", info, 17) == 0) {
        log_w("audioI2SVersion: %s", info + 17);
        if(strcmp(info + 17, "3.3.0") <= 0) {
            log_w("You need at least audioI2S library version 3.3.0");
        }
        else{ log_w("> 3.0.0, Version is: %s", info + 17);}
    }
}

````

closed 2025-07-16 by github-actions[bot]