schreibfaul1/ESP32-audioI2S #1079: Exception in V3.3.2  when playing huge mp3-files

Joe91, 2025-06-25:
When playing this file, I get an exception.
This doesn't happen in v3.2.0

[Crackdown.zip](https://github.com/user-attachments/files/20908690/Crackdown.zip)

Here some logs:
```

D [44917] Freier Speicher: 115312 Bytes
N [44917] Playlist-Generierung
N [44919] Modus: Einzelner Track
N [44921] Neue Playlist mit 1 Titel(n) empfangen
D [44921] Free heap: 114172
I [44985] info        : PSRAM found, inputBufferSize: 630773 bytes
I [44996] info        : buffers freed, free Heap: 117764 bytes
I [44998] info        : Reading file: "/Crackdown.mp3"
I [45001] info        : MP3Decoder has been initialized, free Heap: 93092 bytes , free stack 5460 DWORDs
N [45017] '/Crackdown.mp3' wird abgespielt (1 von 1)
D [45085] no cover image for SD-card audio
I [45493] info        : Content-Length: 11820753
I [45493] info        : ID3 framesSize: 74268
I [45493] info        : ID3 version: 2.3
I [45500] info        : ID3 normal frames
I [45532] id3data     : Title: Crackdown
I [45562] id3data     : Artist: Marcus Intalex, Spirit
I [45587] id3data     : Album: Crackdown
I [45612] id3data     : ContentType: Dance & DJ
I [45637] id3data     : Composer: Marcus Kaye
I [45661] id3data     : Conductor: 
I [45686] id3data     : Track: 1/1
I [45712] id3data     : Year: 2021
I [45737] id3data     : Band: Marcus Intalex, Spirit
I [45787] id3data     : Copyright: Metalheadz Ltd
I [45811] id3data     : PartOfSet: 1/1
I [45944] info        : Audio-Length: 11746485
CORRUPT HEAP: Bad tail at 0x3f8df8d8. Expected 0xbaad5678 got 0x00000000

assert failed: multi_heap_free IDF\components\heap\multi_heap_poisoning.c:279 (head != NULL)

Backtrace: 0x40084f64:0x3ffd21a0 0x4009603d:0x3ffd21c0 0x40099ed9:0x3ffd21e0 0x40098bab:0x3ffd2320 0x4008563f:0x3ffd2340 0x40099f09:0x3ffd2360 0x401026ba:0x3ffd2380 0x4010346c:0x3ffd2500 0x4010399a:0x3ffd2520 0x401098c3:0x3ffd2570 0x400d3d7d:0x3ffd25a0 0x400d4d6c:0x3ffd2630 0x400ef6e4:0x3ffd2650 0x4018c78f:0x3ffd2670 0x4009640e:0x3ffd2690
  #0  0x40084f64 in panic_abort at C:/Users/Jo/.platformio/packages/framework-espidf@src-b1208ff3612f5ed4a85e916decb6f42c/components/esp_system/panic.c:454
  #1  0x4009603d in esp_system_abort at C:/Users/Jo/.platformio/packages/framework-espidf@src-b1208ff3612f5ed4a85e916decb6f42c/components/esp_system/port/esp_system_chip.c:87
  #2  0x40099ed9 in __assert_func at C:\Users\Jo\.platformio\packages\framework-espidf@src-b1208ff3612f5ed4a85e916decb6f42c\components\newlib/assert.c:80 
  #3  0x40098bab in multi_heap_free at C:\Users\Jo\.platformio\packages\framework-espidf@src-b1208ff3612f5ed4a85e916decb6f42c\components\heap/multi_heap_poisoning.c:279 (discriminator 1)
  #4  0x4008563f in heap_caps_free at C:\Users\Jo\.platformio\packages\framework-espidf@src-b1208ff3612f5ed4a85e916decb6f42c\components\heap/heap_caps_base.c:75
  #5  0x40099f09 in free at C:\Users\Jo\.platformio\packages\framework-espidf@src-b1208ff3612f5ed4a85e916decb6f42c\components\newlib/heap.c:39
  #6  0x401026ba in PsramDeleter::operator()(void*) const at .pio/libdeps/lolin_d32_pro_sdmmc_pe/ESP32-audioI2S/src/psram_unique_ptr.hpp:18
      (inlined by) std::__uniq_ptr_impl<char, PsramDeleter>::reset(char*) at C:/Users/Jo/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/unique_ptr.h:205
      (inlined by) std::unique_ptr<char, PsramDeleter>::reset(char*) at C:/Users/Jo/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/unique_ptr.h:503
      (inlined by) ps_ptr<char>::reset() at .pio/libdeps/lolin_d32_pro_sdmmc_pe/ESP32-audioI2S/src/psram_unique_ptr.hpp:951
      (inlined by) Audio::read_ID3_Header(unsigned char*, unsigned int) at .pio/libdeps/lolin_d32_pro_sdmmc_pe/ESP32-audioI2S/src/Audio.cpp:1958
  #7  0x4010346c in Audio::readAudioHeader(unsigned long) at .pio/libdeps/lolin_d32_pro_sdmmc_pe/ESP32-audioI2S/src/Audio.cpp:1255 (discriminator 1)      
  #8  0x4010399a in Audio::processLocalFile() at .pio/libdeps/lolin_d32_pro_sdmmc_pe/ESP32-audioI2S/src/Audio.cpp:3375 (discriminator 1)
  #9  0x401098c3 in Audio::loop() at .pio/libdeps/lolin_d32_pro_sdmmc_pe/ESP32-audioI2S/src/Audio.cpp:2561
  #10 0x400d3d7d in AudioPlayer_Loop() at src/AudioPlayer.cpp:872
  #11 0x400d4d6c in AudioPlayer_Cyclic() at src/AudioPlayer.cpp:246
  #12 0x400ef6e4 in loop() at src/main.cpp:234
  #13 0x4018c78f in loopTask(void*) at C:/Users/Jo/.platformio/packages/framework-arduinoespressif32-src-6738aaddd9fb901216d5f60e093380b6/cores/esp32/main.cpp:74
  #14 0x4009640e in vPortTaskWrapper at C:/Users/Jo/.platformio/packages/framework-espidf@src-b1208ff3612f5ed4a85e916decb6f42c/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```
All my other files I've tested work perfectly fine and also this file does in v3.2.0.

There is lots of heap and psram available, so it might be some memory-leak or so.

This is how the memory looks like when playing an other song:

Memory:

Free heap:           91940
Largest free block:  69620
Free PSRAM heap:     3285792
Largest PSRAM block: 3276788

schreibfaul1, 2025-06-26:
Hi Joe91,
I can reproduce the error with your file.
Originally I had a 1KB length limit for ID3Tags. Then I increased the buffer to 4KB without a length check.
But your file has a "PRIV" frame with > 8KB which caused the error. 
 There is nothing in the PRIV-TAG except "www.Amazon.com", the rest is empty.

Now it will work.

Joe91, 2025-06-26:
Your fixed workd for me. Thanks a lot!