schreibfaul1/ESP32-audioI2S #1084: Can't compile the I2Saudio example

SpectraCoder, 2025-06-27:
I'm trying to get the I2Saudio example to work on an Lolin D32 Pro, but the compiler gives me these cryptic errors. I'm unable to upload the sketch to the ESP32. Any idea how to resolve this? 

`In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32/Arduino.h:43,
                 from c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\celt.h:38,
                 from c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\opus_decoder.cpp:12:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:13:17: error: expected ',' or '...' before numeric constant
   13 | #define TF_CS   4   // TF (Micro SD Card) CS pin
      |                 ^
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:20:12: note: in expansion of macro 'TF_CS'
   20 | #define SS TF_CS
      |            ^~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.h:1031:48: note: in expansion of macro 'SS'
 1031 | void     silk_resampler_private_down_FIR(void* SS, int16_t out[], const int16_t in[], int32_t inLen);
      |                                                ^~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:13:17: error: expected ',' or '...' before numeric constant
   13 | #define TF_CS   4   // TF (Micro SD Card) CS pin
      |                 ^
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:20:12: note: in expansion of macro 'TF_CS'
   20 | #define SS TF_CS
      |            ^~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.h:1033:47: note: in expansion of macro 'SS'
 1033 | void     silk_resampler_private_IIR_FIR(void* SS, int16_t out[], const int16_t in[], int32_t inLen);
      |                                               ^~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:13:17: error: expected ',' or '...' before numeric constant
   13 | #define TF_CS   4   // TF (Micro SD Card) CS pin
      |                 ^
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:20:12: note: in expansion of macro 'TF_CS'
   20 | #define SS TF_CS
      |            ^~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.h:1035:54: note: in expansion of macro 'SS'
 1035 | void     silk_resampler_private_up2_HQ_wrapper(void* SS, int16_t* out, const int16_t* in, int32_t len);
      |                                                      ^~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32/Arduino.h:43,
                 from c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\celt.h:38,
                 from c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.h:17,
                 from c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:15:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:13:17: error: expected ',' or '...' before numeric constant
   13 | #define TF_CS   4   // TF (Micro SD Card) CS pin
      |                 ^
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:20:12: note: in expansion of macro 'TF_CS'
   20 | #define SS TF_CS
      |            ^~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.h:1031:48: note: in expansion of macro 'SS'
 1031 | void     silk_resampler_private_down_FIR(void* SS, int16_t out[], const int16_t in[], int32_t inLen);
      |                                                ^~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:13:17: error: expected ',' or '...' before numeric constant
   13 | #define TF_CS   4   // TF (Micro SD Card) CS pin
      |                 ^
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:20:12: note: in expansion of macro 'TF_CS'
   20 | #define SS TF_CS
      |            ^~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.h:1033:47: note: in expansion of macro 'SS'
 1033 | void     silk_resampler_private_IIR_FIR(void* SS, int16_t out[], const int16_t in[], int32_t inLen);
      |                                               ^~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:13:17: error: expected ',' or '...' before numeric constant
   13 | #define TF_CS   4   // TF (Micro SD Card) CS pin
      |                 ^
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:20:12: note: in expansion of macro 'TF_CS'
   20 | #define SS TF_CS
      |            ^~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.h:1035:54: note: in expansion of macro 'SS'
 1035 | void     silk_resampler_private_up2_HQ_wrapper(void* SS, int16_t* out, const int16_t* in, int32_t len);
      |                                                      ^~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:13:17: error: expected ',' or '...' before numeric constant
   13 | #define TF_CS   4   // TF (Micro SD Card) CS pin
      |                 ^
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\d32_pro/pins_arduino.h:20:12: note: in expansion of macro 'TF_CS'
   20 | #define SS TF_CS
      |            ^~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3537:52: note: in expansion of macro 'SS'
[... 73 lines cut ...]
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3836:43: error: too many arguments to function 'void silk_resampler_private_IIR_FIR(void*)'
 3836 |             silk_resampler_private_IIR_FIR(S, out, S->delayBuf, S->Fs_in_kHz);
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3611:6: note: declared here
 3611 | void silk_resampler_private_IIR_FIR(void*         SS,    /* I/O  Resampler state             */
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3837:43: error: too many arguments to function 'void silk_resampler_private_IIR_FIR(void*)'
 3837 |             silk_resampler_private_IIR_FIR(S, &out[S->Fs_out_kHz], &in[nSamples], inLen - S->Fs_in_kHz);
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3611:6: note: declared here
 3611 | void silk_resampler_private_IIR_FIR(void*         SS,    /* I/O  Resampler state             */
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3840:44: error: too many arguments to function 'void silk_resampler_private_down_FIR(void*)'
 3840 |             silk_resampler_private_down_FIR(S, out, S->delayBuf, S->Fs_in_kHz);
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3537:6: note: declared here
 3537 | void silk_resampler_private_down_FIR(void*         SS,    /* I/O  Resampler state             */
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3841:44: error: too many arguments to function 'void silk_resampler_private_down_FIR(void*)'
 3841 |             silk_resampler_private_down_FIR(S, &out[S->Fs_out_kHz], &in[nSamples], inLen - S->Fs_in_kHz);
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ESP32-audioI2S-master\src\opus_decoder\silk.cpp:3537:6: note: declared here
 3537 | void silk_resampler_private_down_FIR(void*         SS,    /* I/O  Resampler state             */
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

schreibfaul1, 2025-06-27:
I'm not very familiar with the Arduino IDE.
One possibility would be to download the zip file from Github and unzip it.
Then copy the src folder into your project. next to your *.ino
And then `#include "src/Audio.h"`

SpectraCoder, 2025-06-27:
Thanks for your quick response! I just tried something different, and it seems I can work around this problem when selecting a general "ESP WROVER MODULE" instead of the specific "LOLIN D32 PRO" in the Arduino IDE. Now it compiles fine, and it seems to work correctly. 🤷‍♂️

closed 2025-08-12 by github-actions[bot]