schreibfaul1/ESP32-audioI2S #1102: ESP32-audioI2S@^2.3.0 update 3.3.2

steelmak, 2025-07-12:
Hello, dear schreibfaul1.
My name is Timofey, I am from Russia. In PlatformIO I found and installed ESP32-audioI2S@^2.3.0 and successfully compiled the project. Web radio music plays. I want to use the Audio::getVUlevel function from version 3.3.2.
I specified the path to version 3.3.2 in the settings, but the project does not build. Errors appear, files are missing.
#include <NetworkClient.h>
#include <NetworkClientSecure.h>
#include <driver/i2s_std.h>
Probably additional settings are needed in the platformio.ini file.
I use the board ESP32-S3 Supermini (ESP32-S3FH4R2)
Now I have it like this:

[env]
framework = arduino
monitor_speed = 115200
lib_deps =

[env:esp32-s3-SuperMini]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
board_build.arduino.memory_type = qio_qspi
board_build.flash_mode = qio
board_build.psram_type = qio
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304
board_build.partitions = default.csv
build_flags =
-DARDUINO_USB_CDC_ON_BOOT=1
-DBOARD_HAS_PSRAM
board_build.f_cpu = 240000000L
lib_deps =
esphome/ESP32-audioI2S@^2.3.0

Please help me with the settings; I'm learning and don't understand everything.
My mail is [ntv1978@mail.ru](mailto:ntv1978@mail.ru)
I am not sure that you will read my question and I am not sure that I will get an answer here.
Thank you!

schreibfaul1, 2025-07-12:
This is new `.pio\build\esp32s3\https_server.crt.S' not found`

added it

board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt

2MB PSRAM is rare, if it is not octal you have to set quad in `menuconfig`.
The size of the PSRAM is recognised automatically

steelmak, 2025-07-12:
https://aliexpress.ru/item/1005007524094315.html
ESP32-S3 Supermini (ESP32-S3FH4R2)
I liked the compact size, large number of pins and low price.

[image]

[image]

schreibfaul1, 2025-07-12:
Yes, I have added the certificate files. https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF/blob/master/platformio.ini
If it compiles and you are in an endless loop afterwards, it is probably the PSRAM

[image]

steelmak, 2025-07-12:
I replaced (copied) the platformio.ini file
Selected env:esp32s3
By default framework = arduino, espidf
Then pressed the PlatformIO: Clean button
Then successfully built the project (there were several warnings).
Tomorrow I will try working with the board.
Thanks.

steelmak, 2025-07-13:
Hello, dear schreibfaul1.
I managed to make some progress.
1. In _menuconfig_ you need to select
(X) Quad Mode PSRAM
( ) Octal Mode PSRAM

[image]

2. In the _platformio.ini_ file, open the lines

    board_build.partitions = boards/dev-4MB.csv    ;  4MB Flash
    board = ESP32-S3-DevKitC-1-N8R2  ;  8 MB Flash, 2MB have quad PSRAM --> must change in menuconfig

3. In the  _ESP32-S3-DevKitC-1-N8R2.json_ we correct in the file

    "flash_size": "4MB",
    "maximum_size": 4194304,

[image]

4. We assemble the project and observe

[image]

5. In the default code I leave the definition of I2S lines.

_#define I2S_DOUT      1
#define I2S_BCLK      2
#define I2S_LRC       3_

This connection was previously checked in the previous version - ESP32-audioI2S@^2.3.0
Web radio plays correctly; there is sound.
I couldn't figure out how to set up the terminal to receive responses from the board. I connected an external USB-UART.

[image]

Note: The SCK line is connected to GND - a jumper is soldered on the board.
Note 2: In your examples on the diagrams the DAC power supply is connected to +3.3V or +5V. I tested with 5V power supply.

[image]

6. After switching on, the board connects to the access point very quickly, 1-2 seconds.
To evaluate the loop's performance, I added an LED to the board.

int Led = 48;
bool LedState;
unsigned long TickLed;

steelmak, 2025-07-13:
6. After switching on, the board connects to the access point very quickly, 1-2 seconds.
To evaluate the loop's performance, I added an LED to the board.

int Led = 48;
bool LedState;
unsigned long TickLed;
...
    WiFi.setHostname("Radio"); // <===
    WiFi.begin(ssid.c_str(), password.c_str());
    pinMode(Led, OUTPUT);
    while (WiFi.status() != WL_CONNECTED){
     LedState = ! LedState; 
     digitalWrite(Led, LedState);
     delay(500);
    }    
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // default 0...21
    audio.connecttohost("http://85.236.172.22:8000/sammarusya"); // 
}

void loop() {
    audio.loop();
    vTaskDelay(1);
    if (millis() > TickLed){
    TickLed = millis() + 100L; 
    LedState = ! LedState; 
    digitalWrite(Led,LedState);
    }
}

The home router sees our board.

[image]

However, there is no sound (no playback).
Next, let's look at the data from the Port.

steelmak, 2025-07-13:
[image]

[image]

[image]

steelmak, 2025-07-14:
Signals on the BCLK and LRC lines.
No DOUT signal.
No useful signal on other board pins (except UART).

[image]
[image]

schreibfaul1, 2025-07-19:
One explanation is that the frequency at the I2S output is always 48KHz.
Previously it was dependent on the sample rate of the source, usually 44.1 or 22.05KHz

steelmak, 2025-07-20:
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp: In member function 'bool I2SClass::initSTD(uint32_t, i2s_data_bit_width_t, i2s_slot_mode_t, int8_t)':
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp:26:3: warning: missing initializer for member 'i2s_chan_config_t::allow_pd' [-Wmissing-field-initializers]
   26 |   }
      |   ^
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp:344:32: note: in expansion of macro 'I2S_DEFAULT_CFG'
  344 |   i2s_chan_config_t chan_cfg = I2S_DEFAULT_CFG();
      |                                ^~~~~~~~~~~~~~~
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp: In member function 'bool I2SClass::initTDM(uint32_t, i2s_data_bit_width_t, i2s_slot_mode_t, int8_t)':
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp:26:3: warning: missing initializer for member 'i2s_chan_config_t::allow_pd' [-Wmissing-field-initializers]
   26 |   }
      |   ^
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp:453:32: note: in expansion of macro 'I2S_DEFAULT_CFG'
  453 |   i2s_chan_config_t chan_cfg = I2S_DEFAULT_CFG();
      |                                ^~~~~~~~~~~~~~~
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp: In member function 'bool I2SClass::initPDMtx(uint32_t, i2s_data_bit_width_t, i2s_slot_mode_t)':
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp:26:3: warning: missing initializer for member 'i2s_chan_config_t::allow_pd' [-Wmissing-field-initializers]
   26 |   }
      |   ^
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp:544:32: note: in expansion of macro 'I2S_DEFAULT_CFG'
  544 |   i2s_chan_config_t chan_cfg = I2S_DEFAULT_CFG();
      |                                ^~~~~~~~~~~~~~~
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp: In member function 'bool I2SClass::initPDMrx(uint32_t, i2s_data_bit_width_t, i2s_slot_mode_t)':
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp:26:3: warning: missing initializer for member 'i2s_chan_config_t::allow_pd' [-Wmissing-field-initializers]
   26 |   }
      |   ^
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_I2S/src/ESP_I2S.cpp:628:32: note: in expansion of macro 'I2S_DEFAULT_CFG'
  628 |   i2s_chan_config_t chan_cfg = I2S_DEFAULT_CFG();
      |                                ^~~~~~~~~~~~~~~
Compiling .pio\build\esp32s3\ESP_SR.cpp.o
Compiling .pio\build\esp32s3\esp32-hal-sr.c.o
C:/Users/User/.platformio/packages/framework-arduinoespressif32/libraries/ESP_SR/src/esp32-hal-sr.c:10:2: warning: #warning Compatible partition must be selected for ESP_SR 
to work [-Wcpp]
   10 | #warning Compatible partition must be selected for ESP_SR to work
      |  ^~~~~~~

[image]

steelmak, 2025-07-26:
Good day, dear schreibfaul1.
Thank you. Your invaluable advice helped me.
The radio works, the volume indicator works.
I made time synchronization, a weather forecast with a running line, 
configured 20 stations (names and addresses) via the web interface, 
and the scheduled operation mode.
Now it remains to make the case.
The case is the most complex and unfinished part.
I am attaching a photo and a link to a video (in Russian).
[https://rutube.ru/video/private/048571d6a69bc1809dfa625c38b43280/?p=bCjWOZ7expupXULmmkUllA?r=plwd](url)

[image]

[image]

[image]

[image]

steelmak, 2025-07-30:
Good day, dear schreibfaul1.
Question - is there a way (flag or function) to determine the active state of audio playback?
One of the web stations stops playing for unknown reasons. Here is its address
StURL = "https://ic1.radiosignal.one/vanya-mp3"

Repeat call of the function starts playback
audio.connecttohost(StURL.c_str());

The body of the web receiver is almost ready.

[image]

steelmak, 2025-08-02:
Good day, dear schreibfaul1.
I use a running line to output Streamtitle.
Some radio stations (news, talk) do not have Streamtitle data.
When switching to these stations, the "Streamtitle" line contains garbage and, accordingly, a certain length. 
After some time, the line is cleared and acquires zero length. 
But the garbage manages to appear on the screen at the moment of switching.
How to fix this?

An example of such a station
http://orfeyfm.hostingradio.ru:8034/orfeyfm128.mp3

```
const char *txtStreamtitle = NULL; // global pointer to string
uint16_t LenStreamtitle;

// E V E N T S showstreamtitle
void audio_showstreamtitle(const char *info){
    if ( strlen(info)>7 && (LenStreamtitle != strlen(info)))
    {
      LenStreamtitle = strlen(info);
      txtStreamtitle = info;
    }
}
```

schreibfaul1, 2025-08-04:
It looks as if station [x](http://orfeyfm.hostingradio.ru:8034/orfeyfm128.mp3) is not transmitting any streaming titles.
In this case, the "Icy description" is displayed once at the beginning.

If you want to use the "info" outside the event you can create a local copy

````c++
char *txtStreamtitle = NULL; // global pointer to string

// E V E N T S showstreamtitle
void audio_showstreamtitle(const char *info){
    if (strlen(info))  {
        if(txtStreamtitle){free(txtStreamtitle), txtStreamtitle = NULL;}
        txtStreamtitle = strdup(info);
    }
}
````

steelmak, 2025-08-04:
This is how I do it (but I think this is a boring code):

```
uint16_t vu = audio.getVUlevel();
uint8_t Lval =  vu >> 8;
uint8_t Rval = vu & 0xFF; 

 uint16_t sum = (Lval + Rval) / 2;
 if (sum>159          ) {Lval >>= 4; Rval >>= 4;}
 if (sum>79 && sum<160) {Lval >>= 3; Rval >>= 3;}
 if (          sum< 80) {Lval >>= 2; Rval >>= 2;}
```

closed 2025-09-18 by github-actions[bot]