schreibfaul1/ESP32-audioI2S #1106: audio.connecttospeech not working

Dcmwrnd, 2025-07-18:
Hi Schreibfaul1,

Recently I updated the Audio library to v3.3.2q and the 'audio.connecttospeech' is not working. It was working before on v3.3.2f. This new version is giving continuous error on the Serial Monitor output here:-

WiFi connected.
IP address: 192.168.0.6
info        inputBufferSize: 655349 bytes[0m
info        buffers freed, free Heap: 226956 bytes[0m
info        connect to "translate.google.com.vn"[0m
info        c:\Users\David Chiang\Documents\Arduino2\libraries\ESP32-audioI2S-master\src\Audio.cpp:3879:[31m m_currentHost is empty[0m
info        c:\Users\David Chiang\Documents\Arduino2\libraries\ESP32-audioI2S-master\src\Audio.cpp:3879:[31m m_currentHost is empty[0m
info        c:\Users\David Chiang\Documents\Arduino2\libraries\ESP32-audioI2S-master\src\Audio.cpp:3879:[31m m_currentHost is empty[0m
info        c:\Users\David Chiang\Documents\Arduino2\libraries\ESP32-audioI2S-master\src\Audio.cpp:3879:[31m m_currentHost is empty[0m
info        c:\Users\David Chiang\Documents\Arduino2\libraries\ESP32-audioI2S-master\src\Audio.cpp:3879:[31m m_currentHost is empty[0m

Here's my sketch below.
Regards.

/*
Hardware:
MCU:	ESP32-S3 N16R8
Audio Board:	PCM 5102A

Compile with:-
Arduino IDE:	v2.3.6
Board Manager:	v3.2.1
Audio.h:	v3.3.2f

Board configuration in Arduino IDE v2.3.6
-----------------------------------------
Tools -> Board -> esp32 -> ESP32S3 Dev Module

Tools -> USB CDC On Boot: "Disabled"
Tools -> CPU Frequency: "240MHz (WiFi)"
Tools -> Core Debug Level: "None"
Tools -> USB DFU On Boot: "Disabled"
Tools -> Erase All Flash Before Sketch Upload: "Disabled" 
Tools -> Events Run On: "Core 1"
Tools -> Flash Mode: "QIO 80MHz"
Tools -> Flash Size: "16MB (128Mb)"
Tools -> Jtag Adapter: "Disabled"
Tools -> Arduino Runs On: "Core 1"
Tools -> USB Firmware MSC On Boot: "Disabled"
Tools -> Partition Scheme: "Custom" (7.8MB APP, OTA / 512KB SPIFFS)
Tools -> PSRAM: "OPI PSRAM"
Tools -> Upload Mode: "UART0 / Hardware CDC"
Tools -> Upload Speed: "921600"
Tools -> USB Mode: "Hardware CDC and JTAG"
Tools -> Zigbee Mode: "Disabled"
*/

#include <WiFi.h>
#include "Audio.h"

#define I2S_DOUT 13
#define I2S_BCLK 12
#define I2S_LRC  14

int eos = 0;

Audio audio;

const char* ssid = "xxxxxx";
const char* password = "xxxxxx";

void setup() {

  Serial.begin(115200);
  while (!Serial);
  delay(200);

  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(12); // 0...21

  Serial.print("\nConnecting to Local Wifi Network: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while (!WiFi.isConnected()) {
  delay(1000);
  }

  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("\nWiFi connected.");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  } else {
    Serial.println("\nNo WiFi connection.\n");
  }
}

void loop() {
  vTaskDelay(1);
  audio.loop();

  if (eos == 0) {
    eos++;
    audio.connecttospeech("Ladies and gentlemen, welcome to Internet of Things.", "en");
  }

  if (eos == 2) {
    eos++;
    audio.connecttohost("http://live.dancemusic.ro:7000/stream");
  }
}

// optional
void audio_info(const char *info){
  Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
  Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
  Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
  Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
  Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
  Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
  Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
  Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
  Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
  Serial.print("eof_speech  ");Serial.println(info);
  eos++;
}

schreibfaul1, 2025-08-07:
Thank you, I will take a look at that in the near future.

schreibfaul1, 2025-08-07:
Please use
````c++
void audio_eof(const char *info){  //end of file
    Serial.print("eof     ");Serial.println(info);
}
````

for all eof events,
audio_eof_speech() no longer works

Dcmwrnd, 2025-08-26:
Hi Schreibfaul1,

It happened again, after Google TTS speech, it does not trigger 'void audio_eof(const char *info)' on the latest v3.4.2b

Regards