schreibfaul1/ESP32-audioI2S #1118: Example Compilation fails under Arduino IDE

rodolfozunino, 2025-07-30:
I'm using Arduino IDE 2.3.2, ESP32-AudioI2S downloaded most recent version as of today

I installed the Library in the Arduino IDE from ZIP file, and copied/pasted the basic example provided in the GitHub main page (I'm not posting it here because I did not modify anything therein)

Platform: ESP32 Dev module, Huge APP, PSRAM-> Enabled  (I also tried PSRAM disabled, same result)

Compilation reports the following errors (sorry for brute pasting), which seem to relate to cstdint include defs:
What am I doing wrong?

Thanks for helping, 

In file included from c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:303,
                 from c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:33,
                 from C:\Temp\Nuova cartella\sketch_jul30a\sketch_jul30a.ino:3:
C:/Users/rodolfo/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/c++/13.2.0/cstdint:48:1: error: expected unqualified-id before 'namespace'
   48 | namespace std
      | ^~~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:479:5: error: 'requires' does not name a type
  479 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:479:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:520:5: error: 'requires' does not name a type
  520 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:520:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:560:5: error: 'requires' does not name a type
  560 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:560:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:581:5: error: 'requires' does not name a type
  581 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:581:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:602:5: error: 'requires' does not name a type
  602 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:602:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:625:5: error: 'requires' does not name a type
  625 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:625:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:666:5: error: 'requires' does not name a type
  666 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:666:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:708:5: error: 'requires' does not name a type
  708 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:708:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:762:5: error: 'requires' does not name a type
  762 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:762:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:833:5: error: 'requires' does not name a type
  833 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:833:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:848:5: error: 'requires' does not name a type
  848 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:848:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:874:5: error: 'requires' does not name a type
  874 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:874:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:899:5: error: 'requires' does not name a type
  899 |     requires std::is_same_v<U, char>
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:899:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:938:5: error: 'requires' does not name a type
[... 17 lines cut ...]
      |     ^~~~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:1422:5: note: 'requires' only available with '-std=c++20' or '-fconcepts'
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp: In member function 'void ps_ptr<T>::truncate_at(std::size_t)':
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/psram_unique_ptr.hpp:1465:25: error: too few arguments to function 'size_t strlen(const char*)'
 1465 |         if(pos >= strlen()) return;
      |                   ~~~~~~^~
In file included from C:\Users\rodolfo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Arduino.h:29,
                 from C:\Temp\Nuova cartella\sketch_jul30a\sketch_jul30a.ino:1:
C:/Users/rodolfo/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/string.h:41:10: note: declared here
   41 | size_t   strlen (const char *);
      |          ^~~~~~
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h: In member function 'ps_ptr<char> Audio::urlencode(const char*, bool)':
c:\Users\rodolfo\Documents\Arduino\libraries\ESP32-audioI2S-master\src/Audio.h:734:17: error: 'class ps_ptr<char>' has no member named 'shrink_to_fit'
  734 |         encoded.shrink_to_fit();
      |                 ^~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\rodolfo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\libraries\WiFi
  Not used: C:\Users\rodolfo\Documents\Arduino\libraries\WiFiNINA
Multiple libraries were found for "SD.h"
  Used: C:\Users\rodolfo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\libraries\SD
  Not used: C:\Users\rodolfo\AppData\Local\Arduino15\libraries\SD
  Not used: C:\Users\rodolfo\Documents\Arduino\libraries\SD
exit status 1

Compilation error: exit status 1

schreibfaul1, 2025-07-30:
You can unzip the file from Github and copy the src folder into your project next to the ino file.

And then:
`#include "src/Audio.h"`

rodolfozunino, 2025-07-30:
Thanks, after upgrading to ESP32 board ver 3.2.1 (I used 3.1.0 for other projects) compiles correctly