schreibfaul1/ESP32-audioI2S #1120: hang up after certain time

aber68, 2025-07-30:
hello,

i use ai-thinker (ESP board version 3.30, libs are on latest) module with 4MB flas and PSRAM as an arduino internet radio. after a certain time (40-80min) the RSSI goes from -5x to -7x, the inbuffer drops and seconds later the module reboots and starts playing normally - following numbers are debugged:

**RSSI 			-57
Bit rate		128000
in Buffer		524288
Buffer filled	259200
Buffer free	265088

RSSI			-70
WiFi connection lost

RSSI 			-56
Bit rate		128000
in Buffer		524288
Buffer filled	3072
Buffer free	521216**

I have no idea what's going on here.

best regards
andreas

tinti61, 2025-08-04:
Hello,

i observe exact the same behavior on some ESP32. The range is from RSSI -34 to RSSI -76. The RSSI values are read from the repeater/accesspoint. The time for reboot is abaout 10 to 20 minutes. I tested several esp32 DevKitC V4. The accesspoint are several AVM Fritz!Repeater and a TP-Link accesspoint. I notice it is independent of the accesspoint, distance, channel and temperature. I believe with this fault there is a batch production of ESP32, because some are ok others not.

best regards
Thomas

aber68, 2025-08-04:
For me, this addition has almost solved the problem (the short pauses for reconnection are practically inaudible and don't occur very often).

```
   if (!checkBufferFilled && audio.inBufferFilled() > 16000) checkBufferFilled = true;  // 16 KB als start Füllerkennung
   if (millis() - checkBuffer > 3000) {         // debug Interval
      checkBuffer = millis();
      if (checkBufferFilled && audio.inBufferFilled() < 15000) {
         // LOG("Buffer nearly empty ( ", audio.inBufferFilled(), ") - reconnecting ...");
         audio.stopSong();
         audio.connecttohost(Stream2play.c_str());
         checkBufferFilled = false;             // Neustart der Füllerkennung
      }
   }
```

closed 2025-09-18 by github-actions[bot]