schreibfaul1/ESP32-audioI2S #1132: Hangup when playing/loading next larger mp3 file

thomas-mueller898, 2025-08-08:
Dear schreibfaul1,
I ran into a problem using your library (still V.3.30 because the newer versions didn't work) and an ESP32-S3 Dev module. I'm using the newest Arduino IDE (2.3.6) and the newest ESP32-Core. Here's the code that I use:

`#include "Arduino.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"

#define SD_CS          5
#define SPI_MOSI      10    // SD Card
#define SPI_MISO      14
#define SPI_SCK       18

#define I2S_DOUT      13
#define I2S_BCLK      12    // I2S
#define I2S_LRC       11

Audio audio;

void SDSetup() {
  pinMode(SD_CS, OUTPUT);
  digitalWrite(SD_CS, HIGH);
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SD.begin(SD_CS);
}

void next() {
  audio.stopSong();
  audio.connecttoFS(SD, "/music/test2.mp3");
}

void setup() {
  SDSetup();

  Serial.begin(115200);

  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(15); // 0...21
  audio.connecttoFS(SD, "/music/test.mp3");
}
void loop() {
  audio.loop();
  if (!audio.isRunning()) { 
    next();
  }
}
`
Any help is appreciated.
When test.mp3 and test2.mp3 are **short files** (ca.5 s), it works flawlessly, but with larger files, it crashed before or after loading test2.mp3.

Here are some screnshots:
[image]
Because the backtrace wasn't 100% visible: Backtrace: 0x4204e13e:0x3fcebd90 0x42049b56:0x3fcebdc0 0x42047ac1:0x3fcebde0 0x42047c8d:0x3fcebe00 0x42047cdd:0x3fcebe20 0x4208ce16:0x3fcebe50 0x4208e8f6:0x3fcebe70 0x4209069d:0x3fcebeb0 0x4205e521:0x3fcebee0 0x420e59e3:0x3fcebf00 0x420e5686:0x3fcebf20 0x420e507e:0x3fcebf40 0x420e5101:0x3fcebf70 0x42048b0f:0x3fcebf90 0x420486bd:0x3fcebfb0 0x4200bdbc:0x3fcebfd0 0x42010a85:0x3fcec000 0x4200228d:0x3fcec020 0x4204f8c8:0x3fcec040 0x4037f2e9:0x3fcec060

[image]

Thanks in advice.

schreibfaul1, 2025-08-08:
Please try using the `audio_eof()` event and the latest version.
Or, for an older version, `audio_eof_mp3()`

````c++
....
....
Audio audio;

void SDSetup() {
    pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    SD.begin(SD_CS);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // default 0...21
    audio.connecttoFS(SD, "/music/test.mp3");
}

void loop(){
    audio.loop();
    vTaskDelay(1);
}

void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}

int idx = 0;
void audio_eof(const char *info){  //end of file
    Serial.print("eof         ");Serial.println(info);
    idx++;
    if(idx == 1) audio.connecttoFS(SD, "/music/test2.mp3");
    if(idx == 2) audio.connecttoFS(SD, "/music/test3.mp3");
    if(idx == 3) Serial.println("stop");
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}

thomas-mueller898, 2025-08-09:
Thanks! I tried the eof_mp3() function and it worked for both short and long mp3s
!

closed 2025-09-24 by github-actions[bot]