schreibfaul1/ESP32-audioI2S #1135: An m3u8 that doesn't work (json)

stef-ladefense, 2025-08-13:
Is there any hope of being able to decode this type of stream?

`https://live.m6radio.quortex.io/webpHJPXnXrN7B6J7Q8mcqmxP/webradio/rtl/202/audio-64000/index.m3u8
`
audio_info: buffers freed, free Heap: 170272 bytes
audio_info: connect to: "live.m6radio.quortex.io" on port 443 path "/webpHJPXnXrN7B6J7Q8mcqmxP/webradio/rtl/202/audio-64000/index.m3u8"
audio_info: SSL has been established in 344 ms
audio_info: ContentType application/json not supported

Thank

schreibfaul1, 2025-08-15:
That was a tough one.  I hope it's running smoothly now.

stef-ladefense, 2025-08-17:
Hello, I tested it with 3.4.1j, and the m3u8 station still doesn't work.
-->
audio_info: buffers freed, free Heap: 182472 bytes
audio_info: connect to: "live.m6radio.quortex.io" on port 443 path "/webpHJPXnXrN7B6J7Q8mcqmxP/webradio/rtl/202/audio-64000/index.m3u8"
audio_info: SSL has been established in 366 ms
audio_info: redirect to new host "https://hls.rtl.fr/NDgwYWYxMWE5NDgwODY4MTYyNjZjYTA1M2ZmMDJmYWMuMTc1NTQ5MTUwMQ~~/radio/webOU6MESMYnoHzvZJe8ie5Mi/webradio/les-grosses-tetes/short/index.m3u8"
audio_info: next URL: "https://hls.rtl.fr/NDgwYWYxMWE5NDgwODY4MTYyNjZjYTA1M2ZmMDJmYWMuMTc1NTQ5MTUwMQ~~/radio/webOU6MESMYnoHzvZJe8ie5Mi/webradio/les-grosses-tetes/short/index.

However, now that it's new, this stations no longer works (it worked with <=3.4.1i). 

https://rfimonde-96k.ice.infomaniak.ch/rfimonde-96k.mp3
http://ouifm.ice.infomaniak.ch/ouifm-high.aac
http://icecast.radiofrance.fr/fb1071-midfi.aac
http://icecast.radiofrance.fr/franceculture-midfi.aac
http://icecast.radiofrance.fr/franceinfo-midfi.aac
http://icecast.radiofrance.fr/franceinter-midfi.aac
http://icecast.radiofrance.fr/francemusique-midfi.aac
http://europe1.lmn.fm/europe1.aac
http://europe2.lmn.fm/europe2.aac
http://icecast.skyrock.net/s/natio_aac_64k

-->
audio_info: buffers freed, free Heap: 182748 bytes
audio_info: connect to: "rfm.lmn.fm" on port 80 path "/rfm.aac"
audio_info: Connection has been established in 26 ms
audio_info: chunked data transfer
audio_info: icy-name: RFM
audio_info: icy-genre:
audio_info: icy-bitrate: 128000
audio_info: AACDecoder has been initialized
audio_info: StreamUrl=''
->streamtitle 52.46.4D.00.
->streamtitle RFM
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: aac_decoder.cpp:183: Gain control not yet implemented
audio_info: syncword not found
audio_info: syncword found at pos 99
audio_info: syncword found at pos 0
audio_info: aac_decoder.cpp:183: Gain control not yet implemented
audio_info: syncword not found
audio_info: syncword found at pos 118
audio_info: syncword found at pos 0
...
audio_info: aac_decoder.cpp:183: Gain control not yet implemented
audio_info: syncword found at pos 379
audio_info: syncword found at pos 0
audio_info: aac_decoder.cpp:183: Bitstream value not allowed by specification
audio_info: syncword found at pos 442
audio_info: syncword found at pos 0
audio_info: aac_decoder.cpp:183: Unexpected channel configuration change
audio_info: AAC channel config changed to 2
audio_info: aac_decoder.cpp:183: Bitstream value not allowed by specification
audio_info: syncword found at pos 359
audio_info: syncword found at pos 0
audio_info: aac_decoder.cpp:183: Unexpected channel configuration change
audio_info: AAC channel config changed to 2
audio_info: aac_decoder.cpp:183: Bitstream value not allowed by specification
audio_info: syncword found at pos 334
audio_info: syncword found at pos 0
audio_info: aac_decoder.cpp:183: Gain control not yet implemented

schreibfaul1, 2025-08-24:
Some stations require the port to be specified in the HTTP request header, while others do not if they are standard ports, which is strange.

Arielhh, 2025-09-07:
I still have that issue with 3.4.2i (the latest) when trying to play that station that was playing fine in the past https://bss1.neterra.tv/veselina/veselina.m3u8

closed 2025-09-22 by stef-ladefense